jsk-ros-pkg/jsk_recognition #634: travisのテストが終わらない

garaemon, 2015-01-22:
プログラムが増えてきて、結構な頻度で50分以上のコンパイル時間がかかってしまうようです

どうするのが正解でしょうか？

1. 頑張ってコンパイル時間が短いコードを書く
2. jsk_pcl_rosとjsk_perceptionのテストを分ける(依存をなくす)
3. ほぼnodeletなのだなら、片っ端から.cppをincludeして一つの巨大.cppにしてコンパイルしてしまう

k-okada, 2015-01-22:
2ですかね．

garaemon, 2015-01-22:
メッセージに依存があるのが△なので、jsk_recognition_msgsみたいなものを作ってそこにまとめる方向ですかね。

garaemon, 2015-01-22:
以下のような手順ですかね

1. jsk_recognition_msgsを追加。

  jsk_commonにあるべきパッケージだが、試行錯誤しやすいようにjsk_recognitionに配置

2. debリリース
3. jsk_travisにどのパッケージをコンパイルするかの指示ができるようにする
4. jsk_pcl_rosとjsk_perceptionをjsk_recogntion_msgsを使う形に変更。テストもかえる
5. jsk_control, jsk_visualizationもjsk_recognition_msgsを使うように変更する

garaemon, 2015-01-24:
* jsk_recognition_msgsを追加してrelease

garaemon, 2015-01-27:
jsk_visualization is ready
* https://github.com/jsk-ros-pkg/jsk_visualization/pull/296
* https://github.com/jsk-ros-pkg/jsk_visualization/pull/295

garaemon, 2015-01-28:
jsk_recognition is almost ready
* https://github.com/jsk-ros-pkg/jsk_recognition/pull/665

garaemon, 2015-01-28:
jsk_control is almost ready
* https://github.com/jsk-ros-pkg/jsk_control/pull/205

garaemon, 2015-01-29:
almost done

garaemon, 2015-02-01:
a lot of work done!!

closed 2015-02-01 by garaemon