jsk-ros-pkg/jsk_recognition #701: (ros::roseus-add-srvs "jsk_pcl_ros") fails.

furushchev, 2015-02-13:
after `catkin build`, roseus command `(ros::roseus-add-srvs "jsk_pcl_ros")` fails.

```lisp
$ roseus
1.irteusgl$ ros::roseus-add-srvs "jsk_pcl_ros"
/opt/ros/hydro/share/euslisp/jskeus/eus/Linux64/bin/irteusgl 0 error: no such package JSK_RECOGNITION_MSGS in (apply #'load-org-for-ros fullname args)
```

But

```lisp
$ roseus
1.irteusgl$ ros::roseus-add-msgs "jsk_recognition_msgs"
nil
2.irteusgl$ ros::roseus-add-srvs "jsk_pcl_ros"
nil
```

it works.

garaemon, 2015-02-13:
Have you compiled services?

✉︎ Ryohei Ueda

furushchev, 2015-02-13:
@garaemon I just compiled as `catkin build jsk_pcl_ros`,
then reopen terminal, then sourced /opt/ros/hydro, ~/ros/hydro_parent/devel, ~/ros/hydro/devel.

garaemon, 2015-02-13:
`catkin build jsk_pcl_ros` does not generate services/messages for roseus.
Please make dependency in sub-project like drc_task_common:
https://github.com/jsk-ros-pkg/jsk_demos/blob/master/jsk_2015_06_hrp_drc/drc_task_common/catkin.cmake#L4

furushchev, 2015-02-13:
@garaemon thanks for advice...
now let's talk with example `detect_cans_in_fridge_201202`,
I changed catkin settings as follow:

```diff
$ git diff
diff --git a/detect_cans_in_fridge_201202/CMakeLists.txt b/detect_cans_in_fridge
index e4b64b1..91622c6 100644
--- a/detect_cans_in_fridge_201202/CMakeLists.txt
+++ b/detect_cans_in_fridge_201202/CMakeLists.txt
@@ -3,6 +3,7 @@ project(detect_cans_in_fridge_201202)
 
 find_package(catkin REQUIRED COMPONENTS
   jsk_perception
+  jsk_recognition_msgs
   jsk_pcl_ros
   pr2_navigation_self_filter
   jsk_pr2_startup
diff --git a/detect_cans_in_fridge_201202/package.xml b/detect_cans_in_fridge_20
index c8922c9..83c2fa1 100644
--- a/detect_cans_in_fridge_201202/package.xml
+++ b/detect_cans_in_fridge_201202/package.xml
@@ -18,6 +18,7 @@
   <!-- Dependencies needed to compile this package. -->
   <build_depend>roseus</build_depend>
   <build_depend>jsk_perception</build_depend>
+  <build_depend>jsk_recognition_msgs</build_depend>
   <build_depend>jsk_pcl_ros</build_depend>
   <build_depend>pr2_navigation_self_filter</build_depend>
   <build_depend>jsk_pr2_startup</build_depend>
```

then compiled with `catkin build detect_cans_in_fridge_201202 --force-cmake`, after which still occurs the same error...

garaemon, 2015-02-13:
Please check ~/ros/hydro/devel/share/roseus/ros/jsk_pcl_ros directory and if you have it, remove it and run catkin build again

garaemon, 2015-02-13:
And also confirm you dont have ~/.ros/roseus directory

furushchev, 2015-02-13:
I executed as follow:

```bash
$ rm -rf ~/.ros/roseus
$ rm -rf ~/ros/hydro/devel/roseus/ros/jsk_pcl_ros
$ rm -rf ~/ros/hydro/devel/roseus/ros/jsk_recognition_msgs
# new terminal
$ catkin build detect_cans_in_fridge_201202 --force-cmake
# new terminal
$ source /opt/ros/hydro/setup.bash
$ source ~/ros/hydro_parent/devel/setup.bash
$ source ~/ros/hydro/devel/setup.bash
```

and,,, it still occurs error:

```bash
/opt/ros/hydro/share/euslisp/jskeus/eus/Linux64/bin/irteusgl roseus-error: no su
ch package JSK_RECOGNITION_MSGS in (apply #'load-org-for-ros fullname args), exi
tting...
[ INFO] [1423838187.447530687]: cell* ROSEUS_EXIT(context*, int, cell**)
[detect_cans-3] process has died [pid 16267, exit code 1, cmd /home/furushchev/r
os/hydro/src/jsk-ros-pkg/jsk_roseus/roseus/bin/roseus /home/furushchev/ros/hydro
/src/jsk-ros-pkg/jsk_demos/detect_cans_in_fridge_201202/euslisp/detect_cans.l __
name:=detect_cans __log:=/home/furushchev/.ros/log/51cbd16c-b374-11e4-b0d8-00151
7c3b779/detect_cans-3.log].
log file: /home/furushchev/.ros/log/51cbd16c-b374-11e4-b0d8-001517c3b779/detect_
cans-3*.log
```

now my env is:

```bash
$ ls ~/ros/hydro/devel/share/roseus/ros/jsk_pcl_ros
manifest.l  srv
$ ls ~/ros/hydro/devel/share/roseus/ros/jsk_recognition_msgs/
manifest.l  msg
$ ls ~/.ros/roseus/hydro/jsk_p
jsk_pcl_ros_EnvironmentLock.so        jsk_perception_WhiteBalance.so
jsk_perception_EuclideanSegment.so    jsk_perception_WhiteBalancePoints.so
jsk_perception_SetTemplate.so
```

garaemon, 2015-02-13:
Why do you have .sos under ~/.ros/roseus after `catkin build`?

furushchev, 2015-02-14:
@garaemon Sorry, this was just because of my bad bashrc.
I removed ros settings, re-removed `$HOME/.ros/roseus`, then cleaned by `catkin clean --all` and `catkin build`.
As you have said, `$HOME/.ros/roseus` directory no more exists, but still occurs the same error.

```lisp
 $ roseus 
configuring by "/opt/ros/hydro/share/euslisp/jskeus/eus//lib/eusrt.l"
;; readmacro ;; object ;; packsym ;; common ;; constants ;; stream ;; string ;; loader ;; pprint ;; process ;; hashtab ;; array ;; mathtran ;; eusdebug ;; eusforeign ;; coordinates ;; tty ;; history ;; toplevel ;; trans ;; comp ;; builtins ;; par ;; intersection ;; geoclasses ;; geopack ;; geobody ;; primt ;; compose ;; polygon ;; viewing ;; viewport ;; viewsurface ;; hid ;; shadow ;; bodyrel ;; dda ;; helpsub ;; eushelp ;; xforeign ;; Xdecl ;; Xgraphics ;; Xcolor ;; Xeus ;; Xevent ;; Xpanel ;; Xitem ;; Xtext ;; Xmenu ;; Xscroll ;; Xcanvas ;; Xtop ;; Xapplwin 
;; pixword ;; RGBHLS ;; convolve ;; piximage ;; pbmfile ;; image_correlation ;; oglforeign ;; gldecl ;; glconst ;; glforeign ;; gluconst ;; gluforeign ;; glxconst ;; glxforeign ;; eglforeign ;; eglfunc ;; glutil ;; gltexture ;; glprim ;; gleus ;; glview ;; toiv-undefined ;; fstringdouble irtmath irtutil irtc irtgeoc irtgraph pgsql irtgeo euspqp pqp irtscene irtmodel irtdyna irtrobot irtsensor irtbvh irtcollada irtpointcloud irtx eusjpeg euspng png irtimage irtglrgb 
;; extending gcstack 0x4f29420[16374] --> 0x535e960[32748] top=3789
irtgl irtviewer 
EusLisp 9.10( 1.0.2) for Linux64 created on host08.storm.ros.org(Sat Jan 24 05:53:25 UTC 2015)
roseus ;; loading roseus("f37fe05") on euslisp((9.10 host08.storm.ros.org Sat Jan 24 05:53:25 UTC 2015  1.0.2))
eustf roseus_c_util 1.irteusgl$ ros::roseus-add-srvs "jsk_pcl_ros"
/opt/ros/hydro/share/euslisp/jskeus/eus/Linux64/bin/irteusgl 0 error: no such package JSK_RECOGNITION_MSGS in (apply #'load-org-for-ros fullname args)
```

my detail env is:

```bash
~/ros/hydro$ catkin config
--------------------------------------------------------------------------------
Profile:                     default
Extending:          [cached] /home/furushchev/ros/hydro_parent/devel:/opt/ros/hydro
Workspace:                   /home/furushchev/ros/hydro
Source Space:       [exists] /home/furushchev/ros/hydro/src
Build Space:        [exists] /home/furushchev/ros/hydro/build
Devel Space:        [exists] /home/furushchev/ros/hydro/devel
Install Space:     [missing] /home/furushchev/ros/hydro/install
DESTDIR:                     None
--------------------------------------------------------------------------------
Isolate Develspaces:         False
Install Packages:            False
Isolate Installs:            False
--------------------------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
--------------------------------------------------------------------------------
Workspace configuration appears valid.
--------------------------------------------------------------------------------
~/ros/hydro$
~/ros/hydro$ echo $CMAKE_PREFIX_PATH
/home/furushchev/ros/hydro/devel:/home/furushchev/ros/hydro_parent/devel:/opt/ros/hydro
```

garaemon, 2015-02-14:
Check ~/ros/hydro/devel/share/roseus/ros/jsk_pcl_ros/srv

furushchev, 2015-02-14:
```bash
$ tree ~/ros/hydro/devel/share/roseus/ros/jsk_pcl_ros
/home/furushchev/ros/hydro/devel/share/roseus/ros/jsk_pcl_ros
|-- manifest.l
`-- srv
    |-- CallPolygon.l
    |-- CallSnapIt.l
    |-- CheckCircle.l
    |-- EnvironmentLock.l
    |-- EuclideanSegment.l
    |-- ICPAlign.l
    |-- ICPAlignWithBox.l
    |-- PolygonOnEnvironment.l
    |-- RobotPickupReleasePoint.l
    |-- SetDepthCalibrationParameter.l
    |-- SetPointCloud2.l
    |-- SwitchTopic.l
    |-- TowerPickUp.l
    |-- TowerRobotMoveCommand.l
    `-- TransformScreenpoint.l

1 directory, 16 files
```

```bash
$ cat manifest.l 
;;
;; DO NOT EDIT THIS FILE
;;
(ros::load-ros-package "jsk_pcl_ros")
(ros::load-ros-package "pcl_msgs")
(ros::load-ros-package "sensor_msgs")
(ros::load-ros-package "geometry_msgs")
(ros::load-ros-package "jsk_recognition_msgs")
```

It seems there is no problem.

garaemon, 2015-02-14:
I understand. please call ros::roseus-add-msgs "jsk_recognition_msgs" first

furushchev, 2015-02-14:
So, let's get back to beginning of this discussion (https://github.com/jsk-ros-pkg/jsk_recognition/issues/701#issue-57596910).

1. just calling `(ros::roseus-add-srvs "jsk_pcl_ros")` fails.
2. calling it after `(ros;;roseus-add-msgs "jsk_recognition_msgs")` works.

Most all of code is written only `(ros::roseus-add-srvs "jsk_pcl_ros")` in a file.
Must we fix this? Is it correct behavior of euslisp?

garaemon, 2015-02-14:
(ros::load-ros-manifest "jsk_pcl_ros") should work

furushchev, 2015-02-14:
Umm, ok, so you mean:

1. use `(ros::load-ros-manifest <package>)` to load entire msg/srvs with resolving dependencies.
2. use `(ros::roseus-add-msgs <package>)` to load msg/srvs only of the package.

Is it correct?

garaemon, 2015-02-14:
Yes, but if you can write patch for add-msgs to resolve dependency, it sounds good.But I think we should change geneus scrip to add require in msg.l and it's not so easy task

✉︎ Ryohei Ueda

Umm, ok, so you mean:

closed 2015-03-10 by furushchev