jsk-ros-pkg/jsk_recognition #715: jenkins testing does not end within 50 minutes

garaemon, 2015-02-17:
One solution is to use ccache with shareding cache by different docker containers.
https://github.com/jsk-ros-pkg/jsk_travis/issues/53

garaemon, 2015-02-17:
@k-okada Can we use multiple processors on jenkins?

garaemon, 2015-02-17:
I plan to change parallel option from `-p1` to `-p8` or something like that.

garaemon, 2015-02-17:
Another solution is to separate jsk_pcl_ros into more smaller packages. for example, jsk_pcl_ros_util, jsk_pcl_ros_base, jsk_pcl_ros_core.

And jsk_pcl_ros depends on them. In testing phase, we can compile them separately

k-okada, 2015-02-17:
> @k-okada Can we use multiple processors on jenkins?

I think so, using -p8 on jenkins is ncie to do.

garaemon, 2015-02-17:
https://github.com/jsk-ros-pkg/jsk_recognition/pull/716

garaemon, 2015-02-19:
solved by using jenkins.jsk

closed 2015-02-19 by garaemon