jsk-ros-pkg/jsk_recognition #762: What is the best way to make color image from organized point cloud?

aginika, 2015-03-09:
I want to create color image generated from organized point cloud (using pass_throughed point cloud as input, so some parts are null. I hope that those invalid points become black in the color image).

I used depth_image_creator, but it was monochrome.

garaemon, 2015-03-09:
We should use PointIndices and MaskImage.

1. generate PointIndices from organized pointcloud (You need to implement it)
2. convert PointIndices to MaskImage (jsk_pcl/PointIndicesToMaskImage)
3. apply mask image to the original image (jsk_perception/ApplyMaskImage)

aginika, 2015-03-10:
@garaemon  Thanks a lot !!
I added to #765 which you suggest to implement.
And I realied to generate image.
Thanks.

closed 2015-03-10 by aginika