jsk-ros-pkg/jsk_recognition #864: How can I check pointcloud is empty or not

h-kamada, 2015-04-25:
I have to check whether a pointcloud is empty or not.
If input empty pointcloud to kdtree.nearestKSearch, an error occurs.
[pcl::KdTreeFLANN::setInputCloud] Cannot create a KDTree with an empty input cloud!

aginika, 2015-04-25:
Maybe I don't understand what you mean fully, but below does not works for you?
```
pointcloud->points.empty ()
```

h-kamada, 2015-04-25:
Sorry for not explaining enough.
Input pointcloud is multisense/organized_image_points2.
In this topic , I think depth unknown points are set Inf or NaN. (Is this misunderstanding??)
So, if all points I chose were Inf or NaN, nearestKsearch fails. (I think it's reason of error)

I want to know if there are valid points (not Inf or NaN) in pointcloud.

aginika, 2015-04-25:
If you want to check one by one, u can use `isnan`
```
PointT p = cloud->points[i];
 if (!isnan(p.x) && !isnan(p.y) && !isnan(p.z)) {
     non_nan_cloud->points.push_back(p);
```

Just removing , you might be able to use below function 
```
std::vector<int> indices;
pcl::removeNaNFromPointCloud(*m.cloud,*m.cloud, indices);
```

aginika, 2015-04-25:
Belows are documantation

http://docs.pointclouds.org/trunk/namespacepcl.html
http://docs.pointclouds.org/trunk/filter_8hpp_source.html#l00046

h-kamada, 2015-04-25:
While reading http://docs.pointclouds.org/trunk/filter_8hpp_source.html#l00046 , I found is_dense attribute. 
This is it.
Thank you!

closed 2015-04-25 by h-kamada

h-kamada, 2015-04-25:
is_dense is true if no points are invalid. So, it was not I want.