jsk-ros-pkg/jsk_recognition #873: imagesiftで特徴点が見つからない

h-kamada, 2015-04-29:
冷蔵庫デモで使われている冷蔵庫マークの写真に関して、手元のPCで読み込みを行うと特徴点が見つからず、カメラ画像とのマッチングがそもそも行えなくなる。

[詳細]
jsk_perception/point_pose_extractorにおいて、テンプレートとなる画像を指定し、初期化の際にそのテンプレート画像を入力とし、imagesiftに対して特徴点を見つけるようservicecallしているのが現在の実装となっている。
この初期化において、imagesiftに渡された画像の特徴点が一つも見つからない。

h-kamada, 2015-04-29:
古田さんの手元のPCでは問題なく特徴点が見つかった。
launch をあげた際のterminalの表示は以下

NODES
  /
    camera (usb_cam/usb_cam_node)
    camera_image (topic_tools/relay)
    camera_info (topic_tools/relay)
    debug_view (image_view/image_view)
    fridge_demo_sift (imagesift/imagesift)
    fridge_detector (jsk_perception/point_pose_extractor)
    multi_view (image_view/image_view)

auto-starting new master
process[master]: started with pid [31914]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 57feb92e-ee2e-11e4-9b44-28d24433c19d
process[rosout-1]: started with pid [31927]
started core service [/rosout]
process[camera-2]: started with pid [31939]
process[camera_image-3]: started with pid [31990]
process[camera_info-4]: started with pid [32006]
process[fridge_demo_sift-5]: started with pid [32020]
process[fridge_detector-6]: started with pid [32040]
process[debug_view-7]: started with pid [32090]
process[multi_view-8]: started with pid [32128]
[ INFO] [1430284358.266420341]: imagesift: image: 36(size=921600), num: 936, sift time: 0.704s, total: 1.339s
initialize templates done
[ INFO] [1430284358.647103247]: imagesift: image: 57(size=921600), num: 931, sift time: 0.379s, total: 0.381s
[ INFO] [1430284358.837212554]: imagesift: image: 68(size=921600), num: 974, sift time: 0.188s, total: 0.190s
[ INFO] [1430284358.841551837]: imagesift: image: 0(size=45087), num: 0, sift time: 0.004s, total: 0.004s
[ERROR] [1430284358.841770608]: Template image was not set.
[ INFO] [1430284359.078881430]: imagesift: image: 74(size=921600), num: 1015, sift time: 0.234s, total: 0.237s
[ INFO] [1430284359.176700217]: imagesift: image: 81(size=921600), num: 971, sift time: 0.096s, total: 0.098s
^C[multi_view-8] killing on exit
[debug_view-7] killing on exit
[fridge_detector-6] killing on exit
[fridge_demo_sift-5] killing on exit
[camera_info-4] killing on exit
[camera_image-3] killing on exit
[camera-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

h-kamada, 2015-04-29:
上のterminalのlogにおいて
imagesift: image: 36(size=921600), num: 936, sift time: 0.704s, total: 1.339s
のようにsizeが921600となっている方は、カメラ画像の特徴抽出で、こちらはkeypointsが936個見つかるなど上手くいっている。

一方、
[ INFO] [1430284358.841551837]: imagesift: image: 0(size=45087), num: 0, sift time: 0.004s, total: 0.004s
とsizeが45087となっている方がテンプレートとしている画像で、こちらの特徴抽出が上手くいっていない。
テンプレートとして用いている画像はデフォルトと変わらないdetect_cans_in_fridge_201202/data/openni_rgb_fridge.pngを用いている。

古田さんのPCでは、このsizeが148830となっており、サイズが違うので何かがおかしい。

h-kamada, 2015-04-29:
launchファイルは以下

<launch>
  <node pkg="usb_cam" type="usb_cam_node" name="camera" />
  <node pkg="topic_tools" type="relay" name="camera_image"
        args="/camera/image_raw /multisense/left/image_rect_color" />
  <node pkg="topic_tools" type="relay" name="camera_info"
        args="/camera/camera_info /multisense/left/camera_info" />

  <node name="fridge_demo_sift" pkg="imagesift"
        type="imagesift" launch-prefix="nice -n +10" output="screen" >
    <remap from="image" to="/multisense/left/image_rect_color" />
    <remap from="camera_info" to="/multisense/left/camera_info"/>
    <param name="use_mask" value="false" />
    <remap from="mask" to="/attention_clipper/output/mask"/>
  </node>
  <node pkg="jsk_perception" type="point_pose_extractor"
        name="fridge_detector">
    <param name="template_filename" value="$(find detect_cans_in_fridge_201202)/data/openni_rgb_fridge.png"/>
    <param name="window_name" value="fridge_handle_upper"/>
    <param name="object_width" value="0.231"/>
    <param name="object_height" value="0.196"/>
    <param name="relative_pose" value="0.22 0.05 0.075  0.707107 0 -0.707107 0"/>
    <param name="viewer_window" value="false" />
  </node>

  <node pkg="image_view" type="image_view" name="debug_view">
    <remap from="image" to="/fridge_detector/debug_image" />
  </node>
  <node pkg="image_view" type="image_view" name="multi_view">
    <remap from="image" to="/multisense/left/image_rect_color" />
  </node>
</launch>

h-kamada, 2015-04-29:
間違えました。
<launch>
  <node pkg="usb_cam" type="usb_cam_node" name="camera" />
  <node pkg="topic_tools" type="relay" name="camera_image"
        args="/camera/image_raw /multisense/left/image_rect_color" />
  <node pkg="topic_tools" type="relay" name="camera_info"
        args="/camera/camera_info /multisense/left/camera_info" />

  <node name="fridge_demo_sift" pkg="imagesift"
        type="imagesift" launch-prefix="nice -n +10" output="screen" >
    <remap from="image" to="/multisense/left/image_rect_color" />
    <remap from="camera_info" to="/multisense/left/camera_info"/>
    <param name="use_mask" value="false" />
    <remap from="mask" to="/attention_clipper/output/mask"/>
  </node>
  <node pkg="jsk_perception" type="point_pose_extractor"
        name="fridge_detector">
    <param name="template_filename" value="$(find detect_cans_in_fridge_201202)/data/openni_rgb_fridge.png"/>
    <param name="window_name" value="fridge_handle_upper"/>
    <param name="object_width" value="0.231"/>
    <param name="object_height" value="0.196"/>
    <param name="relative_pose" value="0.22 0.05 0.075  0.707107 0 -0.707107 0"/>
    <param name="viewer_window" value="false" />
  </node>

  <node pkg="image_view" type="image_view" name="debug_view">
    <remap from="image" to="/fridge_detector/debug_image" />
  </node>
  <node pkg="image_view" type="image_view" name="multi_view">
    <remap from="image" to="/multisense/left/image_rect_color" />
  </node>
</launch>

h-kamada, 2015-04-29:
launchファイルうまくはれないんですね

h-kamada, 2015-04-29:
```
<launch>
  <node pkg="usb_cam" type="usb_cam_node" name="camera" />
  <node pkg="topic_tools" type="relay" name="camera_image"
        args="/camera/image_raw /multisense/left/image_rect_color" />
  <node pkg="topic_tools" type="relay" name="camera_info"
        args="/camera/camera_info /multisense/left/camera_info" />

  <node name="fridge_demo_sift" pkg="imagesift"
        type="imagesift" launch-prefix="nice -n +10" output="screen" >
    <remap from="image" to="/multisense/left/image_rect_color" />
    <remap from="camera_info" to="/multisense/left/camera_info"/>
    <param name="use_mask" value="false" />
    <remap from="mask" to="/attention_clipper/output/mask"/>
  </node>
  <node pkg="jsk_perception" type="point_pose_extractor"
        name="fridge_detector">
    <param name="template_filename" value="$(find detect_cans_in_fridge_201202)/data/openni_rgb_fridge.png"/>
    <param name="window_name" value="fridge_handle_upper"/>
    <param name="object_width" value="0.231"/>
    <param name="object_height" value="0.196"/>
    <param name="relative_pose" value="0.22 0.05 0.075  0.707107 0 -0.707107 0"/>
    <param name="viewer_window" value="false" /><!-- to disable window-->
  </node>

  <node pkg="image_view" type="image_view" name="debug_view">
    <remap from="image" to="/fridge_detector/debug_image" />
  </node>
  <node pkg="image_view" type="image_view" name="multi_view">
    <remap from="image" to="/multisense/left/image_rect_color" />
  </node>
</launch>
```

garaemon, 2015-04-29:
intrinsic matrixが0なのが問題でした。
エラーメッセージで対応

closed 2015-04-29 by garaemon