jsk-ros-pkg/jsk_recognition #907: Resize point cloud from Kinect2

aginika, 2015-05-05:
I tried to use OrganizedPassthrough and I can see the resized output in rviz, but the output topic has  1920 x 1080  according to `rostopic echo `'s width and hegiht ...(I expected 640 * 480 as I described in launch)
Can I really resize it?
Is it enough fast no matter whether I got really resized output or not?

garaemon, 2015-05-05:
filtered points are represented as nan point to keep it organized

aginika, 2015-05-05:
Is there any nodelet which can really trim the points and publish trimmed smaller points?(e.g. input: 1920x1080 -> outpu : 640 x480)

garaemon, 2015-05-05:
Passthrough with keep organized false or add flag to unorder pointcloud to organized passthrough

garaemon, 2015-05-05:
or apply crop_decimate to original depth image but i don't know it can handle float image

aginika, 2015-05-05:
hmm,Thanks a lot, I will use the organized passthrough

closed 2015-05-05 by aginika