jsk-ros-pkg/jsk_recognition #986: cannot change number of particles in particle filter tracker

wesleypchan, 2015-06-25:
Tried setting the particle_num param to reduce the number of particles to speed up tracker, but in rviz it seems to still show a large number of particles.

After some initial debugging, it seems like the number of particles is getting set properly to particle_num initially, but after a few calls to cloud_cb it gets reverted back to 1000 for some reason.

wesleypchan, 2015-06-25:
After more debugging, it seems like the line of code causing the problem is tracker_->compute().
After calling tracker_->compute(), the number of particles gets reverted to 1000.

wesleypchan, 2015-06-25:
Seems like setting max_particle_num_ = particle_num solves the problem, but I'm not sure if this is the correct fix.

garaemon, 2015-06-25:
particle_num is just an initial number of particles . The number of
particles changes

◉  ryohei

wesleypchan, 2015-06-25:
so if I want to limit the number of particles to make the tracker run faster what would be the proper way of doing it?

garaemon, 2015-06-25:
max_particle_num should work for the purpose

wesleypchan, 2015-07-30:
changing max_particle_num worked

closed 2015-07-30 by wesleypchan