jsk-ros-pkg/jsk_recognition #987: Particle filter tracker is broken?

wesleypchan, 2015-06-25:
There seems to be something wrong with the particle filter tracker as testing with it today, the performance seems to have degraded significantly compared to when I last used it about 9 months ago.

Indeed, a simple test of tracking a static mug placed on the ground between the current version of the code and the version 9 months ago showed that, the code from 9 months ago was able to track the static mug properly, but with the current code, the tracking result quickly drifts to the ground. 

The two versions I tested and compared are: 

1. commit  6adf320b0b8558a5c82aaa9d2de0d7c3cb33a49b from Sept 10 2014
video at http://www.jsk.t.u-tokyo.ac.jp/~wesleyc/videos/pcl_tracker_commit_6adf320.ogv

2. commit 3bb870449b1bc70f5bb76422eadbd7dd8fd921a4 at today June 24 2015 (before the change of removing the -std option in CMakeList.txt)
video at http://www.jsk.t.u-tokyo.ac.jp/~wesleyc/videos/pcl_tracker_commit_3bb8704.ogv

I've test on two machines: 
A: Ubuntu 12.04 Hydro
B: Ubuntu 14.04 Indigo

In both cases the same problem can be seen.

garaemon, 2015-06-25:
Please check it without -std option

wesleypchan, 2015-06-26:
Checked out the lasted version where the -std option has been removed, tested again but the problem is still there and the tracker stills drifts to the ground immediately.

garaemon, 2015-06-26:
please revert jsk_pcl_ros and find a commit which embed this bug

wesleypchan, 2015-06-26:
Here are the test results: 

Commits on Jan 25, 2015 desktop
1a9cf2c55d950f01070e35078e8e9cb8a3efb9e0
Drifts. Broken.

Commits on Jan 25, 2015 desktop
e4648bd419ffc7370356a9a9ac4e7907e1925e30
Drifts. Broken.

Commits on Jan 25, 2015 desktop
5c0d158aa93a2fbee7586e7fbd02dfde485ecd3a
OK

Commits on Jan 15, 2015 desktop
a9e5cd54507c440c31289c10b37374c0037b04ea
OK

Commits on Sep 10, 2014
6adf320b0b8558a5c82aaa9d2de0d7c3cb33a49b
OK

Seems like the problem appeared after this commit:
https://github.com/jsk-ros-pkg/jsk_recognition/commit/e4648bd419ffc7370356a9a9ac4e7907e1925e30

garaemon, 2015-06-26:
https://github.com/jsk-ros-pkg/jsk_recognition/commit/e4648bd419ffc7370356a9a9ac4e7907e1925e30#diff-af831319b3693155a673228b3b08e6d3R135
this line may be problem. Try without *10?

◉  ryohei

wesleypchan, 2015-06-26:
Trying without * 10 fixes the problem. However in the latest version of the code that line has been changed to 
coherence->setMaximumDistance(max_distance);

so how should we fix this?

garaemon, 2015-06-26:
Check default value of max_distance

wesleypchan, 2015-06-26:
max_distance is defaulted to 0.1 in the previous line
    pnh_->param("max_distance", max_distance, 0.1);  

and this param does not appear to be provided in the tracking.launch or the tracking_sample.launch

garaemon, 2015-06-26:
Let's re-introduce `~octree_resolution` parameter

wesleypchan, 2015-06-26:
re-introducing octree_resolution worked. 
Thanks.

I've sent a PR.

garaemon, 2015-06-26:
I misunderstood the code, I think you can decrease max_distance parameter by rosparam.

garaemon, 2015-06-26:
without changes of codes

wesleypchan, 2015-06-26:
octree_resolution is set to 0.01 in the cpp file. 

So do you mean I should leave the cpp file unchanged, and set the rosparam max_distance to 0.01                                  in my own files when I use the tracker?

garaemon, 2015-06-26:
> So do you mean I should leave the cpp file unchanged, and set the rosparam max_distance to 0.01 in my own files when I use the tracker?

Yes, or change the default value of max_distance

garaemon, 2015-06-26:
Please check where ~max_distance is used.

wesleypchan, 2015-06-26:
max_distance is not used anywhere else in particle_filter_tracking_nodelet.cpp.

So is it safe to chagne

wesleypchan, 2015-06-26:
or should it be changed to 
pnh_->param("max_distance", max_distance, octree_resolution);
?

garaemon, 2015-06-26:
Please change the line

wesleypchan, 2015-06-26:
should I change it to 
pnh_->param("max_distance", max_distance, 0.01);
or 
pnh_->param("max_distance", max_distance, octree_resolution);
?

garaemon, 2015-06-26:
> pnh_->param("max_distance", max_distance, 0.01);

looks good to me

wesleypchan, 2015-06-26:
I've change it and committed it.

closed 2015-06-26 by garaemon