jsk-ros-pkg/jsk_recognition #1009: compilation error in jsk_recognition

zweistein, 2015-07-21:
I tired to install your jsk_recognition package in ROS indigo and I have compilation errors, could you be so kind to help me? I installed all necessary pkg: jsk_common-masterm, jsk_common_msgs-master, multisense_ros, navigation-indigo-devel, SLIC-Superpixels-master, and I have opencv-3.0.0-rc1. 

The whole issue is posted on stakowerflow: http://stackoverflow.com/questions/31538142/opencv-linemod-not-cv-member

I would like to use your chessboard_detection and position estimation package, and while I try to run the   roslaunch checkerboard_detector capture.launch some nodes can not be started, like  [jsk_pcl_ros/attention_clipper], [jsk_pcl_ros/point_indices_to_mask_image] due to incomplete pkg compilation.

Thank you in advance for your help.

k-okada, 2015-07-21:
how about install ros-indigo-jsk-perception? we're not using opencv3 for
now.

zweistein, 2015-07-21:
Will it work with opencv3? I can give a try to install like that.

k-okada, 2015-07-21:
I think it is linked against opencv2

zweistein, 2015-07-21:
Could you please give me the versions of dependencies so I can setup our environment to use your jsk pgk?

garaemon, 2015-07-21:
Our environment is normal hydro environment, which includes opencv 2.4 and pcl 1.7.2

k-okada, 2015-07-22:
it is also working on indigo
```
$ aptitude show ros-indigo-jsk-perception
Package: ros-indigo-jsk-perception
New: yes
State: not installed
Version: 0.2.13-0trusty-20150708-0318-+0000
Priority: extra
Section: misc
Maintainer: Kei Okada <k-okada@jsk.t.u-tokyo.ac.jp>
Architecture: amd64
Uncompressed Size: 22.1 M
Depends: libboost-filesystem1.54.0, libboost-system1.54.0, libc6 (>= 2.14),
libconsole-bridge0.2, libgcc1 (>= 1:4.1.1), libgomp1 (>= 4.2.1),
libopencv-calib3d2.4,
         libopencv-core2.4, libopencv-features2d2.4, libopencv-flann2.4,
libopencv-highgui2.4, libopencv-imgproc2.4, libopencv-legacy2.4,
libopencv-ml2.4, libopencv-video2.4,
         libstdc++6 (>= 4.6), libeigen3-dev, ros-indigo-angles,
ros-indigo-cv-bridge, ros-indigo-driver-base,
ros-indigo-dynamic-reconfigure, ros-indigo-geometry-msgs,
         ros-indigo-image-geometry, ros-indigo-image-transport,
ros-indigo-image-view2, ros-indigo-jsk-recognition-msgs,
ros-indigo-jsk-topic-tools, ros-indigo-message-runtime,
         ros-indigo-mk, ros-indigo-nodelet, ros-indigo-pcl-ros,
ros-indigo-posedetection-msgs, ros-indigo-rosbuild, ros-indigo-roscpp,
ros-indigo-rospack, ros-indigo-rospy,
         ros-indigo-rostest, ros-indigo-sensor-msgs, ros-indigo-std-msgs,
ros-indigo-tf
Conflicts: ros-indigo-jsk-perception
Description: jsk_perception

Homepage: http://ros.org/wiki/jsk_perception
```

zweistein, 2015-07-22:
I tired to migrate your code to OpenCV3.0-rc1 and I got stuck with an opencv functione used in snake_segmentation.cpp (jsk_perception). The function si cvSnakeImage() which seams to be dropped from the 3.0 version. Do you have any idea how could this be solved?

zweistein, 2015-07-22:
I did what you suggested, installed opencv2.4.11 and apt-get install ros-indigo-jsk*. Now when I try to launch `roslaunch checkerboard_detector capture.launch` ROS throws an error: 
`ERROR: cannot launch node of type [jsk_topic_tools/static_transform_pose_stamped]: can't locate node [static_transform_pose_stamped] in package [jsk_topic_tools]` 
On the git repo I found the source of this node but in my install there is no sign of this. Do you know what am I missing?

k-okada, 2015-07-22:
we're working on opencv3 compatibility at
https://github.com/jsk-ros-pkg/jsk_recognition/pull/1011, once it is
merged, you can compile with opencv3

garaemon, 2015-07-23:
I found some executable are not installed.
https://github.com/jsk-ros-pkg/jsk_common/pull/1046

But if you compiled jsk_topic_tools from source code, you should be able to find it...

wkentaro, 2016-10-30:
Can this issue be closed?
It seems too old and the condition is changed with the current code.

closed 2019-04-17 by YutoUchimi