jsk-ros-pkg/jsk_recognition #1038: Stochastic mask image for keypoint based detection

garaemon, 2015-08-03:
Just an idea
cc: @h-kamada 

1. Define mask image in stochastic representation (defined in probability density function)
2. Extract keypoints as usual
3. sample points like RANSAC based detection but sample them according to pre-defined stochastic mask image. cf. [sample with replacement](http://www.ma.utexas.edu/users/parker/sampling/repl.htm)
4. do not compute feature vectors if probability is under threshold (= mask image)

h-kamada, 2015-08-07:
Sorry for late response.
I can't understand several things.

  > Define mask image in stochastic representation (defined in probability density function)
  
  this part is realized by attention_clipper node and roi_clipper node in jsk_pcl_ros package.
> Extract keypoints as usual

extract by surf or sift..etc

> sample points like RANSAC based detection but sample them according to pre-defined stochastic mask image. cf. sample with replacement

I can't understand this part. How to sample points? Sample points (like particle filter) only in the region of stochastic mask image? (Is it correct?)  And if so, how to resample?

> do not compute feature vectors if probability is under threshold (= mask image)

Each coordinate's probability is given by calculating probability density destribution using sampled points?

garaemon, 2015-08-07:
> this part is realized by attention_clipper node and roi_clipper node in jsk_pcl_ros package.

It's a binary representation. If object is out of ROI, we cannot recover from it.

> I can't understand this part. How to sample points? Sample points (like particle filter) only in the region of stochastic mask image? (Is it correct?) And if so, how to resample?

RANSAC sample points based on purely random sampling. We can add bias to sampling according to a priori probability.

> do not compute feature vectors if probability is under threshold (= mask image)

> Each coordinate's probability is given by calculating probability density destribution using sampled points?

These two lines do not correspond.

* feature vector computation is heavy
* we need to compute them in order to compute corresponding keypoint pairs
* Now we may optout this computation based on mask image
* But if we use probabilistic representation, we need other technique to reduce computation
  * One solution is just provide threshold to select keypoints to compute descriptor.

wkentaro, 2015-11-16:
> this part is realized by attention_clipper node and roi_clipper node in jsk_pcl_ros package.
It's a binary representation. If object is out of ROI, we cannot recover from it.

actually I don't follow all your discussions, but how do you get the first ROI?
What @garaemon means with Stochastic mask image based detection is like below?
1. making the first ROI bigger
2. split it in many patterns randomly
3. compute matches in each decomposed ROIs