jsk-ros-pkg/jsk_recognition #1045: PoseWithCovarianceStampedがうまく表示されない

h-kamada, 2015-08-11:
#1041 のPRを使ってPoseWithCovarianceStampedを表示しようとしているのですが、対角成分が大きくなると表示すべきpointcloudの領域が増えてしまい、重くなりうまく表示ができません。

garaemon, 2015-08-11:
rqt_reconfigureで点群の解像度が設定できるので変えてみてください

garaemon, 2015-08-11:
https://github.com/jsk-ros-pkg/jsk_recognition/tree/master/jsk_pcl_ros#parameters-31

h-kamada, 2015-08-11:
* 対角成分を0.3とした場合

[image]

* 対角成分を1とした場合

[image]

* 対角成分を3とした場合

[image]

h-kamada, 2015-08-11:
このような表示になるのですが、感覚的に対角成分を増加させるとむしろ領域が狭まっていきそうな気がするのですが、勘違いでしょうか？

garaemon, 2015-08-11:
https://ja.wikipedia.org/wiki/%E3%83%95%E3%82%A1%E3%82%A4%E3%83%AB:Normal_Distribution_PDF.svg

sigmaが大きい方が分布がだらだらと広くなるんじゃない?

garaemon, 2015-08-11:
`sigma^2=3`だと3sigmaまで表示するプログラムなので、`sqrt(3) * 3 = 5.19`なので+-5m程度の描画になるように作った気がします

h-kamada, 2015-08-11:
あ、勘違いですすみません。
adaptive kalmanfilterを導入してみると、対角成分が68くらいの時にこれくらいのスケールです
[image]

garaemon, 2015-08-11:
単位がmmだったりはしない?

h-kamada, 2015-08-11:
単位mmでした。
mに直してしまうと今度は小さくなりすぎてしまうので、自分のコードのどこかがまだおかしいかもしれません、、

closed 2015-08-11 by h-kamada

garaemon, 2015-08-11:
逆にresolutionを小さくしてみると可視化できるかも。これもsigmaの大きさに依存させて決定した方がよさそうだね。もしくはサンプリングの個数を固定にした方がいいか

garaemon, 2015-08-11:
分散の値は正しそうなのかな？
やりたいこととしては、実はKFなんか使わなくても単純に直近のデータ列から分散を計算してみればいいだけな気はしてきた

2015年8月11日火曜日、Ryohei Ueda<garaemon@gmail.com>さんは書きました:

>
> 逆にresolutionを小さくしてみると可視化できるかも。これもsigmaの大きさに依存させて決定した方がよさそうだね。もしくはサンプリングの個数を固定にした方がいいか
>
> 2015年8月11日火曜日、h-kamada<notifications@github.com
>
>> Closed #1045 <https://github.com/jsk-ros-pkg/jsk_recognition/issues/1045>
>>
>>
>
>
> ✉︎ from iPhone

k-okada, 2015-08-14:
KFかParticleFilterかは別にして現在の対象の市姿勢を推定するのにはやはりトラッキングは必要かと．