jsk-ros-pkg/jsk_recognition #1094: catkin_build jsk_pcl_rosが通らない

mmurooka, 2015-08-25:
Ubuntu12.04 hydroで #1081以降コンパイルが通らないです．
Ubuntu14.04 indigoでは普通に通っています．
```
==> Expanding alias 'bt' from 'catkin bt' to 'catkin b --this'
==> Expanding alias 'b' from 'catkin b --this' to 'catkin build --this'
--------------------------------------------------------------------------------
Profile:                     default
Extending:          [cached] /home/jskuser/ros/hydro_parent/devel:/opt/ros/hydro
Workspace:                   /home/jskuser/ros/hydro
Source Space:       [exists] /home/jskuser/ros/hydro/src
Build Space:        [exists] /home/jskuser/ros/hydro/build
Devel Space:        [exists] /home/jskuser/ros/hydro/devel
Install Space:     [missing] /home/jskuser/ros/hydro/install
DESTDIR:                     None
--------------------------------------------------------------------------------
Isolate Develspaces:         False
Install Packages:            False
Isolate Installs:            False
--------------------------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
--------------------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
--------------------------------------------------------------------------------
Workspace configuration appears valid.
-------------------------------------------------------------------------------- 
Found '331' packages in 0.4 seconds. 
Starting ==> dynamic_tf_publisher                                               
Starting ==> image_view                                                         
Starting ==> jsk_footstep_msgs                                                  
Starting ==> jsk_recognition_msgs                                               
Starting ==> pcl_conversions                                                    
Starting ==> sklearn                                                            
Finished <== pcl_conversions      [ 0.4 seconds ]                               
Finished <== sklearn              [ 0.4 seconds ]                               
Starting ==> pcl_ros                                                            
Finished <== dynamic_tf_publisher [ 0.5 seconds ]                               
Finished <== image_view           [ 0.5 seconds ]                               
Starting ==> jsk_topic_tools                                                    
Finished <== jsk_recognition_msgs [ 0.7 seconds ]                               
Finished <== jsk_footstep_msgs    [ 0.7 seconds ]                               
Finished <== pcl_ros              [ 0.8 seconds ]                               
Finished <== jsk_topic_tools      [ 0.8 seconds ]                               
Starting ==> image_view2                                                        
Finished <== image_view2          [ 0.6 seconds ]                               
Starting ==> jsk_pcl_ros                                                        
                                                                                
[jsk_pcl_ros] ==> '/home/jskuser/ros/hydro/build/jsk_pcl_ros/build_env.sh /usr/bin/make --jobserver-fds=3,5 -j' in '/home/jskuser/ros/hydro/build/jsk_pcl_ros'
[  0%] [  0%] [  0%] [  0%] Built target geometry_msgs_generate_messages_cpp
Built target sensor_msgs_generate_messages_cpp
Built target jsk_recognition_msgs_generate_messages_cpp
Built target pcl_msgs_generate_messages_cpp
[  0%] [  0%] [  0%] [  0%] Built target jsk_footstep_msgs_generate_messages_cpp
Built target add_color_from_image
Built target attention_clipper
Built target border_estimator
[  0%] Built target capture_stereo_synchronizer
[ 15%] Built target jsk_pcl_ros_gencfg
[ 16%] [ 16%] [ 16%] [ 17%] Built target colorize_distance_from_plane
Built target add_point_indices
Built target colorize_random_foreset_result2
Built target bilateral_filter
[ 17%] [ 17%] Built target color_histogram_matcher
Built target bounding_box_filter
[ 17%] [ 18%] Built target colorize_random_forest_result
Built target boundingbox_occlusion_rejector
[ 18%] [ 19%] [ 19%] [ 19%] [ 19%] [ 20%] [ 20%] [ 21%] Built target depth_calibration
[... 117 lines cut ...]
/usr/include/pcl-1.7/pcl/recognition/impl/linemod/line_rgbd.hpp:  bool pcl::LineRGBD<PointXYZT, PointRGBT>::loadTemplates(const string&, size_t) :
/usr/include/pcl-1.7/pcl/recognition/impl/linemod/line_rgbd.hpp:110:89: : transform(std::basic_string<char>::iterator&, std::basic_string<char>::iterator&, std::basic_string<char>::iterator&, <unresolved overloaded function type>) 
/usr/include/pcl-1.7/pcl/recognition/impl/linemod/line_rgbd.hpp:110:89: : :
/usr/include/c++/4.6/bits/stl_algo.h:4871:5: : template<class _IIter, class _OIter, class _UnaryOperation> _OIter std::transform(_IIter, _IIter, _OIter, _UnaryOperation)
/usr/include/c++/4.6/bits/stl_algo.h:4907:5: : template<class _IIter1, class _IIter2, class _OIter, class _BinaryOperation> _OIter std::transform(_IIter1, _IIter1, _IIter2, _OIter, _BinaryOperation)
make[2]: *** [CMakeFiles/build_check.dir/build_check.cpp.o]  1
make[1]: *** [CMakeFiles/build_check.dir/all]  2
make: *** [all]  2
[jsk_pcl_ros] <== '/home/jskuser/ros/hydro/build/jsk_pcl_ros/build_env.sh /usr/bin/make --jobserver-fds=3,5 -j' failed with return code '2'
                                                                                 
Failed   <== jsk_pcl_ros          [ 16.9 seconds ]                              
[build] There were '1' errors:                                                  

Failed to build package 'jsk_pcl_ros' because the following command:

# Command to reproduce:
cd /home/jskuser/ros/hydro/build/jsk_pcl_ros && /home/jskuser/ros/hydro/build/jsk_pcl_ros/build_env.sh /usr/bin/make --jobserver-fds=3,5 -j; cd -

# Path to log:
cat /home/jskuser/ros/hydro/build/build_logs/jsk_pcl_ros.log

Exited with return code: 2 
                                                                                
[build] Runtime: 20.5 seconds 
```

mmurooka, 2015-08-25:
すいません @YuOhara です

garaemon, 2015-08-25:
https://github.com/jsk-ros-pkg/jsk_recognition/pull/1093
が必要かも

YuOhara, 2015-08-25:
すいません１４．０４で通ったは間違いで，
pclをsourceから入れたPCでmakeが通ってます

k-okada, 2015-08-25:
travis は通っているように見えるが．

garaemon, 2015-08-25:
> travis は通っているように見えるが．

はい、ただしいです。というわけでコンパイルは通るはず、なんですが。

https://github.com/jsk-ros-pkg/jsk_recognition/pull/1093
がないと実行はできない予感はしています。

garaemon, 2015-08-25:
これは結局どれがエラーなんだろうか. `std::transform`?

garaemon, 2015-08-25:
みてみましたが、やはりbuild_check.cpp.inがあやしすぎるか。
とりあえず無視していても大丈夫なので進めておいてください > @YuOhara 

build_check.cpp.in -> build_check.cppでやっていることは
1. include/jsk_pcl_rosのすべてのヘッダをincludeする
2. jsk_pcl_nodelets.xmlに書いてあるクラスのインスタンスをすべて作る

garaemon, 2015-08-25:
build/jsk_pcl_ros/build_check.cppを貼ってみてください。
僕の手元では
```c++
// -*- mode: c++ -*-
/*********************************************************************
 * Software License Agreement (BSD License)
 *
 *  Copyright (c) 2015, JSK Lab
 *  All rights reserved.
 *
 *  Redistribution and use in source and binary forms, with or without
 *  modification, are permitted provided that the following conditions
 *  are met:
 *
 *   * Redistributions of source code must retain the above copyright
 *     notice, this list of conditions and the following disclaimer.
 *   * Redistributions in binary form must reproduce the above
 *     copyright notice, this list of conditions and the following
 *     disclaimer in the documentation and/o2r other materials provided
 *     with the distribution.
 *   * Neither the name of the JSK Lab nor the names of its
 *     contributors may be used to endorse or promote products derived
 *     from this software without specific prior written permission.
 *
 *  THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS
 *  "AS IS" AND ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT
 *  LIMITED TO, THE IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS
 *  FOR A PARTICULAR PURPOSE ARE DISCLAIMED. IN NO EVENT SHALL THE
 *  COPYRIGHT OWNER OR CONTRIBUTORS BE LIABLE FOR ANY DIRECT, INDIRECT,
 *  INCIDENTAL, SPECIAL, EXEMPLARY, OR CONSEQUENTIAL DAMAGES (INCLUDING,
 *  BUT NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR SERVICES;
 *  LOSS OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER
 *  CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN CONTRACT, STRICT
 *  LIABILITY, OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE) ARISING IN
 *  ANY WAY OUT OF THE USE OF THIS SOFTWARE, EVEN IF ADVISED OF THE
 *  POSSIBILITY OF SUCH DAMAGE.
 *********************************************************************/
#define BOOST_PARAMETER_MAX_ARITY 7

#include "/home/lueda/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/interactive_cuboid_likelihood.h"
#include "/home/lueda/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/random_util.h"
#include "/home/lueda/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/roi_clipper.h"
#include "/home/lueda/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/grid_line.h"
#include "/home/lueda/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/voxel_grid_downsample_manager.h"
#include "/home/lueda/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/linemod.h"
#include "/home/lueda/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/grid_sampler.h"
#include "/home/lueda/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/normal_estimation_omp.h"
#include "/home/lueda/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/parallel_edge_finder.h"
#include "/home/lueda/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/pcd_reader_with_pose.h"
#include "/home/lueda/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/pointcloud_database_server.h"
#include "/home/lueda/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/plane_rejector.h"
#include "/home/lueda/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/joint_state_static_filter.h"
#include "/home/lueda/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/snapit.h"
#include "/home/lueda/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/colorize_distance_from_plane.h"
#include "/home/lueda/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/pointcloud_to_cluster_point_indices.h"
#include "/home/lueda/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/handle_estimator.h"
#include "/home/lueda/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/mask_image_filter.h"
#include "/home/lueda/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/spherical_pointcloud_simulator.h"
#include "/home/lueda/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/environment_plane_modeling.h"
#include "/home/lueda/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/keypoints_publisher.h"
#include "/home/lueda/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/plane_supported_cuboid_estimator.h"
#include "/home/lueda/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/hinted_handle_estimator.h"
#include "/home/lueda/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/incremental_model_registration.h"
#include "/home/lueda/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/multi_plane_sac_segmentation.h"
#include "/home/lueda/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/colorize_random_points_RF.h"
#include "/home/lueda/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/ros_collaborative_particle_filter.h"
#include "/home/lueda/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/depth_image_creator.h"
#include "/home/lueda/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/normal_estimation_integral_image.h"
[... cut ...]
jsk_pcl_ros::OctreeChangePublisher instance_89;
jsk_pcl_ros::TfTransformCloud instance_90;
jsk_pcl_ros::ColorizeMapRandomForest instance_91;
jsk_pcl_ros::NormalEstimationOMP instance_92;
jsk_pcl_ros::HeightmapConverter instance_93;
jsk_pcl_ros::HeightmapToPointCloud instance_94;
jsk_pcl_ros::HeightmapMorphologicalFiltering instance_95;
jsk_pcl_ros::HeightmapTimeAccumulation instance_96;
jsk_pcl_ros::PCDReaderWithPose instance_97;
jsk_pcl_ros::PointCloudMoveitFilter instance_98;

  std::cout << "Hello World" << std::endl;
  return 0;
}
```

garaemon, 2015-08-25:
ちなみに、build_checkは実行できません。
```
$ rosrun jsk_pcl_ros build_check
[FATAL] [1440542827.657635126]: You must call ros::init() before creating the first NodeHandle
Couldn't find an AF_INET address for []
Couldn't find an AF_INET address for []
[ERROR] [1440542827.678959328]: [registerPublisher] Failed to contact master at [:0].  Retrying...
Couldn't find an AF_INET address for []
```

YuOhara, 2015-08-25:
```
[http://hrp2016v:10016][192.168.96.103] jskuser@fuji:~/ros/hydro/build/jsk_pcl_ros$ cat build_check.cpp 
// -*- mode: c++ -*-
/*********************************************************************
 * Software License Agreement (BSD License)
 *
 *  Copyright (c) 2015, JSK Lab
 *  All rights reserved.
 *
 *  Redistribution and use in source and binary forms, with or without
 *  modification, are permitted provided that the following conditions
 *  are met:
 *
 *   * Redistributions of source code must retain the above copyright
 *     notice, this list of conditions and the following disclaimer.
 *   * Redistributions in binary form must reproduce the above
 *     copyright notice, this list of conditions and the following
 *     disclaimer in the documentation and/o2r other materials provided
 *     with the distribution.
 *   * Neither the name of the JSK Lab nor the names of its
 *     contributors may be used to endorse or promote products derived
 *     from this software without specific prior written permission.
 *
 *  THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS
 *  "AS IS" AND ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT
 *  LIMITED TO, THE IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS
 *  FOR A PARTICULAR PURPOSE ARE DISCLAIMED. IN NO EVENT SHALL THE
 *  COPYRIGHT OWNER OR CONTRIBUTORS BE LIABLE FOR ANY DIRECT, INDIRECT,
 *  INCIDENTAL, SPECIAL, EXEMPLARY, OR CONSEQUENTIAL DAMAGES (INCLUDING,
 *  BUT NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR SERVICES;
 *  LOSS OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER
 *  CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN CONTRACT, STRICT
 *  LIABILITY, OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE) ARISING IN
 *  ANY WAY OUT OF THE USE OF THIS SOFTWARE, EVEN IF ADVISED OF THE
 *  POSSIBILITY OF SUCH DAMAGE.
 *********************************************************************/
#define BOOST_PARAMETER_MAX_ARITY 7

#include "/home/jskuser/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/interactive_cuboid_likelihood.h"
#include "/home/jskuser/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/polygon_appender.h"
#include "/home/jskuser/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/region_growing_segmentation.h"
#include "/home/jskuser/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/geo_util.h"
#include "/home/jskuser/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/organized_multi_plane_segmentation.h"
#include "/home/jskuser/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/voxel_grid_downsample_decoder.h"
#include "/home/jskuser/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/color_filter.h"
#include "/home/jskuser/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/attention_clipper.h"
#include "/home/jskuser/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/joint_state_static_filter.h"
#include "/home/jskuser/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/pointcloud_database_server.h"
#include "/home/jskuser/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/geometric_consistency_grouping.h"
#include "/home/jskuser/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/fisheye_sphere_publisher.h"
#include "/home/jskuser/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/organized_pointcloud_to_point_indices.h"
#include "/home/jskuser/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/depth_calibration.h"
#include "/home/jskuser/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/organize_pointcloud.h"
#include "/home/jskuser/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/normal_estimation_integral_image.h"
#include "/home/jskuser/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/centroid_publisher.h"
#include "/home/jskuser/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/tf_listener_singleton.h"
#include "/home/jskuser/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/planar_pointcloud_simulator.h"
#include "/home/jskuser/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/plane_supported_cuboid_estimator.h"
#include "/home/jskuser/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/feature_registration.h"
#include "/home/jskuser/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/colorize_height_2d_mapping.h"
#include "/home/jskuser/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/grid_index.h"
#include "/home/jskuser/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/incremental_model_registration.h"
#include "/home/jskuser/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/grid_sampler.h"
#include "/home/jskuser/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/tf_transform_cloud.h"
#include "/home/jskuser/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/grid_map.h"
#include "/home/jskuser/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/polygon_flipper.h"
[... cut ...]
jsk_pcl_ros::OctreeChangePublisher instance_89;
jsk_pcl_ros::TfTransformCloud instance_90;
jsk_pcl_ros::ColorizeMapRandomForest instance_91;
jsk_pcl_ros::NormalEstimationOMP instance_92;
jsk_pcl_ros::HeightmapConverter instance_93;
jsk_pcl_ros::HeightmapToPointCloud instance_94;
jsk_pcl_ros::HeightmapMorphologicalFiltering instance_95;
jsk_pcl_ros::HeightmapTimeAccumulation instance_96;
jsk_pcl_ros::PCDReaderWithPose instance_97;
jsk_pcl_ros::PointCloudMoveitFilter instance_98;

  std::cout << "Hello World" << std::endl;
  return 0;
}
```

garaemon, 2015-08-25:
* include guardがかぶっているものがある。
* なぜかlinemodをincludeすると動かない環境がある

garaemon, 2015-08-25:
* headerのインクルード順が環境依存

garaemon, 2015-08-25:
headerをアルファベット順にしてincludeしてみたところ、@YuOhara 環境と同じエラーが出ました。

garaemon, 2015-08-25:
headerに`using namespace`があるのが原因のよう

garaemon, 2015-08-25:
長い戦いでしたが、解決しました。

build_checkによって、実装されていないメソッドがあった場合には

garaemon, 2015-08-25:
長い戦いでしたが、解決しました。

build_checkによって、実装されていないメソッドがあった場合には以下のようなエラーが出ます.

```
Linking CXX executable /home/garaemon/ros/hydro/devel/lib/jsk_pcl_ros/build_check
/home/garaemon/ros/hydro/devel/lib/libjsk_pcl_ros.so: undefined reference to `jsk_pcl_ros::BilateralFilter::hoge()'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/garaemon/ros/hydro/devel/lib/jsk_pcl_ros/build_check] Error 1
make[1]: *** [CMakeFiles/build_check.dir/all] Error 2
make: *** [all] Error 2
```

closed 2015-08-25 by garaemon