jsk-ros-pkg/jsk_recognition #1181: [checkerboard_detector]  different pose b/w PoingCloud and Detected Pose

wkentaro, 2015-09-26:
[image]
[image]

There is difference between PoseStamped (which is published by checkerboard_detector) and PointCloud when visualize.

Am I missing something?

I'm using checkerboard
* board_type: chess
* grid_size: 6 x 5
* rect_size: 0.03 x 0.03

garaemon, 2015-09-26:
Is camera intrinsic parameter calibrated?

wkentaro, 2015-09-26:
Not so much, maybe that is the cause.

garaemon, 2015-09-26:
check debug image and confirm checkerboard is detected correctly first.
If it is correct, you are using size of checkerboard or intrinsic camera parameter.

wkentaro, 2015-09-26:
[image]

checkerboard seems detected correctly.
I will do calibration of kinect2.

garaemon, 2015-09-26:
Looks good, intrinsic parameter and checkerboard size may be wrong.

wkentaro, 2015-09-26:
is checkerboard size also wrong?
that size is written on the board.

garaemon, 2015-09-26:
> that size is written on the board.

measure it and 1mm difference of checkerboard size will result in 1cm error of estimated checkeboard pose.

wkentaro, 2015-09-26:
I see, I will.

wkentaro, 2015-09-27:
The checkerboard was printed with reduction of scale, so the actual rectsize was 0.0278 not 0.03.
After changing the specified size, the distance looks correct but pointcloud does not.
https://drive.google.com/open?id=0B9P1L--7Wd2vTWc2dnJfcnlUTzQ

And I calibrated kinect2 on baxter head, and it now looks better.
https://drive.google.com/open?id=0B9P1L--7Wd2vZUxGSXFiaHppSFE

Thank you.

closed 2015-09-27 by wkentaro