jsk-ros-pkg/jsk_recognition #1187: [jsk_pcl_ros] ClusterPointIndicesDecomposer does not publish clouds

wkentaro, 2015-10-02:
[image]

I used the launch file in jsk_pcl_ros/launch.
but actually there is no clouds.

The launch file I used was below (edited some lines):
https://github.com/wkentaro/jsk_recognition/blob/cluster-pi-trial/jsk_pcl_ros/launch/euclidean_segmentation.launch

garaemon, 2015-10-02:
try ClusterPointIndicesDecomposer instead of ClusterPointIndicesDecomposerZAxis

wkentaro, 2015-10-02:
I also tried it, but seems not publish clouds.
and It seems that the feature is not used recently.

garaemon, 2015-10-02:
check timestamp of inputs of cluster decomposer

garaemon, 2015-10-02:
/pcl_nodelet/voxelgrid/output and /pcl_nodelet/clustering/output

wkentaro, 2015-10-02:
I found the reason, ClusterPointIndices does not prepare publications before subscribing cluster_point_indices because number of clusters is unknown.
That is the reason why there is no clouds topics in rostopic list because ConnectedBasedNodelet does not subscribe topics until subscribed.

wkentaro, 2015-10-02:
I can see clouds after subscribing one of the topics.

garaemon, 2015-10-02:
I see, tricky. Please use debug_output (http://jsk-recognition.readthedocs.org/en/latest/jsk_pcl_ros/nodes/cluster_point_indices_decomposer.html#publishing-topics)

I don't recomment to use `~output%02d` topics

wkentaro, 2015-10-02:
BTW, are there simple command line tool to apply point indices to point
cloud?

I'd like to use it when viewing on rviz.

garaemon, 2015-10-02:
pcl_msgs/PointIndices? You can use pcl/ExtractIndices

wkentaro, 2015-10-02:
Not for writing code but command line tool.
Like:

rosrun jsk_pcl_ros pointindices_to_pointcloud ~input:=/kinect2/qhd/points
~target:=/pointcloudindices

closed 2015-10-03 by garaemon