jsk-ros-pkg/jsk_recognition #1188: [jsk_pcl_ros] AttentionClipper dies unexpectedly when subscribe its topic

wkentaro, 2015-10-02:
[image]

There are no error..

garaemon, 2015-10-02:
Your tf looks illegal

wkentaro, 2015-10-02:
I think I transform the tf correctly.

https://github.com/wkentaro/jsk_apc/blob/filter-pod-outer/jsk_2015_apc_common/launch/in_kiva_pod_clipper.launch

And without any change of code, it is now fixed.. I don't know the reason..

garaemon, 2015-10-02:
Do you use `rosbag play -l`?

wkentaro, 2015-10-02:
At present?
No I'm using the realtime data in lab.

wkentaro, 2015-10-02:
In fact, I doubt the change of jsk_topic_tools/ConnectedBasedNodelet. because nodelet_manager dies in this issue.

wkentaro, 2015-10-02:
And I'm doing `git checkout` and `catkin bt` recursively...

garaemon, 2015-10-02:
hmm, please test with gdb.

wkentaro, 2015-10-02:
> hmm, please test with gdb.

Actually, I don't have the error at present. I don't know why as said above..
I will do the test when I encountered same problem.

wkentaro, 2015-10-03:
I think I found the reason.
AttentionClipper dies when negative value is specified as initial_pos_(x|y|z).
I don't know where is the cause of this at present.

wkentaro, 2015-10-03:
these lines may be the cause.
https://github.com/jsk-ros-pkg/jsk_common/blob/master/jsk_topic_tools/src/rosparam_utils.cpp#L48-L55

```
throw std::runtime_error("the value cannot be converted into double")
```

does not appear the screen of roslaunch.

garaemon, 2015-10-03:
`XmlRpc::XmlRpcValue::TypeDouble` cannot catch negative values?

wkentaro, 2015-10-03:
Yeah I doubt that.

wkentaro, 2015-10-03:
I don't know how to do print debug in that file (rosparam_utils.cpp) because ROS_WARN does not appear..

garaemon, 2015-10-03:
> I don't know how to do print debug in that file (rosparam_utils.cpp) because ROS_WARN does not appear..

really? it's strange but please use `std::cerr` as quick hack.

wkentaro, 2015-10-03:
I confirmed that  `XmlRpc::XmlRpcValue::TypeDouble` does not catch negative value.

**no negative**
[image]

**with negative**
[image]

You can focus on `jsk_topic_tools 8` message.

wkentaro, 2015-10-03:
Actually I'm not familiar to C++ libraries.
Do you know some alternatives or how to fix this?

garaemon, 2015-10-03:
can you show me `rosparam get` result?

wkentaro, 2015-10-03:
dimension_x: 0.25
dimension_y: 0.22
dimension_z: 0.22
frame_id: /kiva_pod_base
initial_pos: [-.21, 0.29, 0.36]
initial_rot: [0, 0, 0]

​

garaemon, 2015-10-03:
> initial_pos: [-.21, 0.29, 0.36]

-0.21? Do you have 0 in your launch file?

wkentaro, 2015-10-03:
oh, -.21 does not work for Xml:Rpc.
It is fixed.
Sorry, I thought this can be used (becausee python allow)..

wkentaro, 2015-10-03:
Xml:Rpc allows .29 but not -.29.

garaemon, 2015-10-03:
regular expression should be `^[+-]?(\d*\.\d+|\d+\.?\d*)([eE][+-]?\d+|)` but libxml is probably wrongly implemented.

wkentaro, 2015-10-03:
I see.
Should we warn about it in this?
https://github.com/jsk-ros-pkg/jsk_common/pull/1157

garaemon, 2015-10-03:
> Should we warn about it in this?

Yes, it should help users a lot

garaemon, 2015-10-03:
> jsk-ros-pkg/jsk_common#1157

I've commented about improvements.

wkentaro, 2015-10-04:
Closed via https://github.com/jsk-ros-pkg/jsk_common/pull/1157

closed 2015-10-04 by wkentaro