jsk-ros-pkg/jsk_recognition #1215: add krish segmentation/tracking modules

k-okada, 2015-10-13:
@iKrishneel 
[ ] add segmetation module
[ ] add tracking modulule
[ ] add building object objet

k-okada, 2015-10-13:
@wkentaro  will provide `rosbag` data

wkentaro, 2015-10-13:
Ok, please tell me what kind of log data you need.

iKrishneel, 2015-10-13:
@wkentaro I will need the dataset of APC.

wkentaro, 2015-10-13:
The shelf model is available here:
http://amazonpickingchallenge.org/gazebo_pod.shtml?

The object models are available here:
http://rll.berkeley.edu/amazon_picking_challenge/
(you can just run python script here: http://rll.berkeley.edu/amazon_picking_challenge/scripts/downloader.py)

k-okada, 2015-10-13:
we need `rosbag` data for APC tasks,
I think you have rviz screen shot video for RSJ and asked to re-take them
which the robot actually grab object from the bin, if you have bag data
during that procedure. That exactly what we need. If you do not have them,
it is nice chance to take them.

wkentaro, 2015-10-13:
Actually I don't have bagfiles which contains pointcloud.
So I will take that rosbag. When is the deadline? @iKrishneel

iKrishneel, 2015-10-13:
This is to test some on going research algorithm. You can take the data at your suitable time.

wkentaro, 2015-10-13:
Okay, thanks.

wkentaro, 2015-11-06:
[image]

for log files
```
$ jsk_data get 2015-11-06-17-10-59_apc_pick_verify_return.tgz
$ tar zxvf 2015-11-06-17-10-59_apc_pick_verify_return.tgz 20151106-171059_apc_pick_verify_return/
$ cat 20151106-171059_apc_pick_verify_return/README.md
```

wkentaro, 2015-11-06:
VIDEO: https://drive.google.com/open?id=0B9P1L--7Wd2vTjJzRk55VUNMNjA

wkentaro, 2015-11-06:
Please wait for a while for log uploading and video encoding.

iKrishneel, 2015-11-09:
@wkentaro thanks I will look into it. :+1: