jsk-ros-pkg/jsk_recognition #1239: split location of scripts and nodes

wkentaro, 2015-10-15:
I think scripts and nodes should be located in different directories.

I mean as 
**scripts** for example:

*  https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/jsk_perception/scripts/download_trained_data.sh
* https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/jsk_perception/scripts/check_cascadeclassifier.py

and 
**nodes** for example:

* https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/jsk_perception/scripts/image_publisher.py
* https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/jsk_perception/scripts/label_image_decomposer.py

do you have any idea? @garaemon 

One option is to move **scripts** to `bin` dir under each package root dir.

garaemon, 2015-10-15:
node_scripts directory?

garaemon, 2015-10-15:
change  as you like. :relaxed:

wkentaro, 2015-10-15:
thanks. :sunglasses:

PR #1242 (2015-10-15): [jsk_perception] Move ros node scripts/ -> node_scripts/
Closes #1239
commit 8cbad4bc (2015-10-15): [jsk_perception] Move ros node scripts/ -> node_scripts/

Closes #1239
closed 2015-10-15 by garaemon