jsk-ros-pkg/jsk_recognition #1261: [jsk_pcl_ros] Are there MergePointClouds?

wkentaro, 2015-10-19:
I have two kinect2's and I have two point cloud topics.
Are there way to merge these topics?

wkentaro, 2015-10-19:
cc: @YuOhara

garaemon, 2015-10-19:
https://github.com/ros-perception/perception_pcl/blob/jade-devel/pcl_ros/pcl_nodelets.xml#L127

garaemon, 2015-10-19:
PointCloudConcatenateDataSynchronizer and please confirm your pcl_ros includes [this PR](https://github.com/ros-perception/perception_pcl/pull/85)

wkentaro, 2015-10-19:
Thanks. :smile:

closed 2015-10-19 by wkentaro

garaemon, 2015-10-19:
When you want to merge two pointclouds, be careful about timestamp.

wkentaro, 2015-10-19:
I'm going to merge two point clouds at the first node of the pass through nodes.

wkentaro, 2015-10-19:
I have a lots of below error message:
```
[ERROR] [1445254970.000790841]: Lookup would require extrapolation into the past.  Requested time 1445254959.134954699 but the earliest data is at time 1445254960.172250049, when looking up transform from frame [kinect2_torso_rgb_optical_frame] to frame [kinect2_head_rgb_optical_frame]
[ERROR] [1445254976.463017667]: Lookup would require extrapolation into the past.  Requested time 1445254965.792854681 but the earliest data is at time 1445254966.578855736, when looking up transform from frame [kinect2_torso_rgb_optical_frame] to frame [kinect2_head_rgb_optical_frame]
```
do you know about this?

garaemon, 2015-10-19:
https://github.com/garaemon/perception_pcl/blob/4ec3d5fd9d2c724eb60643ee6ce3b096bb765f6e/pcl_ros/src/pcl_ros/io/concatenate_data.cpp#L205

transformation between kinect2_torso_rgb_optical_frame and kinect2_head_rgb_optical_frame can be solved

wkentaro, 2015-10-19:
once in three times it succeeds. I think..

garaemon, 2015-10-19:
need to call `waitForTransform` in PointCloudConcatenateDataSynchronizer before solving tf.
`tf::MessageFilter` is better https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/jsk_pcl_ros/src/tf_transform_cloud_nodelet.cpp#L102

wkentaro, 2015-10-19:
I sent PR: https://github.com/ros-perception/perception_pcl/pull/109