jsk-ros-pkg/jsk_recognition #1262: [question] Is that possible to merge two images from two Kinects?

wkentaro, 2015-10-19:
as #1261 to point clouds.

garaemon, 2015-10-19:
tile_image?

meging image is difficult because simply merged image does not satisfy
camera model.
if you put the third virtual camera at the middle of them, you can create
image from pointcloud not from images.
i don't know virtual_camera_publisher supprts it.

wkentaro, 2015-10-19:
Thank you for your information.
I will search on google for that.

garaemon, 2015-10-19:
If you need to implement it, [this](https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/jsk_pcl_ros/src/add_color_from_image_nodelet.cpp#L107) is a good sample.

But if you want to implement it in realtime, you may need to use GPU to render pointcloud on image.
keywords: depth buffer, z buffer, offscreen rendering, OpenGL

garaemon, 2015-10-19:
references
* http://research.microsoft.com/en-us/um/redmond/projects/hyperlapse/
* http://repository.dl.itc.u-tokyo.ac.jp/dspace/bitstream/2261/43684/1/48037413.pdf

garaemon, 2015-10-19:
* [DepthImageCreator](https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/jsk_pcl_ros/src/depth_image_creator_nodelet.cpp) can generate depth image from point cloud.

wkentaro, 2015-10-20:
Thanks.
I have tried `convert_pointcloud_to_image` and it does not work on my issue.
http://wiki.ros.org/pcl_ros

I'd like to try implement these feature, but I don't have time for it now.
Maybe I will try in the future.

closed 2015-10-20 by wkentaro