jsk-ros-pkg/jsk_recognition #1269: [bug] [jsk_recognition_utils] Opposite transformation in lookupTransformWithDuration()

wkentaro, 2015-10-20:
current
```
listener->lookupTransform(
  from_frame, to_frame, stamp, transform);
```

**correctly**

```
listener->lookupTransform(
  /*target_frame=*/to_frame, /*source_frame=*/from_frame, /*time=*/stamp, /*transform=*/transform);
```

http://docs.ros.org/jade/api/tf/html/c++/classtf_1_1Transformer.html#ac01a9f8709a828c427f1a5faa0ced42b

wkentaro, 2015-10-20:
see https://github.com/jsk-ros-pkg/jsk_recognition/pull/1271#issuecomment-149628708 for *before* and *after*.

garaemon, 2015-10-21:
https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/jsk_recognition_utils/include/jsk_recognition_utils/tf_listener_singleton.h#L58

```
  tf::StampedTransform lookupTransformWithDuration(
    tf::TransformListener* listener,
    const std::string& to_frame,
    const std::string& from_frame,
    const ros::Time& stamp,
    ros::Duration duration);
```

garaemon, 2015-10-21:
current implementation is correct and lookupTransformWithDuration is already used in some nodelets.
If you want to swap to_fram and from_frame, 
1. rename argument
2. fix cpp implementation
3. fix all lookupTransformWithDuration used

wkentaro, 2015-10-21:
> current implementation is correct

I don't think so. See:
https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/jsk_recognition_utils/src/tf_listener_singleton.cpp#L69-L70

wkentaro, 2015-10-21:
>  lookupTransformWithDuration is already used in some nodelets.

Yeah, so maybe we need test all nodes. :sob:

garaemon, 2015-10-21:
Hmm, I'm not confident about this issue...

garaemon, 2015-10-21:
Start with writing unittest in jsk_recognition_utils.

wkentaro, 2015-10-21:
It's simple, isn't it? from_frame and to_frame should be passed to opposite pos.
https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/jsk_recognition_utils/src/tf_listener_singleton.cpp#L69-L70

because from_frame is source_frame, and to_frame is target_frame.

garaemon, 2015-10-22:
```
$ rosrun tf tf_echo
Usage: tf_echo source_frame target_frame

This will echo the transform from the coordinate frame of the source_frame
to the coordinate frame of the target_frame. 
Note: This is the transform to get data from target_frame into the source_frame.
```

But it uses in opposite manner.
https://github.com/ros/geometry/blob/indigo-devel/tf/src/tf_echo.cpp#L107
```
        echoListener.tf.lookupTransform(source_frameid, target_frameid, ros::Time(), echo_transform);
```

But tf document says
```
void Transformer::lookupTransform	(	const std::string & 	target_frame,
const std::string & 	source_frame,
const ros::Time & 	time,
StampedTransform & 	transform 
)		const
```

is tf document wrong?

garaemon, 2015-10-22:
https://github.com/ros/geometry/issues/108
reported

garaemon, 2015-10-22:
I think order of arguments of `tf::Transformer::lookupTransformation` will not change because it will affect so many codes.

we need to write unittest to be confident about tf.
1. run `static_tf_transform_publisher` with known offset
2. verify the offset by `TfLookupTransformWithDuration`

wkentaro, 2015-10-22:
Ok, TfLookupTransformWithDuration is just a shallow wrapper of
tf::lookupTransform, so I will write a test for it and will change
TfLookupTransformWithDuration according to the result.

wkentaro, 2015-10-22:
I sent PR to test tf::Transform::lookupTrasform: https://github.com/jsk-ros-pkg/jsk_recognition/pull/1274

wkentaro, 2015-12-03:
closes via https://github.com/jsk-ros-pkg/jsk_recognition/pull/1324

closed 2015-12-03 by wkentaro