jsk-ros-pkg/jsk_recognition #1300: [bug] [jsk_pcl_ros] Nodelet manager down without no error message at 438c63a

wkentaro, 2015-11-04:
at https://github.com/jsk-ros-pkg/jsk_recognition/commit/6a1bcef887ee3d27765b5999695da232e2b654b2 it does not down.

The manager handles nodelets in jsk_pcl_ros. And when I checkout to https://github.com/jsk-ros-pkg/jsk_recognition/commit/6a1bcef887ee3d27765b5999695da232e2b654b2 and after recompiling `jsk_recognition_utils` and `jsk_pcl_ros`, this problem is fixed.

So, the changes between

a. https://github.com/jsk-ros-pkg/jsk_recognition/commit/6a1bcef887ee3d27765b5999695da232e2b654b2
b. 438c63a801999827f265932d7675c01e5ab07603

seems to cause this problem.

wkentaro, 2015-11-04:
the result of ldd libjsk_pcl_ros*.so

* [works](https://gist.github.com/wkentaro/8cee95840225af10fa57)
* [does not work](https://gist.github.com/wkentaro/6af99e592221cc1f7142)

wkentaro, 2015-11-04:
commenting out [these](https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/jsk_pcl_ros/CMakeLists.txt#L37-L41) fixed this issue.

mmurooka, 2015-11-04:
In my commits those libraries are additionally linked to libjsk_pcl_ros.so .
Maybe one of those libraries is the cause of the problem.

The difference of ldd result is as follows.
```
cut work.txt -f1 -d"(" | sort > work_sorted.txt
cut notwork.txt -f1 -d"(" | sort > notwork_sorted.txt
diff work_sorted.txt notwork_sorted.txt
```
The result of diff command.
```
310a311,314
> 	libpcl_common.so.1.7 => /usr/lib/libpcl_common.so.1.7 
> 	libpcl_common.so.1.7 => /usr/lib/libpcl_common.so.1.7 
> 	libpcl_common.so.1.7 => /usr/lib/libpcl_common.so.1.7 
> 	libpcl_common.so.1.7 => /usr/lib/libpcl_common.so.1.7 
314a319,320
> 	libpcl_features.so.1.7 => /usr/lib/libpcl_features.so.1.7 
> 	libpcl_features.so.1.7 => /usr/lib/libpcl_features.so.1.7 
317,319c323,325
< 	libpcl_features.so.1.8 => /usr/local/lib/libpcl_features.so.1.8 
< 	libpcl_filters.so.1.8 => /usr/local/lib/libpcl_filters.so.1.8 
< 	libpcl_filters.so.1.8 => /usr/local/lib/libpcl_filters.so.1.8 
---
> 	libpcl_filters.so.1.7 => /usr/lib/libpcl_filters.so.1.7 
> 	libpcl_filters.so.1.7 => /usr/lib/libpcl_filters.so.1.7 
> 	libpcl_filters.so.1.7 => /usr/lib/libpcl_filters.so.1.7 
321a328,329
> 	libpcl_io.so.1.7 => /usr/lib/libpcl_io.so.1.7 
> 	libpcl_io.so.1.7 => /usr/lib/libpcl_io.so.1.7 
323a332,333
> 	libpcl_io_ply.so.1.7 => /usr/lib/libpcl_io_ply.so.1.7 
> 	libpcl_io_ply.so.1.7 => /usr/lib/libpcl_io_ply.so.1.7 
325a336,339
> 	libpcl_kdtree.so.1.7 => /usr/lib/libpcl_kdtree.so.1.7 
> 	libpcl_kdtree.so.1.7 => /usr/lib/libpcl_kdtree.so.1.7 
> 	libpcl_kdtree.so.1.7 => /usr/lib/libpcl_kdtree.so.1.7 
> 	libpcl_kdtree.so.1.7 => /usr/lib/libpcl_kdtree.so.1.7 
333c347,350
< 	libpcl_octree.so.1.8 => /usr/local/lib/libpcl_octree.so.1.8 
---
> 	libpcl_octree.so.1.7 => /usr/lib/libpcl_octree.so.1.7 
> 	libpcl_octree.so.1.7 => /usr/lib/libpcl_octree.so.1.7 
> 	libpcl_octree.so.1.7 => /usr/lib/libpcl_octree.so.1.7 
> 	libpcl_octree.so.1.7 => /usr/lib/libpcl_octree.so.1.7 
338a356,358
> 	libpcl_sample_consensus.so.1.7 => /usr/lib/libpcl_sample_consensus.so.1.7 
> 	libpcl_sample_consensus.so.1.7 => /usr/lib/libpcl_sample_consensus.so.1.7 
> 	libpcl_sample_consensus.so.1.7 => /usr/lib/libpcl_sample_consensus.so.1.7 
342,343c362,365
< 	libpcl_sample_consensus.so.1.8 => /usr/local/lib/libpcl_sample_consensus.so.1.8 
< 	libpcl_search.so.1.8 => /usr/local/lib/libpcl_search.so.1.8 
---
> 	libpcl_search.so.1.7 => /usr/lib/libpcl_search.so.1.7 
> 	libpcl_search.so.1.7 => /usr/lib/libpcl_search.so.1.7 
> 	libpcl_search.so.1.7 => /usr/lib/libpcl_search.so.1.7 
> 	libpcl_search.so.1.7 => /usr/lib/libpcl_search.so.1.7 
346a369,370
> 	libpcl_segmentation.so.1.7 => /usr/lib/libpcl_segmentation.so.1.7 
> 	libpcl_segmentation.so.1.7 => /usr/lib/libpcl_segmentation.so.1.7 
```

wkentaro, 2015-11-04:
these seems to affect euclid_clustering.
```
% rosrun jsk_pcl_ros euclidean_clustering \~input:=/bin_a_extract_pi/output _min_size:=500 _max_size:=100000 __name:=bin_a_euclid_clustering
[ INFO] [1446612660.507271436]: Initializing nodelet with 48 worker threads.
[ WARN] [1446612665.999617481]: '/bin_a_euclid_clustering' subscribes topics only with child subscribers.
[1]    35262 segmentation fault  rosrun jsk_pcl_ros euclidean_clustering \~input:=/bin_a_extract_pi/output
```

[image]

wkentaro, 2015-11-04:
output of gdb back trace
```
Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffe7fff700 (LWP 87055)]
0x00007ffff7bab680 in boost::detail::sp_counted_base::release() ()
   from /opt/ros/indigo/lib/libnodeletlib.so
(gdb) run
The program being debugged has been started already.
Start it from the beginning? (y or n) n
Program not restarted.
(gdb) bt
#0  0x00007ffff7bab680 in boost::detail::sp_counted_base::release() ()
   from /opt/ros/indigo/lib/libnodeletlib.so
#1  0x00007fffc640e8f8 in ~shared_count (this=0x7fff8c001da0, __in_chrg=<optimized out>)
    at /usr/include/boost/smart_ptr/detail/shared_count.hpp:371
#2  ~shared_array (this=0x7fff8c001d98, __in_chrg=<optimized out>)
    at /usr/include/boost/smart_ptr/shared_array.hpp:48
#3  ~KdTreeFLANN (this=0x7fff8c001d40, __in_chrg=<optimized out>)
    at /usr/local/include/pcl-1.8/pcl/kdtree/kdtree_flann.h:139
#4  pcl::KdTreeFLANN<pcl::PointXYZ, flann::L2_Simple<float> >::~KdTreeFLANN (
    this=0x7fff8c001d40, __in_chrg=<optimized out>)
    at /usr/local/include/pcl-1.8/pcl/kdtree/kdtree_flann.h:139
#5  0x00007ffff7bab69e in boost::detail::sp_counted_base::release() ()
   from /opt/ros/indigo/lib/libnodeletlib.so
#6  0x00007fffbc67f5b5 in ~shared_count (this=0x7fff8c001450, __in_chrg=<optimized out>)
    at /usr/include/boost/smart_ptr/detail/shared_count.hpp:371
#7  ~shared_ptr (this=0x7fff8c001448, __in_chrg=<optimized out>)
    at /usr/include/boost/smart_ptr/shared_ptr.hpp:328
#8  pcl::search::KdTree<pcl::PointXYZ, pcl::KdTreeFLANN<pcl::PointXYZ, flann::L2_Simple<float> > >::~KdTree (this=0x7fff8c001410, __in_chrg=<optimized out>)
    at /home/baxter/pcl/search/include/pcl/search/kdtree.h:99
#9  0x00007fffc62fda9a in destroy (this=0x7fff8c001408)
    at /usr/include/boost/smart_ptr/make_shared_object.hpp:57
#10 operator() (this=0x7fff8c001408) at /usr/include/boost/smart_ptr/make_shared_object.hpp:87
#11 boost::detail::sp_counted_impl_pd<pcl::search::KdTree<pcl::PointXYZ, pcl::KdTreeFLANN<pcl::PointXYZ, flann::L2_Simple<float> > >*, boost::detail::sp_ms_deleter<pcl::search::KdTree<pcl::PointXYZ, pcl::KdTreeFLANN<pcl::PointXYZ, flann::L2_Simple<float> > > > >::dispose (
    this=0x7fff8c0013f0) at /usr/include/boost/smart_ptr/detail/sp_counted_impl.hpp:153
#12 0x00007ffff7bab69e in boost::detail::sp_counted_base::release() ()
   from /opt/ros/indigo/lib/libnodeletlib.so
#13 0x00007fffc64b6875 in ~shared_count (this=0x7fffe7ffe808, __in_chrg=<optimized out>)
    at /usr/include/boost/smart_ptr/detail/shared_count.hpp:371
#14 ~shared_ptr (this=0x7fffe7ffe800, __in_chrg=<optimized out>)
    at /usr/include/boost/smart_ptr/shared_ptr.hpp:328
#15 pcl::EuclideanClusterExtraction<pcl::PointXYZ>::~EuclideanClusterExtraction (
    this=0x7fffe7ffe7d0, __in_chrg=<optimized out>)
    at /usr/local/include/pcl-1.8/pcl/segmentation/extract_clusters.h:295
#16 0x00007fffc64b443b in jsk_pcl_ros::EuclideanClustering::extract (this=0xa52b50, input=...)
    at /home/baxter/ros/indigo/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/src/euclidean_cluster_extraction_nodelet.cpp:63
#17 0x00007fffc6305c3c in operator() (a0=..., this=<optimized out>)
    at /usr/include/boost/function/function_template.hpp:767
#18 boost::detail::function::void_function_obj_invoker1<boost::function<void (boost::shared_ptr---Type <return> to continue, or q <return> to quit---
<sensor_msgs::PointCloud2_<std::allocator<void> > const> const&)>, void, boost::shared_ptr<sensor_msgs::PointCloud2_<std::allocator<void> > const> >::invoke(boost::detail::function::function_buffer&, boost::shared_ptr<sensor_msgs::PointCloud2_<std::allocator<void> > const>) (
    function_obj_ptr=..., a0=...) at /usr/include/boost/function/function_template.hpp:153
#19 0x00007fffc6363bb7 in operator() (a0=..., this=0x7fff8c0012a8)
    at /usr/include/boost/function/function_template.hpp:767
#20 ros::SubscriptionCallbackHelperT<boost::shared_ptr<sensor_msgs::PointCloud2_<std::allocator<void> > const> const&, void>::call (this=0x7fff8c0012a0, params=...)
    at /home/baxter/ros/indigo_parent/install/include/ros/subscription_callback_helper.h:144
#21 0x00007ffff78adad9 in ros::SubscriptionQueue::call() ()
   from /home/baxter/ros/indigo_parent/install/lib/libroscpp.so
#22 0x00007ffff783ff19 in ros::CallbackQueue::callOneCB(ros::CallbackQueue::TLS*) ()
   from /home/baxter/ros/indigo_parent/install/lib/libroscpp.so
#23 0x00007ffff783f639 in ros::CallbackQueue::callOne(ros::WallDuration) ()
   from /home/baxter/ros/indigo_parent/install/lib/libroscpp.so
#24 0x00007ffff7bc2230 in nodelet::detail::CallbackQueue::callOne() ()
   from /opt/ros/indigo/lib/libnodeletlib.so
#25 0x00007ffff7bc3426 in nodelet::detail::CallbackQueueManager::workerThread(nodelet::detail::CallbackQueueManager::ThreadInfo*) () from /opt/ros/indigo/lib/libnodeletlib.so
#26 0x00007ffff59bda4a in ?? () from /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.54.0
#27 0x00007ffff579c182 in start_thread (arg=0x7fffe7fff700) at pthread_create.c:312
#28 0x00007ffff6bad47d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:111
```

mmurooka, 2015-11-04:
```
#4  pcl::KdTreeFLANN<pcl::PointXYZ, flann::L2_Simple<float> >::~KdTreeFLANN (
    this=0x7fff8c001d40, __in_chrg=<optimized out>)
    at /usr/local/include/pcl-1.8/pcl/kdtree/kdtree_flann.h:139
```
octmapとリンクするとKdTreeが動かなくなる？
調べてみます．

wkentaro, 2015-11-04:
テストケース: https://gist.github.com/wkentaro/956e993e938ce0ad0989

mmurooka, 2015-11-04:
僕の手元では、hydroでjsk_recognitionのorigin/masterで再現しませんでした。

https://gist.github.com/mmurooka/a2d99dae5dc39902645e
で、確認したところSIGVは再現しませんでしたが、
```
% make
[100%] Building CXX object CMakeFiles/kdtree_search.dir/kdtree_search.cpp.o
Linking CXX executable kdtree_search
/usr/bin/ld: warning: libpcl_segmentation.so.1.7, needed by /opt/ros/indigo/lib/liboctomap_server.so, may conflict with libpcl_segmentation.so.1.8
/usr/bin/ld: warning: libpcl_features.so.1.8, needed by /usr/local/lib/libpcl_segmentation.so, may conflict with libpcl_features.so.1.7
/usr/bin/ld: warning: libpcl_kdtree.so.1.7, needed by /usr/lib/libpcl_features.so, may conflict with libpcl_kdtree.so.1.8

```
と、コンパイル時にでて、pclがソースから入っていて干渉しているようなので、
これが関係あるかもしれないです。

wkentaro, 2015-11-04:
ソースから入れると依存関係はどう解決するのがいいか、知っている人いますか。
すべて（octreeなどpclに依存しているものすべて）をソースというのが解でしょうか。

k-okada, 2015-11-04:
ソースから入れたら，基本はそれに依存しているものも全てコンパイルしなおしではないでしょうか

wkentaro, 2015-11-04:
*a*のパッケージがpcl-1.7にリンク
*b*のパッケージがpcl-1.8にリンク
*c*のパッケージが*a*,*b*にリンク

*a* = octomap_server
*b* = pcl_segmentation
*c* = jsk_pcl_ros

というときにsegfaultというのがこのissueの原因だと思うのですが，
*a*がpcl-1.7が1.8にリンクしてくれないとダメなので*a*もソースから入れるしかないでしょうか．

そうなると，pcl-1.7にリンクしてしまっているパッケージ（依存しているもの）は全てソースから入れなければいけなくなりますが，そういうものですか？

k-okada, 2015-11-04:
そうだとおもいます。なのでapiをきがるにかえるといいこちがないわけですが、
ところでpcl_segmentation はどこにあるでしょうか？なぜpcl1.8に依存していますか？

wkentaro, 2015-11-04:
となると避ける方法は①とことんソースでいれる or ②pcl-1.8を諦めるということになりますね．．

/usr/local/lib/libpcl_features.so.1.8
https://github.com/jsk-ros-pkg/jsk_recognition/issues/1300#issuecomment-153712264

でpclをソースからインストールしたものだと思います．
(おそらく @aginika がapcの為に入れていたのだと思います）

k-okada, 2015-11-05:
そもそなぜ1.8が入っているかが知りたいですが
十分な理由があればrosinstall_generatorを駆使して
ソースから入れればいいと思います

wkentaro, 2015-11-05:
@aginika にも確認しましたが，入れた覚えが特にないということなので，消します．

wkentaro, 2015-11-06:
pclを消すと治りました。

closed 2015-11-06 by wkentaro