jsk-ros-pkg/jsk_recognition #1308: Segfault SLICSuperPixels

wkentaro, 2015-11-09:
https://gist.github.com/wkentaro/84e4cd9ee84a206cef3e
```
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffef179700 (LWP 12514)]
[New Thread 0x7fffee978700 (LWP 12515)]
[New Thread 0x7fffee177700 (LWP 12516)]
[New Thread 0x7fffed976700 (LWP 12517)]
[New Thread 0x7fffed175700 (LWP 12518)]
[New Thread 0x7fffec974700 (LWP 12519)]
[New Thread 0x7fffe7fff700 (LWP 12521)]
[New Thread 0x7fffe77fe700 (LWP 12522)]
[New Thread 0x7fffe6ffd700 (LWP 12523)]
[New Thread 0x7fffe67fc700 (LWP 12524)]
[New Thread 0x7fffe5ffb700 (LWP 12525)]
[New Thread 0x7fffe57fa700 (LWP 12526)]
[New Thread 0x7fffe4ff9700 (LWP 12527)]
[ INFO] [1447083483.356138142]: Initializing nodelet with 8 worker threads.
[New Thread 0x7fffdbfff700 (LWP 12528)]
[New Thread 0x7fffad0be700 (LWP 12533)]
[New Thread 0x7fffab4d0700 (LWP 12562)]
[New Thread 0x7fffab0cf700 (LWP 12563)]
[New Thread 0x7fffaacce700 (LWP 12565)]
[New Thread 0x7fffaa8cd700 (LWP 12566)]
[New Thread 0x7fffaa0cb700 (LWP 12569)]
[New Thread 0x7fffaa4cc700 (LWP 12567)]
[New Thread 0x7fffa9cca700 (LWP 12570)]
*** Error in `/home/wkentaro/ros/indigo/devel/lib/jsk_perception/slic_super_pixels': munmap_chunk(): invalid pointer: 0x00007fff9f5674a0 ***

Program received signal SIGABRT, Aborted.
[Switching to Thread 0x7fffec974700 (LWP 12519)]
0x00007ffff6d83cc9 in __GI_raise (sig=sig@entry=6) at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
56      ../nptl/sysdeps/unix/sysv/linux/raise.c: No such file or directory.
```

garaemon, 2015-12-05:
give me gdb output

wkentaro, 2015-12-06:
not repriduced in your environment?

wkentaro, 2015-12-06:
https://github.com/jsk-ros-pkg/jsk_recognition/pull/1378

garaemon, 2015-12-10:
> not repriduced in your environment?

I have seen SEGV a lot of times.

garaemon, 2015-12-10:
I think slic does not need to release `cv::Mat`.

https://github.com/garaemon/SLIC-Superpixels/blob/master/slic.cpp#L24

garaemon, 2015-12-10:
https://github.com/garaemon/SLIC-Superpixels/pull/1

It may solve the issue.

wkentaro, 2015-12-12:
It does not work, but I don't use this node so it's ok.
this node maybe works only for small images. 
**360x486**
[[image]

**1080x1920**
[image]

garaemon, 2015-12-20:
I confirmed this issue.

garaemon, 2015-12-20:
Is it easy to rewrite it with scikit image?

wkentaro, 2015-12-20:
> Is it easy to rewrite it with scikit image?

Probably, but I don't have the motivation for that now, because I'm not using `slic` currently.
And `slic` seems used with other algorithms and in that case it does not need to be a standalone node.
http://scikit-image.org/docs/dev/api/skimage.segmentation.html?highlight=slic#skimage.segmentation.slic

Ridhwanluthra, 2018-02-27:
is this solved?

wkentaro, 2018-03-17:
@Ridhwanluthra I think it has not fixed yet.

PR #2450 (2019-09-20): Update to slic d77d6e8
Fix #1308 

In previous version of submodule `slic`, there was invalid access to element of minus index of `std::vector`.
This caused segmentation fault in `jsk_perception/slic_super_pixels` node during destructor of that vector was called.
Probably this is why `slic_super_pixels` fails when large image is used (reported in #1308 ).

I updated `slic` to the latest commit and confirmed it works with large image.
i.e. `jsk_perception/sample/kiva_pod_image_color.jpg` (size is 1920 x 1080)
[image]
[image]
[image]

Thanks @pazeshun
closed 2019-09-20 by k-okada