jsk-ros-pkg/jsk_recognition #1311: edge detector duplication

aginika, 2015-11-11:
From https://github.com/ros-perception/vision_opencv/issues/80

Below are duplicating
https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/jsk_perception/src/edge_detector.cpp
https://github.com/ros-perception/vision_opencv/blob/indigo/opencv_apps/src/nodelet/edge_detection_nodelet.cpp

@k-okada I didn't intend to write the documents of opencv_apps .... Did you mean jsk_perception?

k-okada, 2015-11-11:
> From ros-perception/vision_opencv#80

opencv_apps are stable version of the jsk_recognition, so if that is ok,
we'll use them and remove correspondence src from jsk_recognition.
so add tutorials to the opencv_apps, wiki is ok, just a few line on how to
run launch or how to run crom command line, and ask kochigami-san to put
image on that

aginika, 2015-11-11:
OK, Thank you. I wrote
http://wiki.ros.org/opencv_apps

closed 2015-11-11 by aginika