jsk-ros-pkg/jsk_recognition #1312: Compile error in organized_edge_detector_nodelet.cpp

snozawa, 2015-11-12:
When I ran catkin build on my laptop, jsk_pcl_ros compile failed.
```
nozawa@test1-pc:~/catkin_ws/ws_robotprogramming/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros$ catkin bt -i -v
```
The error message is here:
```
[jsk_pcl_ros] Built target voxel_grid_downsample_decoder
[jsk_pcl_ros] [ 48%] [ 49%] Built target jsk_pcl_ros_generate_messages_cpp
[jsk_pcl_ros] Built target voxel_grid_downsample_manager
[jsk_pcl_ros] [ 53%] [ 59%] Built target jsk_pcl_ros_generate_messages_lisp
[jsk_pcl_ros] Built target jsk_pcl_ros_generate_messages_eus
[jsk_pcl_ros] [ 65%] Built target jsk_pcl_ros_generate_messages_py
[jsk_pcl_ros] [ 65%] Built target jsk_pcl_ros_gencpp
[jsk_pcl_ros] [ 65%] Built target jsk_pcl_ros_generate_messages
[jsk_pcl_ros] [ 65%] Built target jsk_pcl_ros_base
[jsk_pcl_ros] [ 65%] Built target jsk_pcl_ros_moveit
[jsk_pcl_ros] [ 66%] Built target sample_cube_nearest_point
[jsk_pcl_ros] [ 72%] Built target jsk_pcl_ros_util
[jsk_pcl_ros] Scanning dependencies of target jsk_pcl_ros
[jsk_pcl_ros] [ 72%] [ 72%] [ 73%] [ 73%] Building CXX object CMakeFiles/jsk_pcl_ros.dir/src/plane_supported_cuboid_estimator_nodelet.cpp.o
[jsk_pcl_ros] Building CXX object CMakeFiles/jsk_pcl_ros.dir/src/extract_cuboid_particles_top_n_nodelet.cpp.o
[jsk_pcl_ros] Building CXX object CMakeFiles/jsk_pcl_ros.dir/src/interactive_cuboid_likelihood_nodelet.cpp.o
[jsk_pcl_ros] Building CXX object CMakeFiles/jsk_pcl_ros.dir/src/organized_edge_detector_nodelet.cpp.o
[jsk_pcl_ros] /home/nozawa/catkin_ws/ws_robotprogramming/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/src/organized_edge_detector_nodelet.cpp:39:51: fatal error: pcl/features/organized_edge_detection.h:
[jsk_pcl_ros]  #include <pcl/features/organized_edge_detection.h>
[jsk_pcl_ros]                                                    ^
[jsk_pcl_ros] compilation terminated.
[jsk_pcl_ros] make[2]: *** [CMakeFiles/jsk_pcl_ros.dir/src/organized_edge_detector_nodelet.cpp.o]  1
[jsk_pcl_ros] make[2]: *** ....
[jsk_pcl_ros] make[1]: *** [CMakeFiles/jsk_pcl_ros.dir/all]  2
[jsk_pcl_ros] make: *** [all]  2
[jsk_pcl_ros] <== '/home/nozawa/catkin_ws/ws_robotprogramming/build/jsk_pcl_ros/build_env.sh /usr/bin/make --jobserver-fds=3,5 -j' failed with return code '2'
Failed   <== jsk_pcl_ros           [ 46.3 seconds ]
[build] There were '1' errors:

Failed to build package 'jsk_pcl_ros' because the following command:

# Command to reproduce:
cd /home/nozawa/catkin_ws/ws_robotprogramming/build/jsk_pcl_ros && /home/nozawa/catkin_ws/ws_robotprogramming/build/jsk_pcl_ros/build_env.sh /usr/bin/make --jobserver-fds=3,5 -j; cd -

# Path to log:
cat /home/nozawa/catkin_ws/ws_robotprogramming/build/build_logs/jsk_pcl_ros.log

Exited with return code: 2

[build] Runtime: 49.9 seconds
```

My laptop is:
- Ubuntu: 14.04
- ROS: indigo
- PCL: 1.7.2 (deb)

In CMakeLists.txt, PCL 1.7.1+ is required.
https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/jsk_pcl_ros/CMakeLists.txt#L288
But, on my laptop, PCL version is greater than 1.7.1 and 
`
pcl/features/organized_edge_detection.h
`
is missing.

How can I fix this?

k-okada, 2015-11-12:
duplicates of https://github.com/jsk-ros-pkg/jsk_recognition/pull/1301

k-okada, 2015-11-12:
merged, 

btw, are you going to let B3 students to compile jsk_recognition ;-9

snozawa, 2015-11-12:
Thanks and sorry for existent problem:
```
duplicates of #1301
```

> btw, are you going to let B3 students to compile jsk_recognition ;-9

Yew, currently, deb-installed jsk-pcl-ros seem not to work:
https://github.com/jsk-ros-pkg/jsk_recognition/issues/1313

closed 2015-11-12 by snozawa