jsk-ros-pkg/jsk_recognition #1313: hsi_color_filter does not work on deb ros-indigo-jsk-pcl-ros (0.3.6-0trusty-20150915-1556-+0000)

snozawa, 2015-11-12:
I installed ros-indigo-jsk-pcl-ros from deb  (0.3.6-0trusty-20150915-1556-+0000).

I tried hsi_color_filter, which worked one year before, 
and it failes by linking error.

* How to reproduce problem
- Launch hsi_color_filter: (example on turtlebot + Ubuntu14.04 + indigo)
```
roslaunch jsk_pcl_ros hsi_color_filter.launch DEFAULT_NAMESPACE:=/camera/depth_registered INPUT:=points
```
- Subscribe results:
```
rostopic echo /camera/depth_registered/debug_output
```
- hsi_color_filter died like this:
```
nozawa@test1-pc:~/catkin_ws/ws_robotprogramming/src$ roslaunch jsk_pcl_ros hsi_color_filter.launch DEFAULT_NAMESPACE:=/camera/depth_registered INPUT:=points
... logging to /home/nozawa/.ros/log/041baf44-88fa-11e5-b33d-68f72824216e/roslaunch-test1-pc-3602.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.
^[]2;/opt/ros/indigo/share/jsk_pcl_ros/launch/hsi_color_filter.launch^G
started roslaunch server http://192.168.97.161:35093/

SUMMARY
========

CLEAR PARAMETERS
 * /camera/depth_registered/cluster_decomposer/

PARAMETERS
 * /camera/depth_registered/euclidean_clustering/min_size: 100
 * /camera/depth_registered/euclidean_clustering/tolerance: 0.02
 * /camera/depth_registered/hsi_filter/h_limit_max: 127
 * /camera/depth_registered/hsi_filter/h_limit_min: -128
 * /camera/depth_registered/hsi_filter/i_limit_max: 255
 * /camera/depth_registered/hsi_filter/i_limit_min: 0
 * /camera/depth_registered/hsi_filter/s_limit_max: 255
 * /camera/depth_registered/hsi_filter/s_limit_min: 0
 * /camera/depth_registered/hsi_filter/use_indices: False
 * /rosdistro: indigo
 * /rosversion: 1.11.13

NODES
  /camera/depth_registered/
    cluster_decomposer (nodelet/nodelet)
    euclidean_clustering (nodelet/nodelet)
    hsi_filter (nodelet/nodelet)
    hsi_filter_manager (nodelet/nodelet)

ROS_MASTER_URI=http://localhost:11311
^[]2;/opt/ros/indigo/share/jsk_pcl_ros/launch/hsi_color_filter.launch http://localhost:11311^G
core service [/rosout] found
process[camera/depth_registered/hsi_filter_manager-1]: started with pid [3621]
process[camera/depth_registered/hsi_filter-2]: started with pid [3622]
[ INFO] [1447306704.165496822]: Initializing nodelet with 8 worker threads.
[ INFO] [1447306704.170047185]: Loading nodelet /camera/depth_registered/hsi_filter of type jsk_pcl/HSIColorFilter to manager hsi_filter_manager with the following remappings:
[ INFO] [1447306704.170090843]: /camera/depth_registered/hsi_filter/input -> /camera/depth_registered/points
[ INFO] [1447306704.170101964]: /camera/depth_registered/hsi_filter/output -> /camera/depth_registered/hsi_output
process[camera/depth_registered/euclidean_clustering-3]: started with pid [3663]
[ INFO] [1447306704.212633103]: Loading nodelet /camera/depth_registered/euclidean_clustering of type jsk_pcl/EuclideanClustering to manager hsi_filter_manager with the following remappings:
[ INFO] [1447306704.212690834]: /camera/depth_registered/euclidean_clustering/input -> /camera/depth_registered/hsi_output
process[camera/depth_registered/cluster_decomposer-4]: started with pid [3680]
[ INFO] [1447306704.260163051]: Loading nodelet /camera/depth_registered/cluster_decomposer of type jsk_pcl/ClusterPointIndicesDecomposerZAxis to manager hsi_filter_manager with the following remappings:
[ INFO] [1447306704.260215573]: /camera/depth_registered/cluster_decomposer/boxes -> /camera/depth_registered/boxes
[ INFO] [1447306704.260231197]: /camera/depth_registered/cluster_decomposer/debug_output -> /camera/depth_registered/debug_output
[ INFO] [1447306704.260245093]: /camera/depth_registered/cluster_decomposer/input -> /camera/depth_registered/hsi_output
[ INFO] [1447306704.260257922]: /camera/depth_registered/cluster_decomposer/target -> /camera/depth_registered/euclidean_clustering/output
/opt/ros/indigo/lib/nodelet/nodelet: symbol lookup error: /opt/ros/indigo/lib/libjsk_pcl_ros.so: undefined symbol: _ZN3pcl6search6KdTreeINS_8PointXYZEEC1Eb
[camera/depth_registered/hsi_filter_manager-1] process has died [pid 3621, exit code 127, cmd /opt/ros/indigo/lib/nodelet/nodelet manager __name:=hsi_filter_manager __log:=/home/nozawa/.ros/log/041baf44-88fa-11e5-b33d-68f72824216e/camera-depth_registered-hsi_filter_manager-1.log].                                                                                               
log file: /home/nozawa/.ros/log/041baf44-88fa-11e5-b33d-68f72824216e/camera-depth_registered-hsi_filter_manager-1*.log
[ INFO] [1447306739.412314974]: Bond broken, exiting
[ INFO] [1447306739.511610200]: Bond broken, exiting
[ INFO] [1447306739.514791763]: Bond broken, exiting
[camera/depth_registered/hsi_filter-2] process has finished cleanly                                                                                                                                                                                                                                                                                                                     
log file: /home/nozawa/.ros/log/041baf44-88fa-11e5-b33d-68f72824216e/camera-depth_registered-hsi_filter-2*.log
[camera/depth_registered/euclidean_clustering-3] process has finished cleanly                                                                                                                                                                                                                                                                                                           
log file: /home/nozawa/.ros/log/041baf44-88fa-11e5-b33d-68f72824216e/camera-depth_registered-euclidean_clustering-3*.log
[camera/depth_registered/cluster_decomposer-4] process has finished cleanly                                                                                                                                                                                                                                                                                                             
log file: /home/nozawa/.ros/log/041baf44-88fa-11e5-b33d-68f72824216e/camera-depth_registered-cluster_decomposer-4*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

The cause of error seems to be linking error of pcl::kdtree:
```
/opt/ros/indigo/lib/nodelet/nodelet: symbol lookup error: /opt/ros/indigo/lib/libjsk_pcl_ros.so: undefined symbol: _ZN3pcl6search6KdTreeINS_8PointXYZEEC1Eb
```

This is ros-indigo-jsk-pcl-ros version information:
```
nozawa@test1-pc:~/catkin_ws/ws_robotprogramming/src/robot-programming/dxl_armed_turtlebot/euslisp$ aptitude show ros-indigo-jsk-pcl-ros
パッケージ: ros-indigo-jsk-pcl-ros
状態: インストール済み
自動的にインストールされた: いいえ
バージョン: 0.3.6-0trusty-20150915-1556-+0000
優先度: 特別
セクション: misc
メンテナ: Youhei Kakiuchi <youhei@jsk.t.u-tokyo.ac.jp>
アーキテクチャ: amd64
展開サイズ: 24.9 M
依存: libboost-filesystem1.54.0, libboost-system1.54.0, libboost-thread1.54.0,
        libc6 (>= 2.14), libconsole-bridge0.2, libgcc1 (>= 1:4.1.1), libgomp1
        (>= 4.2.1), libopencv-core2.4, libopencv-imgproc2.4, libpcl-common-1.7,
        libpcl-features-1.7, libpcl-filters-1.7, libpcl-io-1.7,
        libpcl-kdtree-1.7, libpcl-keypoints-1.7, libpcl-octree-1.7,
        libpcl-recognition-1.7, libpcl-search-1.7, libpcl-segmentation-1.7,
        libpcl-surface-1.7, libpcl-tracking-1.7, libstdc++6 (>= 4.6),
        libyaml-cpp0.5, ros-indigo-octomap, libboost-all-dev, libyaml-cpp-dev,
        python-sklearn, ros-indigo-cv-bridge, ros-indigo-diagnostic-msgs,
        ros-indigo-diagnostic-updater, ros-indigo-dynamic-reconfigure,
        ros-indigo-eigen-conversions, ros-indigo-geometry-msgs,
        ros-indigo-image-geometry, ros-indigo-image-transport,
        ros-indigo-image-view2, ros-indigo-interactive-markers,
        ros-indigo-jsk-footstep-msgs, ros-indigo-jsk-recognition-msgs,
        ros-indigo-jsk-recognition-utils, ros-indigo-jsk-topic-tools,
        ros-indigo-laser-assembler, ros-indigo-message-runtime,
        ros-indigo-ml-classifiers, ros-indigo-moveit-core,
        ros-indigo-moveit-ros-perception, ros-indigo-nodelet,
        ros-indigo-pcl-conversions, ros-indigo-pcl-msgs, ros-indigo-pcl-ros,
        ros-indigo-rosboost-cfg, ros-indigo-roscpp-tutorials,
        ros-indigo-sensor-msgs, ros-indigo-sklearn, ros-indigo-std-msgs,
        ros-indigo-std-srvs, ros-indigo-stereo-msgs, ros-indigo-tf,
        ros-indigo-tf-conversions, ros-indigo-tf2-ros,
        ros-indigo-visualization-msgs
競合: ros-indigo-jsk-pcl-ros
説明: jsk_pcl_ros
 
ホームページ: http://ros.org/wiki/jsk_pcl_ros
```

Do we need to release jsk_pcl_ros?

k-okada, 2015-11-12:
aha, jsk_pcl_ros is compiled against pcl 1.7.1 on ros buildfirm and not
working with pcl-1.7.2 on your machine, so even if you release newer
version, it may fail.
Also your computer did not use shadow-fixed, so even if you release newer
version, it will take ~month to use them

another option is to recompile from deb package.

k-okada, 2015-11-12:
```
#!/bin/bash

set -x
set -e
mkdir -p /tmp/jsk-pcl-ros-build
cd /tmp/jsk-pcl-ros-build
if [ ! -e ros-indigo-jsk-pcl-ros_0.3.6.orig.tar.gz ]; then
    wget wget
http://packages.ros.org/ros/ubuntu/pool/main/r/ros-indigo-jsk-pcl-ros/ros-indigo-jsk-pcl-ros_0.3.6.orig.tar.gz
fi
if [ ! -e ros-indigo-jsk-pcl-ros_0.3.6-0trusty.debian.tar.gz ]; then
    wget wget
http://packages.ros.org/ros/ubuntu/pool/main/r/ros-indigo-jsk-pcl-ros/ros-indigo-jsk-pcl-ros_0.3.6-0trusty.debian.tar.gz
fi
rm -fr ros-indigo-jsk-pcl-ros-0.3.6 || echo "ok"
tar -xvzf ros-indigo-jsk-pcl-ros_0.3.6.orig.tar.gz
cd ros-indigo-jsk-pcl-ros-0.3.6
tar -xvzf ../ros-indigo-jsk-pcl-ros_0.3.6-0trusty.debian.tar.gz
sed -i s@0.3.6-0@0.3.6-99@ debian/changelog
dpkg-buildpackage -rfakeroot -b
```
will recompile deb package, so you can put generated deb package and ask
them to install this

wkentaro, 2016-10-30:
Is this fixed for current apt version?

closed 2019-04-17 by YutoUchimi