jsk-ros-pkg/jsk_recognition #1316: How to use Skeletonization?

rkoyama1623, 2015-11-14:
@iKrishneel 

I am trying to recognize truss, so I want to use skeltonization.
I tried skeltonization.launch by using following picture.

* topoic name: /image_raw
[image]

and edge_detector and dilate_mask_image output following pictures.

* topic name: /edge/image
[image]

* topoic name: /dilate_mask_image/output
[image]

However, skeletonization does not output good results. The output is empty as follows.
* topic name: /skeltonization/image_output
[image]

Does anyone have good idea?
skeltonization.launch's contents is as follows:
```
<launch>
  <node pkg="jsk_perception" type="edge_detector" name="edge_detector">
    <remap from="image" to="/image_raw" />
  </node>
  <node pkg="jsk_perception" type="dilate_mask_image" name="dilate_mask_image">
    <remap from="~input" to="/edge/image" />
  </node>
  <node pkg="jsk_perception" type="skeletonization" name="skeletonization">
    <remap from="~input" to="/dilate_mask_image/output" />
  </node>
  <node pkg="image_view" type="image_view" name="edge_view" >
    <remap from="image" to="/edge/image" />
  </node>
  <node pkg="rqt_reconfigure" type="rqt_reconfigure" name="edge_reconfigure" />
</launch>
```

wkentaro, 2015-11-14:
> However, skeletonization does not output good results. The output is empty as follows.

What kind of output do you expect?
I think that output is normal, because the input image is very simple.
Maybe you can change the params with `rqt_reconfigure`. or try more complex input image.

rkoyama1623, 2015-11-14:
最終的に，赤い線のような画像が出て欲しいです．
hough_lineを適応する前処理として，細線化をしたくて，issueを立てました．

[image]

iKrishneel, 2015-11-14:
@rkoyama1623 
I tried the node and its working.

As @wkentaro said, can you try to increase the dilate_mask_image in rqt_reconfigure, perhaps by large amount until both sides of the edges fades as one. Please try to confirm using live stream on Xtion sensor.

@wkentaro 
The output will be single pixel wide skeleton of the edge.

Input edge
[image]

Dilated Edges
[image]

Skeleton
[image]

wkentaro, 2015-11-14:
yeah, I browsed your reference a bit.
http://rosettacode.org/wiki/Zhang-Suen_thinning_algorithm

rkoyama1623, 2015-11-14:
Thank you for advices.
I changed picture to my webcam, and the Dilated Edges are as follows:
[image]

but still /skeltonization/image_output is black.
I also checked jsk_perception version, but it was latest.

> Please try to confirm using live stream on Xtion sensor.

my input image is 2D. Is it OK?

wkentaro, 2015-11-14:
@rkoyama1623 
It works if the input image is 2D mono.
I tried use image you attached llike
```
wget https://cloud.githubusercontent.com/assets/11606776/11163012/1fd3e562-8b02-11e5-9415-6700aa3d1459.png -O koyama.png
rosrun jsk_perception image_publisher _file_name:=koyama.png
rosrun jsk_perception skeletonization \~input:=/image_publisher/output
rosrun image_view image_view image:=/skeletnization/image_output
```
[image]

iKrishneel, 2015-11-14:
@rkoyama1623 
my input image is 2D. Is it OK?
- Yes it works only for 2D image space.

Please check input image type. It only takes single channel 8bit input.

wkentaro, 2015-11-14:
>  It only takes single channel 8bit input.

but there is conversation between input image encoding to 8bit mono one here:
https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/jsk_perception/src/skeletonization_nodelet.cpp#L68-L74

so if there is no errors (liks `cv_bridge exception XXXX`), the conversation should be working.

wkentaro, 2015-11-14:
maybe there is an another reason why the `skeletonization` does not work on @rkoyama1623 's PC.

iKrishneel, 2015-11-14:
If @rkoyama1623 does not get an exception, the encoding is correct so I am wondering what is the reason it does not work when he runs the node.

rkoyama1623, 2015-11-14:
@wkentaro solved my pc problem.
my image_view and rqt_image_view was old (hydro), and they cannot display the message with encoding type of 32FC1.
By using rviz, the image was displayed correctly.

The time seem to come when I must update the ros version to indigo.

Thank you for everything, @wkentaro and @iKrishneel !

wkentaro, 2015-11-14:
@rkoyama1623 
Please add documentation about the output topic's image encoding here:
 http://jsk-recognition.readthedocs.org/en/latest/jsk_perception/nodes/skeletonization.html#publishing-topic.

like http://jsk-recognition.readthedocs.org/en/latest/jsk_perception/nodes/mask_image_to_label.html

wkentaro, 2015-11-14:
you can change doc here. https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/doc/jsk_perception/nodes/skeletonization.md

rkoyama1623, 2015-11-14:
OK, I sent PR.
https://github.com/jsk-ros-pkg/jsk_recognition/pull/1317

closed 2015-11-14 by rkoyama1623