jsk-ros-pkg/jsk_recognition #1322: How can we test c++ function?

wkentaro, 2015-11-15:
I mean function like https://github.com/jsk-ros-pkg/jsk_recognition/pull/1314

wkentaro, 2015-11-15:
gtest is a candidate.
http://wiki.ros.org/gtest

but I have not found method `assert_called_with` in mock for python.

k-okada, 2015-11-15:
is this answer your question?
http://answers.ros.org/question/181708/rostest-minimum-working-example/?answer=181934#post-id-181934
https://github.com/ros/ros_comm/blob/indigo-devel/test/test_roscpp/test/test_header.cpp

wkentaro, 2015-11-15:
yeah, I found gtest and created PR here: https://github.com/jsk-ros-pkg/jsk_common/pull/1238
(I mistook creating issue on jsk_recognition not jsk_common)

Now I'm wondering there is c++ version of `mock.assert_called_with` in python..

wkentaro, 2015-11-15:
I'd like test this kind of c++ function.
https://github.com/jsk-ros-pkg/jsk_common/pull/1229/files

But this function does not return value, so maybe it is difficult.
with python, I managed it like https://github.com/jsk-ros-pkg/jsk_common/pull/1237/files#diff-84305c6b98b19f3cad0718b5dc2501a6R47

wkentaro, 2015-11-15:
**Memo**
Test C++ code with Rostest: https://github.com/jsk-ros-pkg/jsk_recognition/pull/1324

wkentaro, 2015-11-16:
@k-okada
返り値がvoidの関数のテストは、単体ではできないでしょうか。それとも、boolを返り値にするのが正解ですか。

k-okada, 2015-11-17:
わかっていませんが，多分 yes

wkentaro, 2015-11-18:
ありがとうございます。

closed 2015-11-25 by k-okada