jsk-ros-pkg/jsk_recognition #1323: cannot compile jsk_pcl_ros on fuji

garaemon, 2015-11-15:
@mmurooka 

```
[octomap_server] ==> '/home/jskuser/ros/hydro/build/octomap_server/build_env.sh /usr/bin/make --jobserver-fds=3,5 -j' in '/home/jskuser/ros/hydro/build/octomap_server'
[ 10%] Built target octomap_server_gencfg
[ 20%] Building CXX object CMakeFiles/octomap_server.dir/src/OctomapServer.cpp.o
/home/jskuser/ros/hydro/src/OctoMap/octomap_mapping/octomap_server/src/OctomapServer.cpp:  void octomap_server::OctomapServer::initContactSensor(ros::NodeHandle) :
/home/jskuser/ros/hydro/src/OctoMap/octomap_mapping/octomap_server/src/OctomapServer.cpp:327:101: : no matching function for call to robot_self_filter::SelfMask<pcl::PointXYZ>::SelfMask(tf::TransformListener&, std::vector<robot_self_filter::LinkInfo>&, const char [9])
/home/jskuser/ros/hydro/src/OctoMap/octomap_mapping/octomap_server/src/OctomapServer.cpp:327:101: : :
/home/jskuser/ros/hydro/src/pr2_navigation/pr2_navigation_self_filter/include/pr2_navigation_self_filter/self_mask.h:175:2: : robot_self_filter::SelfMask<PointT>::SelfMask(tf::TransformListener&, const std::vector<robot_self_filter::LinkInfo>&) [with PointT = pcl::PointXYZ]
/home/jskuser/ros/hydro/src/pr2_navigation/pr2_navigation_self_filter/include/pr2_navigation_self_filter/self_mask.h:175:2: :    2 3 
/home/jskuser/ros/hydro/src/pr2_navigation/pr2_navigation_self_filter/include/pr2_navigation_self_filter/self_mask.h:145:11: : robot_self_filter::SelfMask<pcl::PointXYZ>::SelfMask(const robot_self_filter::SelfMask<pcl::PointXYZ>&)
/home/jskuser/ros/hydro/src/pr2_navigation/pr2_navigation_self_filter/include/pr2_navigation_self_filter/self_mask.h:145:11: :    1 3 
/home/jskuser/ros/hydro/src/OctoMap/octomap_mapping/octomap_server/src/OctomapServer.cpp:  void octomap_server::OctomapServer::insertContactSensor(std::vector<octomap_msgs::ContactSensor_<std::allocator<void> > >) :
/home/jskuser/ros/hydro/src/OctoMap/octomap_mapping/octomap_server/src/OctomapServer.cpp:378:33: : class robot_self_filter::SelfMask<pcl::PointXYZ> has no member named getMaskContainmentforNamedLink
/home/jskuser/ros/hydro/src/OctoMap/octomap_mapping/octomap_server/src/OctomapServer.cpp:385:42: : class robot_self_filter::SelfMask<pcl::PointXYZ> has no member named getMaskContainmentforNamedLink
make[2]: *** [CMakeFiles/octomap_server.dir/src/OctomapServer.cpp.o]  1
make[1]: *** [CMakeFiles/octomap_server.dir/all]  2
make: *** [all]  2

```

garaemon, 2015-11-15:
```
[http://hrp2016v:10016][192.168.96.103] jskuser@fuji:~/ros/hydro/src/OctoMap/octomap_mapping/octomap_server$ git branch
* insert-contact-sensor
  master
[http://hrp2016v:10016][192.168.96.103] jskuser@fuji:~/ros/hydro/src/OctoMap/octomap_mapping/octomap_server$ git status
# On branch insert-contact-sensor
# Changes not staged for commit:
#   (use "git add <file>..." to update what will be committed)
#   (use "git checkout -- <file>..." to discard changes in working directory)
#
#	modified:   launch/hrp2jsknt_octomap_reaching_shelf.launch
#	modified:   scripts/hrp2_contact_sensor.py
#
```

mmurooka, 2015-11-17:
遅くなりました．
octomap_serverについては，diffはcheckoutしてからmasterにしました．
octomap_server, jsk_pcl_rosのbuild通りました．

closed 2015-11-28 by garaemon