jsk-ros-pkg/jsk_recognition #1331: Request for new version release

snozawa, 2015-11-18:
Request for new version release and tag to include https://github.com/jsk-ros-pkg/jsk_recognition/pull/1301.

Is this release possible through jenkins?

k-okada, 2015-11-18:
released new version http://jenkins.jsk.imi.i.u-tokyo.ac.jp:8080/job/add_tag/146/, but for your purpsoe, please follow my instruction at https://github.com/jsk-ros-pkg/jsk_recognition/issues/1313#issuecomment-156013536

k-okada, 2015-11-18:
and your student uses ros from released repo, not shadow fixed

k-okada, 2015-11-18:
Because of https://github.com/jsk-ros-pkg/jsk_recognition/pull/1213/files
We could not release this 
Any idea?   @mmurooka

k-okada, 2015-11-18:
Btw why this passed the test

mmurooka, 2015-11-18:
Where can I see the error message?

k-okada, 2015-11-18:
hope this has been resolved by
https://github.com/jsk-ros-pkg/jsk_recognition/pull/1334
This mean you rosdep install does not install robot_self_filter, you have
to intentionally install them via source

```
Would you like to try again? [Y/n]? y
Running 'rosdep update'...
Key 'python-dlib' resolved to '['dlib']' with installer 'pip', which does
not match the default installer 'apt'.
Failed to resolve python-dlib on ubuntu:saucy with: Error running
generator: The Debian generator does not support dependencies which are
installed with the 'pip' installer.
python-dlib is depended on by these packages: ['jsk_perception']
<== Failed
Key 'python-dlib' resolved to '['dlib']' with installer 'pip', which does
not match the default installer 'apt'.
Failed to resolve python-dlib on ubuntu:trusty with: Error running
generator: The Debian generator does not support dependencies which are
installed with the 'pip' installer.
python-dlib is depended on by these packages: ['jsk_perception']
<== Failed
Key 'python-gdown' resolved to '['gdown']' with installer 'pip', which does
not match the default installer 'apt'.
Failed to resolve python-gdown on ubuntu:saucy with: Error running
generator: The Debian generator does not support dependencies which are
installed with the 'pip' installer.
python-gdown is depended on by these packages: ['jsk_perception']
<== Failed
Key 'python-gdown' resolved to '['gdown']' with installer 'pip', which does
not match the default installer 'apt'.
Failed to resolve python-gdown on ubuntu:trusty with: Error running
generator: The Debian generator does not support dependencies which are
installed with the 'pip' installer.
python-gdown is depended on by these packages: ['jsk_perception']
<== Failed
Could not resolve rosdep key 'robot_self_filter'
Failed to resolve robot_self_filter on ubuntu:saucy with: Error running
generator: Failed to resolve rosdep key 'robot_self_filter', aborting.
robot_self_filter is depended on by these packages: ['jsk_pcl_ros']
<== Failed
Could not resolve rosdep key 'robot_self_filter'
Failed to resolve robot_self_filter on ubuntu:trusty with: Error running
generator: Failed to resolve rosdep key 'robot_self_filter', aborting.
robot_self_filter is depended on by these packages: ['jsk_pcl_ros']
<== Failed
Some of the dependencies for packages in this repository could not be
resolved by rosdep.
You can try to address the issues which appear above and try again if you
wish.
Would you like to try again? [Y/n]? n
User aborted after rosdep keys were not resolved.
<== The following generator action reported that it is missing one or more
    rosdep keys, but that the key exists in other platforms:
'['/usr/bin/git-bloom-generate', '-y', 'rosdebian', '--prefix',
'release/indigo', 'indigo', '-i', '0']'

If you are absolutely sure that this key is unavailable for the platform in
question, the generator can be skipped and you can proceed with the release.
Skip generator action and continue with release [y/N]? n

<== Error running command '['/usr/bin/git-bloom-generate', '-y',
'rosdebian', '--prefix', 'release/indigo', 'indigo', '-i', '0']'
Release failed, exiting
```

mmurooka, 2015-11-19:
> This mean you rosdep install does not install robot_self_filter, you have to intentionally install them via source 

Yes, the apt package of robot_self_filter is not released now. https://github.com/PR2/robot_self_filter/issues/3
https://github.com/jsk-ros-pkg/jsk_recognition/pull/1213#issuecomment-147377177 is related to this.

> Btw why this passed the test

In travis test,
`Could not resolve rosdep key 'robot_self_filter'` error is ignored by`-r` option.

wkentaro, 2015-12-03:
Now `robot_self_filter` is released. https://github.com/PR2/robot_self_filter/issues/3#issuecomment-161046436
Is that possible to release this package for now?
(I'm trying to test it https://github.com/jsk-ros-pkg/jsk_recognition/pull/1362)

wkentaro, 2015-12-12:
@snozawa This can be close?

closed 2015-12-14 by k-okada