jsk-ros-pkg/jsk_recognition #1356: fails to build jsk_pcl_ros/collision_detector_nodelet.cpp & jsk_pcl_ros/octomap_server_contact_nodelet.cpp

furushchev, 2015-11-29:
catkin build fails on building jsk_pcl_ros/collision_detector_nodelet.cpp & jsk_pcl_ros/octomap_server_contact_nodelet.cpp.

[jsk_pcl_ros_build_log.txt](https://github.com/jsk-ros-pkg/jsk_recognition/files/46561/jsk_pcl_ros_build_log.txt)

in my env (hydro):
`robot_self_filter` is installed from source, and confirmed it is the latest.

```bash
$ rosversion robot_self_filter
0.1.26
```

and `jsk_pcl_ros` is also cloned the latest source.
(HEAD is now at 934d9b1 Merge pull request #1354 from garaemon/measure-tracking-computation-time)

```bash
$ head package.xml 
<package>
  <name>jsk_pcl_ros</name>
  <version>0.3.7</version>
  <description>jsk_pcl_ros</description>
  <maintainer email="youhei@jsk.t.u-tokyo.ac.jp">Youhei Kakiuchi</maintainer>

  <license>BSD</license>

  <url type="website">http://ros.org/wiki/jsk_pcl_ros</url>
  <!-- <url type="bugtracker"></url> -->
```

It seems to be related with https://github.com/PR2/robot_self_filter/pull/4, and https://github.com/PR2/robot_self_filter/pull/5
@mmurooka

mmurooka, 2015-11-29:
> It seems to be related with PR2/robot_self_filter#4, and PR2/robot_self_filter#5

Yes, if you merge those two PRs, build should be succeded.

現状が正解だったのか自信はないのですが，
https://github.com/jsk-ros-pkg/jsk_recognition/pull/1213#issuecomment-147377177 で，
robot_self_filterを入れるならこの2つのPRが反映されていないと通らなくなっています．
同じトラブルを抱える方が多そうだったらどうするのがいいでしょう．

k-okada, 2015-11-30:
> #1213 (comment)

よくわかっていないですが，現状hydro use_deb=trueで通っていないのかな？
https://travis-ci.org/jsk-ros-pkg/jsk_recognition/builds/86870990
は何故通っているのかな．

mmurooka, 2015-11-30:
robot_self_filterが入っていなかったら，
collision_detectorなどのnodeletはコンパイルされないようになっていて，
https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/jsk_pcl_ros/CMakeLists.txt#L409

travisでは，
robot_self_filterはdebでリリースされておらずインストールされていないので，
通っています．

k-okada, 2015-11-30:
では何故このエラーがでるのかな．

wkentaro, 2015-11-30:
古田さんがrobot_self_filterをソースから入れているからだと思います。

k-okada, 2015-11-30:
だったら，ソース上でマージしてコンパイルすればいいですね．

furushchev, 2015-11-30:
返信遅れました。了解です。依存パッケージのマージを待たなくてもマージしていただけるものなのですね。

closed 2015-11-30 by furushchev