jsk-ros-pkg/jsk_recognition #1360: hsi_color_filter.lanchはtfを出力しますか？

ban-masa, 2015-12-02:
```$ roslaunch jsk_pcl_ros hsi_color_filter.launch DEFAULT_NAMESPACE:=/camera/depth_registered INPUT:=points h_min:=40 h_max:=45 s_min:=50```
とすると以下のような結果になるのですが、ロボットモデル上部に表示されているtfは検出されたboxのものなのでしょうか？
[image]

YuOhara, 2015-12-02:
報告ありがとうございます。

モデルの上部に出ている矢印ですが、こちらはTFでなく（TFなら名前が出るはず）、RvizのDisplays/BoundingBoxのshow coordsの結果だと思われます。

本来なら左下の箱の真ん中に出るべきものですが、表示のプログラム(jsk_rviz_plugins)のプログラムにバグがあって、正しく表示されてないのだと思います。おそらく座標変換が間違えているのかと。

YuOhara, 2015-12-02:
ちなみにですが、BoundingBoxのTFを出したいということなら、それも可能で、
https://jsk-recognition.readthedocs.org/en/latest/jsk_pcl_ros/nodes/cluster_point_indices_decomposer.html
~publish_tfというrosparamをtrueにすれば、TFが出ます。

garaemon, 2015-12-05:
再現するbagファイルを送ってください。
トピックは、`/tf`と`//camera/depth_registered/boxes`です

ban-masa, 2015-12-07:
遅くなってしまい申し訳ありません。
/tfと/camera/depth_registered/boxesのbagファイルですがこれでよろしいでしょうか。

[hsi_color_filter_test.zip](https://github.com/jsk-ros-pkg/jsk_recognition/files/54386/hsi_color_filter_test.zip)

garaemon, 2015-12-08:
@ban-mesa 確認してみます

garaemon, 2015-12-08:
@ban-mesa 

たぶんこれで解決されたんじゃないかと思います
https://github.com/jsk-ros-pkg/jsk_visualization/pull/535

[image]

ban-masa, 2015-12-09:
ありがとうございます。
実機で確認したところ、boxの位置に正しく座標系が出るようになりました。

closed 2015-12-09 by garaemon