jsk-ros-pkg/jsk_recognition #1363: [bug] [jsk_pcl_ros] compile error at compiling collision_detector_nodelet.cpp

wkentaro, 2015-12-03:
appears after the release of `ros-indigo-robot-self-filter`. (I found this at travis test on https://github.com/jsk-ros-pkg/jsk_recognition/pull/1361)

assigned to @mmurooka ?
```
[jsk_pcl_ros] In file included from /workspace/ros/ws_jsk_recognition/src/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/self_mask_urdf_robot.h:4:0, 

[jsk_pcl_ros]                  from /workspace/ros/ws_jsk_recognition/src/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/collision_detector.h:50, 

[jsk_pcl_ros]                  from /workspace/ros/ws_jsk_recognition/src/jsk_recognition/jsk_pcl_ros/src/collision_detector_nodelet.cpp:37: 

[jsk_pcl_ros] /opt/ros/indigo/include/robot_self_filter/self_mask.h: In member function 'bool robot_self_filter::SelfMaskUrdfRobot::initKdlConfigure()': 

[jsk_pcl_ros] /opt/ros/indigo/include/robot_self_filter/self_mask.h:643:38: error: 'std::vector<robot_self_filter::SelfMask<pcl::PointXYZ>::SeeLink, std::allocator<robot_self_filter::SelfMask<pcl::PointXYZ>::SeeLink> > robot_self_filter::SelfMask<pcl::PointXYZ>::bodies_' is private 

[jsk_pcl_ros]   std::vector<SeeLink>                bodies_;                    

[jsk_pcl_ros]                                       ^                           

[jsk_pcl_ros] In file included from /workspace/ros/ws_jsk_recognition/src/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/collision_detector.h:50:0, 

[jsk_pcl_ros]                  from /workspace/ros/ws_jsk_recognition/src/jsk_recognition/jsk_pcl_ros/src/collision_detector_nodelet.cpp:37: 

[jsk_pcl_ros] /workspace/ros/ws_jsk_recognition/src/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/self_mask_urdf_robot.h:46:36: error: within this context 

[jsk_pcl_ros]        for(unsigned int i = 0 ; i < bodies_.size() ; i++) {       

[jsk_pcl_ros]                                     ^                             

[jsk_pcl_ros] In file included from /workspace/ros/ws_jsk_recognition/src/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/self_mask_urdf_robot.h:4:0, 

[jsk_pcl_ros]                  from /workspace/ros/ws_jsk_recognition/src/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/collision_detector.h:50, 

[jsk_pcl_ros]                  from /workspace/ros/ws_jsk_recognition/src/jsk_recognition/jsk_pcl_ros/src/collision_detector_nodelet.cpp:37: 

[jsk_pcl_ros] /opt/ros/indigo/include/robot_self_filter/self_mask.h:643:38: error: 'std::vector<robot_self_filter::SelfMask<pcl::PointXYZ>::SeeLink, std::allocator<robot_self_filter::SelfMask<pcl::PointXYZ>::SeeLink> > robot_self_filter::SelfMask<pcl::PointXYZ>::bodies_' is private 

[jsk_pcl_ros]   std::vector<SeeLink>                bodies_;                    

[jsk_pcl_ros]                                       ^                           

[jsk_pcl_ros] In file included from /workspace/ros/ws_jsk_recognition/src/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/collision_detector.h:50:0, 

[jsk_pcl_ros]                  from /workspace/ros/ws_jsk_recognition/src/jsk_recognition/jsk_pcl_ros/src/collision_detector_nodelet.cpp:37: 

[jsk_pcl_ros] /workspace/ros/ws_jsk_recognition/src/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/self_mask_urdf_robot.h:47:28: error: within this context 

[jsk_pcl_ros]          std::string name = bodies_[i].name;                      

[jsk_pcl_ros]                             ^                                     

[jsk_pcl_ros] In file included from /workspace/ros/ws_jsk_recognition/src/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/self_mask_urdf_robot.h:4:0, 

[jsk_pcl_ros]                  from /workspace/ros/ws_jsk_recognition/src/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/collision_detector.h:50, 

[jsk_pcl_ros]                  from /workspace/ros/ws_jsk_recognition/src/jsk_recognition/jsk_pcl_ros/src/collision_detector_nodelet.cpp:37: 

[jsk_pcl_ros] /opt/ros/indigo/include/robot_self_filter/self_mask.h: In member function 'void robot_self_filter::SelfMaskUrdfRobot::updateRobotModel(std::map<std::basic_string<char>, double>&, const Pose&)': 

[jsk_pcl_ros] /opt/ros/indigo/include/robot_self_filter/self_mask.h:643:38: error: 'std::vector<robot_self_filter::SelfMask<pcl::PointXYZ>::SeeLink, std::allocator<robot_self_filter::SelfMask<pcl::PointXYZ>::SeeLink> > robot_self_filter::SelfMask<pcl::PointXYZ>::bodies_' is private 

[jsk_pcl_ros]   std::vector<SeeLink>                bodies_;                    

[jsk_pcl_ros]                                       ^                           

[jsk_pcl_ros] In file included from /workspace/ros/ws_jsk_recognition/src/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/collision_detector.h:50:0, 

[jsk_pcl_ros]                  from /workspace/ros/ws_jsk_recognition/src/jsk_recognition/jsk_pcl_ros/src/collision_detector_nodelet.cpp:37: 

[jsk_pcl_ros] /workspace/ros/ws_jsk_recognition/src/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/self_mask_urdf_robot.h:89:36: error: within this context 

[... 700 lines cut ...]

Build summary:                                                                  

 Successful checkerboard_detector   Successful dynamic_tf_publisher             

 Successful image_view2             Successful imagesift                        

 Successful jsk_data                Successful jsk_network_tools                

 Successful jsk_perception          Successful jsk_recognition_msgs             

 Successful jsk_recognition_utils   Successful jsk_tilt_laser                   

 Successful jsk_tools               Successful jsk_topic_tools                  

 Successful multi_map_server        Successful resized_image_transport          

 Successful virtual_force_publisher                                             

                                                                                

Failed packages:                                                                

 Failed     jsk_pcl_ros                                                         
```

furushchev, 2015-12-03:
related: https://github.com/jsk-ros-pkg/jsk_recognition/issues/1356

wkentaro, 2015-12-03:
I appreciate if this issue is fixed as soon as possible.

garaemon, 2015-12-03:
https://github.com/PR2/robot_self_filter/issues/9

garaemon, 2015-12-04:
I understood. I will fix it
.

wkentaro, 2015-12-04:
now released.
[image]

garaemon, 2015-12-10:
solved

closed 2015-12-10 by garaemon