jsk-ros-pkg/jsk_recognition #1410: Cannot import bounding_rect_of_mask

garaemon, 2015-12-15:
@wkentaro 

http://jenkins.jsk.imi.i.u-tokyo.ac.jp:8080/job/trusty-travis-jsk-ros-pkg-jsk_recognition-indigo-deb-true-true/445/console

```
[jsk_perception] Traceback (most recent call last):                             
[jsk_perception]   File "/workspace/ros/ws_jsk_recognition/src/jsk_recognition/jsk_perception/node_scripts/label_image_decomposer.py", line 7, in <module> 
[jsk_perception]     from jsk_recognition_utils import bounding_rect_of_mask    
[jsk_perception]   File "/workspace/ros/ws_jsk_recognition/devel/lib/python2.7/dist-packages/jsk_recognition_utils/__init__.py", line 35, in <module> 
[jsk_perception]     exec(__fh.read())                                          
[jsk_perception]   File "<string>", line 6, in <module>                         
[jsk_perception] Traceback (most recent call last):                             
[jsk_perception]   File "/workspace/ros/ws_jsk_recognition/src/jsk_recognition/jsk_perception/node_scripts/label_image_decomposer.py", line 7, in <module> 
[jsk_perception]     from jsk_recognition_utils import bounding_rect_of_mask    
[jsk_perception]   File "/workspace/ros/ws_jsk_recognition/devel/lib/python2.7/dist-packages/jsk_recognition_utils/__init__.py", line 35, in <module> 
[jsk_perception]     exec(__fh.read())                                          
[jsk_perception]   File "<string>", line 6, in <module>                         
[jsk_perception]   File "/workspace/ros/ws_jsk_recognition/src/jsk_recognition/jsk_recognition_utils/python/jsk_recognition_utils/visualize.py", line 7, in <module> 
[jsk_perception]     import matplotlib.pyplot as plt                            
[jsk_perception]   File "/workspace/ros/ws_jsk_recognition/src/jsk_recognition/jsk_recognition_utils/python/jsk_recognition_utils/visualize.py", line 7, in <module> 
[jsk_perception]   File "/usr/local/lib/python2.7/dist-packages/matplotlib/pyplot.py", line 29, in <module> 
[jsk_perception]     import matplotlib.pyplot as plt                            
[jsk_perception]   File "/usr/local/lib/python2.7/dist-packages/matplotlib/pyplot.py", line 29, in <module> 
[jsk_perception]     import matplotlib.colorbar                                 
[jsk_perception]   File "/usr/local/lib/python2.7/dist-packages/matplotlib/colorbar.py", line 34, in <module> 
[jsk_perception]     import matplotlib.colorbar                                 
[jsk_perception]     import matplotlib.collections as collections               
[jsk_perception]   File "/usr/local/lib/python2.7/dist-packages/matplotlib/colorbar.py", line 34, in <module> 
[jsk_perception]   File "/usr/local/lib/python2.7/dist-packages/matplotlib/collections.py", line 27, in <module> 
[jsk_perception]     import matplotlib.collections as collections               
[jsk_perception]     import matplotlib.backend_bases as backend_bases           
[jsk_perception]   File "/usr/local/lib/python2.7/dist-packages/matplotlib/collections.py", line 27, in <module> 
[jsk_perception]   File "/usr/local/lib/python2.7/dist-packages/matplotlib/backend_bases.py", line 62, in <module> 
[jsk_perception]     import matplotlib.backend_bases as backend_bases           
[jsk_perception]       File "/usr/local/lib/python2.7/dist-packages/matplotlib/backend_bases.py", line 62, in <module> 
[jsk_perception] import matplotlib.textpath as textpath                         
[jsk_perception]   File "/usr/local/lib/python2.7/dist-packages/matplotlib/textpath.py", line 15, in <module> 
[jsk_perception]     import matplotlib.textpath as textpath                     
[jsk_perception]       File "/usr/local/lib/python2.7/dist-packages/matplotlib/textpath.py", line 15, in <module> 
[jsk_perception] import matplotlib.font_manager as font_manager                 
[jsk_perception]   File "/usr/local/lib/python2.7/dist-packages/matplotlib/font_manager.py", line 1420, in <module> 
[jsk_perception]     import matplotlib.font_manager as font_manager             
[jsk_perception]   File "/usr/local/lib/python2.7/dist-packages/matplotlib/font_manager.py", line 1420, in <module> 
[jsk_perception]     _rebuild()                                                 
[jsk_perception]   File "/usr/local/lib/python2.7/dist-packages/matplotlib/font_manager.py", line 1405, in _rebuild 
[jsk_perception]     _rebuild()                                                 
[jsk_perception]   File "/usr/local/lib/python2.7/dist-packages/matplotlib/font_manager.py", line 1405, in _rebuild 
[jsk_perception]         fontManager = FontManager()                            
[jsk_perception]   File "/usr/local/lib/python2.7/dist-packages/matplotlib/font_manager.py", line 1043, in __init__ 
[jsk_perception] fontManager = FontManager()                                    
[jsk_perception]   File "/usr/local/lib/python2.7/dist-packages/matplotlib/font_manager.py", line 1043, in __init__ 
[jsk_perception]     self.ttffiles = findSystemFonts(paths) + findSystemFonts() 
[jsk_perception]   File "/usr/local/lib/python2.7/dist-packages/matplotlib/font_manager.py", line 323, in findSystemFonts 
[jsk_perception]         for f in get_fontconfig_fonts(fontext):                
[jsk_perception]   File "/usr/local/lib/python2.7/dist-packages/matplotlib/font_manager.py", line 276, in get_fontconfig_fonts 
[jsk_perception] self.ttffiles = findSystemFonts(paths) + findSystemFonts()     
[jsk_perception]   File "/usr/local/lib/python2.7/dist-packages/matplotlib/font_manager.py", line 323, in findSystemFonts 
[jsk_perception]     output = pipe.communicate()[0]                             
[jsk_perception]   File "/usr/lib/python2.7/subprocess.py", line 799, in communicate 
[jsk_perception]     for f in get_fontconfig_fonts(fontext):                    
[jsk_perception]   File "/usr/local/lib/python2.7/dist-packages/matplotlib/font_manager.py", line 276, in get_fontconfig_fonts 
[jsk_perception]         return self._communicate(input)                        
[jsk_perception]   File "/usr/lib/python2.7/subprocess.py", line 1401, in _communicate 
[jsk_perception] output = pipe.communicate()[0]                                 
[jsk_perception]   File "/usr/lib/python2.7/subprocess.py", line 799, in communicate 
[jsk_perception]     return self._communicate(input)                            
[jsk_perception]     stdout, stderr = self._communicate_with_poll(input)        
[jsk_perception]   File "/usr/lib/python2.7/subprocess.py", line 1401, in _communicate 
[jsk_perception]   File "/usr/lib/python2.7/subprocess.py", line 1455, in _communicate_with_poll 
[jsk_perception]         ready = poller.poll()                                  
[jsk_perception] KeyboardInterruptstdout, stderr = self._communicate_with_poll(input) 
[jsk_perception]                                                                
[jsk_perception]   File "/usr/lib/python2.7/subprocess.py", line 1455, in _communicate_with_poll 
[jsk_perception]     ready = poller.poll()                                      
[jsk_perception] KeyboardInterrupt                                              
```

garaemon, 2015-12-16:
@wkentaro Could you rewrite it without matplotlib, for example use opencv instead? 

I think it is because of GUI thread of matplotlib.

We can set proper 'backends' of matplotlib to work it but it is a little bit tricky.

wkentaro, 2015-12-16:
I hope below fixes this problem
https://github.com/jsk-ros-pkg/jsk_recognition/pull/1412/files

garaemon, 2015-12-16:
fixed

closed 2015-12-16 by garaemon