jsk-ros-pkg/jsk_recognition #1413: jsk_pcl_ros test data is too large (~200M)

garaemon, 2015-12-16:
Decrease the size.

garaemon, 2015-12-16:
You can see long log to download it.
http://jenkins.jsk.imi.i.u-tokyo.ac.jp:8080/job/trusty-travis-jsk-ros-pkg-jsk_recognition-jade-deb-false-true/499/console

garaemon, 2015-12-16:
If we can use wget instead of gdown, we can use --timestamping option to avoid re-download the file (we can cache it somewhere)

wkentaro, 2015-12-16:
gdown uses wget inside.
https://github.com/wkentaro/gdown/blob/master/gdown/__init__.py#L14

wkentaro, 2015-12-16:
gdown is necessary to download large file from good drive.
just using wget command does not work.

wkentaro, 2015-12-17:
@garaemon 
I have done implementation to identify if downloaded file is latest by checking md5 hash. https://github.com/jsk-ros-pkg/jsk_recognition/pull/1419
Do you have any idea how to cache it?

wkentaro, 2016-05-07:
This is fixed via 
- https://github.com/jsk-ros-pkg/jsk_recognition/commit/cc275d4bfbdabd515b1e794e5eee96ecb85ab6a0
- https://github.com/jsk-ros-pkg/jsk_travis/pull/220

closed 2016-05-07 by wkentaro