jsk-ros-pkg/jsk_recognition #1421: Release new version

garaemon, 2015-12-17:
Don't know why but failed on ros buildfarm.
http://jenkins.ros.org/view/IbinS64/job/ros-indigo-jsk-pcl-ros_binarydeb_saucy_amd64/100/

k-okada, 2015-12-17:
```

Build timed out (after 120 minutes). Marking the build as aborted.

```

garaemon, 2015-12-17:
too large project...?

garaemon, 2015-12-17:
Because of change of jsk_common, ccache misses cache?

garaemon, 2015-12-17:
Anyway I released 0.3.10

garaemon, 2015-12-17:
Idea
1. split into some packages. jsk_pcl_ros is too large. (But I like this style...)
2. Remove unused programs
3. Concatenate cpp files into one big cpp files

garaemon, 2015-12-17:
4. Send to upstream (best)

In order to send to upstream, we need to move to indigo at least...

garaemon, 2015-12-17:
I found compiler are different.

When it exceeds 120 minutes, `/usr/lib/ccache/x86_64-linux-gnu-g++` is used.
If it run in 120 minutes, `/usr/lib/ccache/c++` is used.

garaemon, 2015-12-17:
Compiler version is updated from 4.8.1 to 4.8.4

garaemon, 2015-12-17:
It is not because of difference of compiler.

http://jenkins.ros.org/view/IbinS64/job/ros-indigo-jsk-pcl-ros_binarydeb_saucy_amd64/97/consoleFull
Here jenkins uses 4.8.1 but it fails.

garaemon, 2015-12-17:
Capability of node are different?

http://jenkins.ros.org/computer/

`_sync` node looks inferior than other nodes

garaemon, 2015-12-17:
I reported
https://github.com/ros-infrastructure/buildfarm/issues/220

garaemon, 2015-12-18:
We need to separate package into few packages...

> They are the same machine. The sync one is slightly older and is special in that it is the only one allowed to run sync jobs. There's always going to be some variability on the build time. This package is very clearly near the edge. Things that can effect the built time are other hosts on the VM, bandwidth available, and how much overlap is in the caches from recent builds.

garaemon, 2015-12-19:
Failed to release. 

http://jenkins.jsk.imi.i.u-tokyo.ac.jp:8080/job/bloom_release/5889/console

```
14:51:49 release indigo/jsk_recognition -> 0.3.11
14:51:49 run bloom for  jsk_recognition
14:51:53 Cloning into '/tmp/tmp0hqSOx'...
14:52:07 origin	https://github.com/tork-a/jsk_recognition-release.git (fetch)
14:52:07 origin	https://github.com/tork-a/jsk_recognition-release.git (push)
14:52:08 fatal: could not read Username for 'https://github.com': No such device or address
14:52:08 ROS Distro index file associate with commit 'f4955b88b2b2dd9655889ce81002aa639b1303db'
14:52:08 New ROS Distro index url: 'https://raw.githubusercontent.com/ros/rosdistro/f4955b88b2b2dd9655889ce81002aa639b1303db/index.yaml'
14:52:08 ==> Fetching 'jsk_recognition' repository from 'https://github.com/tork-a/jsk_recognition-release.git'
14:52:08 ==> Testing for push permission on release repository
14:52:08 ==> git remote -v
14:52:08 ==> git push --dry-run
14:52:08 Cannot push to remote release repository.
```

garaemon, 2015-12-19:
@k-okada Could you fix this issue?

k-okada, 2015-12-20:
```
Running 'rosdep update'...
Could not resolve rosdep key 'ml_classifiers'
Failed to resolve ml_classifiers on ubuntu:trusty with: Error running
generator: Failed to resolve rosdep key 'ml_classifiers', aborting.
ml_classifiers is depended on by these packages: ['jsk_pcl_ros_utils']
<== Failed
Could not resolve rosdep key 'ml_classifiers'
Failed to resolve ml_classifiers on ubuntu:utopic with: Error running
generator: Failed to resolve rosdep key 'ml_classifiers', aborting.
ml_classifiers is depended on by these packages: ['jsk_pcl_ros_utils']
<== Failed
Could not resolve rosdep key 'ml_classifiers'
Failed to resolve ml_classifiers on ubuntu:vivid with: Error running
generator: Failed to resolve rosdep key 'ml_classifiers', aborting.
ml_classifiers is depended on by these packages: ['jsk_pcl_ros_utils']
<== Failed
Could not resolve rosdep key 'sklearn'
Failed to resolve sklearn on ubuntu:trusty with: Error running generator:
Failed to resolve rosdep key 'sklearn', aborting.
sklearn is depended on by these packages: ['jsk_pcl_ros_utils']
<== Failed
Could not resolve rosdep key 'sklearn'
Failed to resolve sklearn on ubuntu:utopic with: Error running generator:
Failed to resolve rosdep key 'sklearn', aborting.
sklearn is depended on by these packages: ['jsk_pcl_ros_utils']
<== Failed
Could not resolve rosdep key 'sklearn'
Failed to resolve sklearn on ubuntu:vivid with: Error running generator:
Failed to resolve rosdep key 'sklearn', aborting.
sklearn is depended on by these packages: ['jsk_pcl_ros_utils']
<== Failed

```
do we remove them from 3rdparty? i think it was not a good idea to remove
something within same rosdistro....

k-okada, 2015-12-20:
sorry, it was jade, so they are not released on them, I can fix this.

garaemon, 2015-12-20:
finally, jsk_recognition deb will be back to normal in a day.

[image]

garaemon, 2015-12-21:
Finally, done

closed 2015-12-21 by garaemon