jsk-ros-pkg/jsk_recognition #1423: Failed to release jade from bloom_release task

garaemon, 2015-12-17:
```
15:23:55 Executing release track 'jade'
15:23:55 ==> bloom-export-upstream /tmp/tmp4uCG6g/upstream git --tag 0.3.10 --display-uri https://github.com/jsk-ros-pkg/jsk_recognition --name jsk_recognition --output-dir /tmp/tmpqmple2
15:23:55 
15:23:55 ==> git-bloom-import-upstream /tmp/tmpqmple2/jsk_recognition-0.3.10.tar.gz  --release-version 0.3.10 --replace
15:23:55 
15:23:55 ==> git-bloom-generate -y rosrelease jade --source upstream -i 0
15:23:55 
15:23:55 ==> git-bloom-generate -y rosdebian --prefix release/jade jade -i 0
15:23:55 <== Error running command '['/usr/bin/git-bloom-generate', '-y', 'rosdebian', '--prefix', 'release/jade', 'jade', '-i', '0']'
15:23:55 ROS Distro index file associate with commit 'd76d361ceacf83a9e02207208204e9ebe4676255'
15:23:55 New ROS Distro index url: 'https://raw.githubusercontent.com/ros/rosdistro/d76d361ceacf83a9e02207208204e9ebe4676255/index.yaml'
15:23:55 ==> Fetching 'jsk_recognition' repository from 'https://github.com/tork-a/jsk_recognition-release.git'
15:23:55 ==> Testing for push permission on release repository
15:23:55 ==> git remote -v
15:23:55 ==> git push --dry-run
15:23:55 ==> Releasing 'jsk_recognition' using release track 'jade'
15:23:55 ==> git-bloom-release jade
15:23:55 Release failed, exiting.
```

k-okada, 2015-12-17:
robot_self_filter is not released on jade, and I created issue for that,
but I'm afraid this just assign  new job to @wkentaro and @garaemon.....
https://github.com/PR2/robot_self_filter/issues/11

garaemon, 2015-12-17:
Wait for next shadow-fixed release

closed 2015-12-19 by garaemon