jsk-ros-pkg/jsk_recognition #1444: Is there way to change dynamic parameter with timestamp?

wkentaro, 2015-12-20:
I'd like to use dynamic_reconfigure for recognition with timestamp.
Currently, the config callback does not see the timestamp https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/jsk_pcl_ros/src/organized_multi_plane_segmentation_nodelet.cpp#L172
and dynamic_reconfigure seems to be not designed to do so.

Any idea to fix this issue?
I think there are two solutions:

1. Change the `configCallback` to make the node to see the timestamp.
1. Change `dynamic_reconfigure` to serve another topic with timstamp (like `dynamic_reconfigure/ReconfigureStamped`). (currently, dynamic_reconfigure serves `set_parameters` service not topic)

wkentaro, 2015-12-20:
The first option may break current system and the nodes' behavior will change, so I think it's not good idea.

wkentaro, 2015-12-20:
Could you please give me some advice? @garaemon @k-okada

garaemon, 2015-12-20:
It depends on your purpose.

Whey you need timestamp on dynamic_reconfigure?

wkentaro, 2015-12-20:
> It depends on your purpose.

Yes, so I don't want to break current system.

> Whey you need timestamp on dynamic_reconfigure?

To change the parameter from **now** (the time I changed the parameter), not the time the `configCallback` is called in each recognition node.

garaemon, 2015-12-20:
> To change the parameter from now (the time I changed the parameter), not the time the configCallback is called in each recognition node.

hmm, it sounds difficult.
Your problem is latency between rqt_dynamic_reconfigure event callback and service call?

wkentaro, 2015-12-20:
> Your problem is latency between rqt_dynamic_reconfigure event callback and service call?

Maybe no, the problem is a node cannot know that what data with what timestamp other node processing.

What I want to do is to specify the timestamp and parameter for a processing, this can be significant in dynamic environment.
At this time, I'm in not so dynamic environment so maybe current dynamic reconfigure system does work.