jsk-ros-pkg/jsk_recognition #1448: [jsk_pcl_ros_utils] fatal error: jsk_recognition_msgs/SetPointCloud2.h: No such file or directory

wkentaro, 2015-12-20:
```
[jsk_pcl_ros_utils] In file included from /home/wkentaro/ros/indigo/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros_utils/src/pointcloud_to_stl_nodelet.cpp:36:0: 
[jsk_pcl_ros_utils] /home/wkentaro/ros/indigo/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros_utils/include/jsk_pcl_ros_utils/pointcloud_to_stl.h:44:49: fatal error: jsk_recognition_msgs/SetPointCloud2.h: No such file or directory 
[jsk_pcl_ros_utils]  #include <jsk_recognition_msgs/SetPointCloud2.h>                          
[jsk_pcl_ros_utils]                                                  ^                         
[jsk_pcl_ros_utils] compilation terminated.                                                    
[jsk_pcl_ros_utils] make[2]: *** [CMakeFiles/jsk_pcl_ros_utils.dir/src/pointcloud_to_stl_nodelet.cpp.o] Error 1 
[jsk_pcl_ros_utils] make[2]: *** Waiting for unfinished jobs....                               
[jsk_pcl_ros_utils] make[1]: *** [CMakeFiles/jsk_pcl_ros_utils.dir/all] Error 2                
[jsk_pcl_ros_utils] make: *** [all] Error 2  
```

garaemon, 2015-12-21:
Latest master should be fixed

wkentaro, 2015-12-21:
thanks, now it seems to be fixed.

closed 2015-12-21 by wkentaro