jsk-ros-pkg/jsk_recognition #1482: 60秒以内に同じlaunchを再び上げるとnodeletが上がらない

eisoku9618, 2016-01-01:
1. 以下のようなjsk_pcl_rosのnodeletを使ったlaunchフィアルを一回上げる（ちゃんと上がる）
1. C-cで止める
1. その後 n秒 待って再び同じlaunchファイルを上げる

ということを良くやるかと思うのですが，n > 60秒 のときは問題なく上がるのですが，n < 60秒のときはnodeletが起動できないです．

```txt
[ERROR] [1451621026.401613259, 4754.982000000]: Failed to load nodelet [/transform_cloud] of type [jsk_pcl/TfTransformCloud] even after refreshing the cache: MultiLibraryClassLoader: Could not create object of class type jsk_pcl_ros::TfTransformCloud as no factory exists for it. Make sure that the library exists and was explicitly loaded through MultiLibraryClassLoader::loadLibrary()
[ERROR] [1451621026.401636862, 4754.982000000]: The error before refreshing the cache was: MultiLibraryClassLoader: Could not create object of class type jsk_pcl_ros::TfTransformCloud as no factory exists for it. Make sure that the library exists and was explicitly loaded through MultiLibraryClassLoader::loadLibrary()
[ERROR] [1451621026.401648487, 4754.982000000]: [int main(int, char**)] Failed to load nodelet [/transform_cloud -- jsk_pcl/TfTransformCloud]
```

roscoreを含め一旦全て落としても，やはり前回上げた時から60秒間待たないとnodeletが上がらないです．

これはそういうものなのでしょうか？それとも僕のlaunchの書き方が悪いでしょうか？
（関係あるかわかりませんが，pcl/PassThroughを2つ上げるものは待たずとも正常に上がりました．）

---

以下確認用スクリプト

---

以下のスクリプトで確認したところ，
１分間以上間隔を開けると大丈夫なのですが，
間隔が55秒だと10回やって10回ともダメでした．

- /tmp/test.launch ： jsk_pcl/TfTransformCloud と pcl/PassThrough を上げる

```xml
<launch>
  <node pkg="jsk_topic_tools" type="standalone_complexed_nodelet" name="aiueo" output="screen">
    <rosparam subst_value="true">                                                                                                                                                                 
    nodelets:                                                                                                                                                                                     
      - name: transform_cloud                                                                                                                                                                     
        type: jsk_pcl/TfTransformCloud                                                                                                                                                            
        remappings:                                                                                                                                                                               
          - from: ~input                                                                                                                                                                          
            to: /multisense/organized_image_points2_color                                                                                                                                         
      - name: laser_x_filter                                                                                                                                                                      
        type: pcl/PassThrough                                                                                                                                                                     
        remappings:                                                                                                                                                                               
          - from: ~input                                                                                                                                                                          
            to: transform_cloud/output                                                                                                                                                            
          - from: /tf                                                                                                                                                                             
            to: /tf_null                                                                                                                                                                          
    </rosparam>
  </node>
  <group ns="laser_x_filter">
    <rosparam>                                                                                                                                                                                    
      filter_field_name: x                                                                                                                                                                        
      filter_limit_min: -5.0                                                                                                                                                                      
      filter_limit_max: 1000.0                                                                                                                                                                    
    </rosparam>
  </group>
  <group ns="transform_cloud">
    <rosparam>                                                                                                                                                                                    
      target_frame_id: odom                                                                                                                                                                       
    </rosparam>
  </group>
</launch>
```

- /tmp/hoge.sh : nodeletの起動に成功するまでC-cと55秒間のsleepとroslaunchを10回繰り返すscript @ tmux

```sh
for i in $(seq 0 9)
do
    tmux send-keys "C-c" C-m
    echo "try "$i
    tmux send-keys "roslaunch /tmp/test.launch 2>&1 | tee /tmp/hoge" C-m
    sleep 2
    num=`grep -c "Failed to load nodelet" /tmp/hoge`
    if [ $num -gt 0 ]
    then
        echo "failed to load nodlet"
        sleep 55s
    else
        echo "success!!"
        break
    fi
done
```

garaemon, 2016-01-01:
おかしいですね。

1. roslaunchがちゃんと死んでいるか
2. pluginlibとlibrospackが怪しいのでソースから入れてデバッグする

k-okada, 2016-01-02:
60秒の間に
```
lsof | grep ros
```
したらなにか分かるかな．

eisoku9618, 2016-01-04:
ありがとうございます．

> roslaunchがちゃんと死んでいるか

``ps aux | grep roslaunch``では出てきませんでしたので，死んでいそうです．

```
lsof | grep ros
```

- nodeletがうまく上がっているとき： https://gist.githubusercontent.com/eisoku9618/f3ad45914ec1639a2964/raw/c5ebc936a723a425f508f147e8065518a24bb8ba/success_lsof_grep_ros.log
- 失敗しているとき： https://gist.githubusercontent.com/eisoku9618/f3ad45914ec1639a2964/raw/c5ebc936a723a425f508f147e8065518a24bb8ba/fail_lsof_grep_ros.log

や

```
lsof -c standalone
```

- nodeletがうまく上がっているとき： https://gist.githubusercontent.com/eisoku9618/f3ad45914ec1639a2964/raw/c5ebc936a723a425f508f147e8065518a24bb8ba/success_lsof_-c_standalone.log
- 失敗しているとき： https://gist.githubusercontent.com/eisoku9618/f3ad45914ec1639a2964/raw/c5ebc936a723a425f508f147e8065518a24bb8ba/fail_lsof_-c_standalone.log

は僕レベルで気づけるこれと言った違いはなさそうでした．

eisoku9618, 2016-01-04:
補足ですが，standalone_complexed_nodeletを使わずに以下のようにしても同じ現象が起きますので，standalone_complexed_nodeletには問題なさそうです．

また，同じjsk_pclのnodeletでも，例えばjsk_pcl/TiltLaserListenerは問題なく動きます．

```xml
<launch>
  <node pkg="nodelet" type="nodelet" args="manager" name="aiueo" />

  <node pkg="nodelet" type="nodelet" name="transform_cloud"
        args="load jsk_pcl/TfTransformCloud aiueo">
    <remap from="~input" to="/multisense/organized_image_points2_color" />
    <rosparam>
      target_frame_id: odom
    </rosparam>
  </node>

  <node pkg="nodelet" type="nodelet" name="laser_x_filter"
        args="load pcl/PassThrough aiueo">
    <remap from="~input" to="transform_cloud/output" />
    <remap from="/tf" to="/tf_null" />
    <rosparam>
      filter_field_name: x
      filter_limit_min: -5.0
      filter_limit_max: 1000.0
    </rosparam>
  </node>
</launch>
```

garaemon, 2016-01-04:
pluginlibの中のdlopenで死んでいるのか，dlsymで死んでいるのか，それともコンストラクタの呼び出しで死んでいるのかonInitの呼び出しで死んでいるのかを見極めるとデバッグできる気がします．

TfTransformCloudはシングルトン作ったりしているので，ちょっと怪しいと言えば怪しい．

loadじゃなくて，standaloneだと動く?

eisoku9618, 2016-01-04:
> pluginlibの中のdlopenで死んでいるのか，dlsymで死んでいるのか，それともコンストラクタの呼び出しで死んでいるのかonInitの呼び出しで死んでいるのかを見極めるとデバッグできる気がします．

pluginlibをソースから入れるのがまだ出来ていませんが，
少なくとも，TfTransformCloud（とその親クラスのDiagnosticNodelet）のコンストラクタやonInitにprint文入れても，nodeletがうまく起動しない時は表示されませんでした．

> loadじゃなくて，standaloneだと動く?

以下のようにstandaloneにしてもダメでした．

```xml
<launch>
  <node pkg="nodelet" type="nodelet" name="transform_cloud"
        args="standalone jsk_pcl/TfTransformCloud">
    <remap from="~input" to="/multisense/organized_image_points2_color" />
    <rosparam>                                                                                                                                                                                    
      target_frame_id: odom                                                                                                                                                                       
    </rosparam>
  </node>
</launch>
```

jvrcのときには動いていたはず，，，と思っていたら，

jsk_pcl_rosのCMakeに

``jsk_pcl_nodelet``

と書かれている方は動くけれども

``jsk_pcl_util_nodelet_deprecated``

と書かれている方は動かない気がしてきました．
以下のようなlaunchで``jsk_pcl_util_nodelet_deprecated``のものを6個くらい確認して，全部nodeletが死んでしまいました．

```launch
<launch>
  <node pkg="nodelet" type="nodelet" name="aaa"
        args="standalone jsk_pcl/PolygonArrayDistanceLikelihood">
  </node>
</launch>
```

YuOhara, 2016-10-15:
治ってると思われます．
```
$ cat test.launch
<launch>
  <node pkg="nodelet" type="nodelet" name="transform_cloud"
        args="standalone jsk_pcl/TfTransformCloud">
    <remap from="~input" to="/multisense/organized_image_points2_color" />
    <rosparam>                                                                                                                                                                                    
      target_frame_id: odom                                                                                                                                                                       
    </rosparam>
  </node>
</launch>
leus@him ~ (ROBOT=HRP2JSKNTS)
$ roslaunch test.launch 
... logging to /home/leus/.ros/log/3be0e7e6-92d6-11e6-adfe-d050990a3345/roslaunch-him-17631.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
WARNING: disk usage in log directory [/home/leus/.ros/log] is over 1GB.
It's recommended that you use the 'rosclean' command.

started roslaunch server http://192.168.96.186:44258/

SUMMARY
========

PARAMETERS
 * /rosdistro: indigo
 * /rosversion: 1.11.20
 * /transform_cloud/target_frame_id: odom

NODES
  /
    transform_cloud (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [17643]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 3be0e7e6-92d6-11e6-adfe-d050990a3345
process[rosout-1]: started with pid [17656]
started core service [/rosout]
process[transform_cloud-2]: started with pid [17675]

```
上のコードで，特にエラーは出ませんでした

k-okada, 2016-10-16:
どこで治ったんでしょうね．

YuOhara, 2016-10-18:
#1499　の前後だと思います。
60秒sleepしないとあがらない、という当初の問題についても、
 #1499で治ってることを確かめました。
前後で比較して、#1499直前は失敗して、1499後は成功しているログです。

（そもそも
問題が二つあって、
* 60秒いないにjsk_pcl_nodeletを含むlaunchをあげるとしぬ
* jsk_pcl_util_nodelet_deprecatedなどはそもそも上がらない。

でした。
#1499 は、後者の問題を解決するものでしたが、
前者の問題も、治っていました。）

なぜこの問題が起こったかについては、 #1499 で考察していただいています。

```
leus@leus-HP-Z620-Workstation:~/ros/indigo/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros$ git checkout 4935720
M	.travis
M	jsk_pcl_ros/CMakeLists.txt
Previous HEAD position was b237670... Merge pull request #1496 from garaemon/support-inlier
HEAD is now at 4935720... [jsk_pcl_ros_utils] Use jsk_pcl_utils prefix instead of jsk_pcl to prevent namespace conflict with jsk_pcl nodelets
leus@leus-HP-Z620-Workstation:~/ros/indigo/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros$ catkin bt | grep fail
leus@leus-HP-Z620-Workstation:~/ros/indigo/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros$ roslaunch /tmp/test.launch | grep fail
WARNING: disk usage in log directory [/home/leus/.ros/log] is over 1GB.
It's recommended that you use the 'rosclean' command.
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
^Cclose failed in file object destructor:
sys.excepthook is missing
lost sys.stderr

leus@leus-HP-Z620-Workstation:~/ros/indigo/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros$ roslaunch /tmp/test.launch | grep fail
WARNING: disk usage in log directory [/home/leus/.ros/log] is over 1GB.
It's recommended that you use the 'rosclean' command.
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
^Cclose failed in file object destructor:
sys.excepthook is missing
lost sys.stderr

leus@leus-HP-Z620-Workstation:~/ros/indigo/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros$ roslaunch /tmp/test.launch | grep fail
WARNING: disk usage in log directory [/home/leus/.ros/log] is over 1GB.
It's recommended that you use the 'rosclean' command.
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
^Cclose failed in file object destructor:
sys.excepthook is missing
lost sys.stderr

leus@leus-HP-Z620-Workstation:~/ros/indigo/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros$ git checkout HEAD^^
M	.travis
M	jsk_pcl_ros/CMakeLists.txt
Previous HEAD position was 4935720... [jsk_pcl_ros_utils] Use jsk_pcl_utils prefix instead of jsk_pcl to prevent namespace conflict with jsk_pcl nodelets
HEAD is now at b237670... Merge pull request #1496 from garaemon/support-inlier
leus@leus-HP-Z620-Workstation:~/ros/indigo/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros$ catkin bt | grep fail
leus@leus-HP-Z620-Workstation:~/ros/indigo/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros$ roslaunch /tmp/test.launch | grep fail
WARNING: disk usage in log directory [/home/leus/.ros/log] is over 1GB.
It's recommended that you use the 'rosclean' command.
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
[ERROR] [1476804316.151839911]: Failed to load nodelet [/transform_cloud] of type [jsk_pcl/TfTransformCloud] even after refreshing the cache: MultiLibraryClassLoader: Could not create object of class type jsk_pcl_ros::TfTransformCloud as no factory exists for it. Make sure that the library exists and was explicitly loaded through MultiLibraryClassLoader::loadLibrary()
[ERROR] [1476804316.151911744]: The error before refreshing the cache was: MultiLibraryClassLoader: Could not create object of class type jsk_pcl_ros::TfTransformCloud as no factory exists for it. Make sure that the library exists and was explicitly loaded through MultiLibraryClassLoader::loadLibrary()
[ERROR] [1476804316.151964887]: [main] Failed to load nodelet [/transform_cloud -- jsk_pcl/TfTransformCloud]
^Cclose failed in file object destructor:
sys.excepthook is missing
lost sys.stderr

leus@leus-HP-Z620-Workstation:~/ros/indigo/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros$ roslaunch /tmp/test.launch | grep fail
WARNING: disk usage in log directory [/home/leus/.ros/log] is over 1GB.
It's recommended that you use the 'rosclean' command.
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
[ERROR] [1476804321.703957092]: Failed to load nodelet [/transform_cloud] of type [jsk_pcl/TfTransformCloud] even after refreshing the cache: MultiLibraryClassLoader: Could not create object of class type jsk_pcl_ros::TfTransformCloud as no factory exists for it. Make sure that the library exists and was explicitly loaded through MultiLibraryClassLoader::loadLibrary()
[ERROR] [1476804321.704025240]: The error before refreshing the cache was: MultiLibraryClassLoader: Could not create object of class type jsk_pcl_ros::TfTransformCloud as no factory exists for it. Make sure that the library exists and was explicitly loaded through MultiLibraryClassLoader::loadLibrary()
[... 14 lines cut ...]
lost sys.stderr

leus@leus-HP-Z620-Workstation:~/ros/indigo/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros$ sleep 60
[... cut ...]
[ERROR] [1476804406.358794180]: The error before refreshing the cache was: MultiLibraryClassLoader: Could not create object of class type jsk_pcl_ros::TfTransformCloud as no factory exists for it. Make sure that the library exists and was explicitly loaded through MultiLibraryClassLoader::loadLibrary()
[ERROR] [1476804406.358825222]: [main] Failed to load nodelet [/transform_cloud -- jsk_pcl/TfTransformCloud]
^Cclose failed in file object destructor:
sys.excepthook is missing
lost sys.stderr

leus@leus-HP-Z620-Workstation:~/ros/indigo/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros$ 
```

k-okada, 2016-10-19:
なるほど
これは重要情報かも知れないですね

YoheiKakiuchi, 2016-10-19:
そもそもの問題の原因に関わるかわからないのですが、
https://github.com/jsk-ros-pkg/jsk_recognition/pull/1499
を見て思ったのは、同じバイナリを指すと思われる記述が jsk_pcl_rosとjsk_pcl_ros_utilsのnodelets.xml
２箇所にされているけれども、これはなにか理由があるんだろうか？

https://github.com/jsk-ros-pkg/jsk_recognition/pull/1499 以前は、それが同じ名前で登録されているように見える。（たぶん片方の記述が間違ったクラスを指している？？）
https://github.com/jsk-ros-pkg/jsk_recognition/pull/1499 は正しいクラスを指す、名前の異なるnodeletが２つあるように見える (たとえばPointCloudToSTLなど）

orikuma, 2016-10-19:
>そもそもの問題の原因に関わるかわからないのですが、
>#1499 <https://github.com/jsk-ros-pkg/jsk_recognition/pull/1499>
>を見て思ったのは、同じバイナリを指すと思われる記述が jsk_pcl_rosとjsk_pcl_ros_utilsのnodelets.xml
>２箇所にされているけれども、これはなにか理由があるんだろうか？

これは後方互換性のためで, もともとjsk_pcl_rosにあったnodeletの一部をjsk_pcl_ros_utilsに移動した後,
過去のプログラム(drc関連のプログラムなど)でjsk_pcl_ros/Hogeで定義されているnodeletを変更しなくても良いよ
うにしたいという意図がありました.

YoheiKakiuchi, 2016-10-19:
別々のファイルに、何の説明もなく重複した記述があるのは気持ち悪いですね。
今後の方針としては、jsk_pcl_rosのものはpcl_rosに記述して、pcl_ros_utilのものはpcl_ros_utilに記述するようにするということでいいんだよね。

orikuma, 2016-10-19:
>  今後の方針としては、jsk_pcl_rosのものはpcl_rosに記述して、pcl_ros_utilのものはpcl_ros_utilに記述するようにするということでいいんだよね。
はい, 後方互換性を担保する必要がない新しいnodeletはそうするのが正しいと思います.

y-fuku-local, 2021-12-06:
・"/prediction/motion_predictor/objects"のトピックが流れてない。
・"/points_no_ground"の値をもとにcostmapを生成している。
・grid_map::GridMapRosConverter::toOccupancyGridのコード解析をすればすべて解決する。

grid_mapについて
https://myenigma.hatenablog.com/entry/20140714/1405343128
GridMapRosConverter.cppについて
http://docs.ros.org/en/jade/api/grid_map_ros/html/GridMapRosConverter_8cpp_source.html
pythonでoccupancygridをpublishする方法
https://memo-methods-in-researching.hatenablog.com/entry/2018/12/10/182938

  for (int idx = 0; idx < 37500; idx++) {
    out_occupancy_grid.data[idx] = 55;
  }
  cout << "out_occupancy_grid.data[0]:" << int(out_occupancy_grid.data[0]) << endl;

grid_map::GridMapRosConverter::toOccupancyGrid(costmap, COMBINED_COSTMAP_LAYER_, grid_min_value_, grid_max_value_, out_occupancy_grid);
・costmap：
・COMBINED_COSTMAP_LAYER_：costmap
・grid_min_value_：0.0
・grid_max_value_：1.0
・out_occupancy_grid：output