jsk-ros-pkg/jsk_recognition #1542: hsi_color_filter_launch  somehow fails on deb ros-indigo-jsk-pcl-ros (0.3.16-0trusty-20160216-210341-0800)

kochigami, 2016-03-07:
I installed ros-indigo-jsk-pcl-ros from deb (0.3.16-0trusty-20160216-210341-0800) , executed ```catkin build``` and tried hsi_color_filter, but it fails.
I referred to  https://github.com/jsk-ros-pkg/jsk_recognition/issues/1313, but I couldn't figure out what I have to do. Sorry for my trouble.

I tried to launch hsi_color_filter.launch as follows: (robot: turtlebot, pc: ubuntu 14.04, ros: indigo)
```
roslaunch jsk_pcl_ros hsi_color_filter.launch DEFAULT_NAMESPACE:=/camera/depth_registered INPUT:=points h_min:=-20 h_max:=30
```
The dying process is as follows:
```
... logging to /home/kochigami/.ros/log/c30c9b84-e43f-11e5-9b39-bc5ff4fcfea0/roslaunch-kochigami-desktop-8604.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
WARNING: disk usage in log directory [/home/kochigami/.ros/log] is over 1GB.
It's recommended that you use the 'rosclean' command.

started roslaunch server http://192.168.97.146:41390/

SUMMARY
========

CLEAR PARAMETERS
 * /camera/depth_registered/cluster_decomposer/

PARAMETERS
 * /camera/depth_registered/euclidean_clustering/min_size: 100
 * /camera/depth_registered/euclidean_clustering/tolerance: 0.02
 * /camera/depth_registered/hsi_filter/h_limit_max: 30
 * /camera/depth_registered/hsi_filter/h_limit_min: -20
 * /camera/depth_registered/hsi_filter/i_limit_max: 255
 * /camera/depth_registered/hsi_filter/i_limit_min: 0
 * /camera/depth_registered/hsi_filter/s_limit_max: 255
 * /camera/depth_registered/hsi_filter/s_limit_min: 0
 * /camera/depth_registered/hsi_filter/use_indices: False
 * /rosdistro: indigo
 * /rosversion: 1.11.16

NODES
  /camera/depth_registered/
    cluster_decomposer (nodelet/nodelet)
    euclidean_clustering (nodelet/nodelet)
    hsi_filter (nodelet/nodelet)
    hsi_filter_manager (nodelet/nodelet)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[camera/depth_registered/hsi_filter_manager-1]: started with pid [8635]
process[camera/depth_registered/hsi_filter-2]: started with pid [8636]
process[camera/depth_registered/euclidean_clustering-3]: started with pid [8637]
process[camera/depth_registered/cluster_decomposer-4]: started with pid [8638]
[ INFO] [1457342632.765618443]: Loading nodelet /camera/depth_registered/hsi_filter of type jsk_pcl/HSIColorFilter to manager hsi_filter_manager with the following remappings:
[ INFO] [1457342632.765899112]: /camera/depth_registered/hsi_filter/input -> /camera/depth_registered/points
[ INFO] [1457342632.765989805]: /camera/depth_registered/hsi_filter/output -> /camera/depth_registered/hsi_output
[ INFO] [1457342632.788947466]: waitForService: Service [/camera/depth_registered/hsi_filter_manager/load_nodelet] has not been advertised, waiting...
[ INFO] [1457342632.791863205]: Loading nodelet /camera/depth_registered/cluster_decomposer of type jsk_pcl/ClusterPointIndicesDecomposerZAxis to manager hsi_filter_manager with the following remappings:
[ INFO] [1457342632.792096633]: /camera/depth_registered/cluster_decomposer/boxes -> /camera/depth_registered/boxes
[ INFO] [1457342632.792163596]: /camera/depth_registered/cluster_decomposer/debug_output -> /camera/depth_registered/debug_output
[ INFO] [1457342632.792210132]: /camera/depth_registered/cluster_decomposer/input -> /camera/depth_registered/hsi_output
[ INFO] [1457342632.792242395]: /camera/depth_registered/cluster_decomposer/target -> /camera/depth_registered/euclidean_clustering/output
[ INFO] [1457342632.797188978]: Loading nodelet /camera/depth_registered/euclidean_clustering of type jsk_pcl/EuclideanClustering to manager hsi_filter_manager with the following remappings:
[ INFO] [1457342632.797520336]: /camera/depth_registered/euclidean_clustering/input -> /camera/depth_registered/hsi_output
[ INFO] [1457342632.797624086]: waitForService: Service [/camera/depth_registered/hsi_filter_manager/load_nodelet] has not been advertised, waiting...
[ INFO] [1457342632.800349345]: waitForService: Service [/camera/depth_registered/hsi_filter_manager/load_nodelet] has not been advertised, waiting...
[ INFO] [1457342632.857194609]: Initializing nodelet with 8 worker threads.
[ INFO] [1457342632.862002167]: waitForService: Service [/camera/depth_registered/hsi_filter_manager/load_nodelet] is now available.
[ INFO] [1457342632.866888560]: waitForService: Service [/camera/depth_registered/hsi_filter_manager/load_nodelet] is now available.
[ INFO] [1457342632.870917927]: waitForService: Service [/camera/depth_registered/hsi_filter_manager/load_nodelet] is now available.
[ERROR] [1457342633.465708165]: Failed to load nodelet [/camera/depth_registered/hsi_filter] of type [jsk_pcl/HSIColorFilter] even after refreshing the cache: MultiLibraryClassLoader: Could not create object of class type jsk_pcl_ros::HSIColorFilter as no factory exists for it. Make sure that the library exists and was explicitly loaded through MultiLibraryClassLoader::loadLibrary()
[ERROR] [1457342633.465771961]: The error before refreshing the cache was: Failed to load library /opt/ros/indigo/lib/libjsk_pcl_ros.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = /opt/ros/indigo/lib/libjsk_pcl_ros.so: undefined symbol: _ZTVN21jsk_recognition_utils17WallDurationTimerE)
[FATAL] [1457342633.466027448]: Service call failed!
[... 41 lines cut ...]
        ros-indigo-eigen-conversions, ros-indigo-geometry-msgs,
        ros-indigo-image-geometry, ros-indigo-image-transport,
        ros-indigo-image-view2, ros-indigo-interactive-markers,
        ros-indigo-jsk-footstep-msgs, ros-indigo-jsk-pcl-ros-utils,
        ros-indigo-jsk-perception, ros-indigo-jsk-recognition-msgs,
        ros-indigo-jsk-recognition-utils, ros-indigo-jsk-topic-tools,
        ros-indigo-kdl-conversions, ros-indigo-kdl-parser,
        ros-indigo-laser-assembler, ros-indigo-message-runtime,
        ros-indigo-ml-classifiers, ros-indigo-moveit-core,
        ros-indigo-moveit-ros-perception, ros-indigo-nav-msgs,
        ros-indigo-nodelet, ros-indigo-octomap-msgs, ros-indigo-octomap-ros,
        ros-indigo-octomap-server, ros-indigo-pcl-conversions,
        ros-indigo-pcl-msgs, ros-indigo-pcl-ros,
        ros-indigo-robot-self-filter, ros-indigo-rosboost-cfg,
        ros-indigo-roscpp-tutorials, ros-indigo-sensor-msgs,
        ros-indigo-std-msgs, ros-indigo-std-srvs, ros-indigo-stereo-msgs,
        ros-indigo-tf, ros-indigo-tf-conversions, ros-indigo-tf2-ros,
        ros-indigo-visualization-msgs
競合: ros-indigo-jsk-pcl-ros
説明: jsk_pcl_ros
 
ホームページ: http://ros.org/wiki/jsk_pcl_ros

```
Could someone help me what I have to do?

k-okada, 2016-03-07:
please check pcl version
> sudo aptitude show libpcl-common-1.7

and your source list
> ls /etc/apt/sources.list.d/
--
◉ Kei Okada

kochigami, 2016-03-07:
Thank you very much for your help.

1. ```aptitude show libpcl-common-1.7 ``` returns 
```
パッケージ: libpcl-common-1.7            
状態: インストール済み
自動的にインストールされた: はい
バージョン: 1.7.1-3+trusty2
優先度: 任意
セクション: libs
メンテナ: Jochen Sprickerhof <launchpad@jochen.sprickerhof.de>
アーキテクチャ: amd64
展開サイズ: 795 k
依存: libboost-system1.54.0, libboost-thread1.54.0, libc6 (>= 2.14), libgcc1 (>= 1:4.1.1), libgomp1 (>= 4.2.1),
        libstdc++6 (>= 4.6)
競合: libpcl-common-1.7
説明: The Point Cloud Library (or PCL) for point cloud processing - runtime
 
ホームページ: http://pointclouds.org/
```
2. The result of ```ls /etc/apt/sources.list.d/``` is below:

```kazam-team-stable-series-trusty.list  ros-latest.list  ros-latest.list.save```

kochigami, 2016-03-07:
This error says that the version of ```jsk_recognition_utils``` is old.
```
[ERROR] [1457349708.914208621]: The error before refreshing the cache was: Failed to load library /opt/ros/indigo/lib/libjsk_pcl_ros.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = /opt/ros/indigo/lib/libjsk_pcl_ros.so: undefined symbol: _ZTVN21jsk_recognition_utils17WallDurationTimerE)
```
and when I updated  
```sudo apt-get install ros-indigo-jsk-recognition-utils```, this program worked.

closed 2016-03-07 by kochigami