jsk-ros-pkg/jsk_recognition #1582: [jsk_recognition/point_pose_extractor.cpp] SetTemplateサービスの使い方が分かりません

kochigami, 2016-04-06:
point_pose_extractor.cppでは，
SetTemplateサービスを使うことでテンプレートの数を増やしていけるのだと思うのですが，
カラー画像データのdata数の指定でエラーが出てしまい困っています．

width: 205 
height: 290 (２つは画像ファイルを画像ビューワーで開いて確認)
step: width * byte_depth * num_channels = 205 * 1 * 3 = 615 
data: step * height = 615 * 290 = 178350
このように設定すると，以下のエラーが出ます．

```
ERROR: service [/SetTemplate] responded with an error: Image is wrongly formed: height * step != size  or  290 * 615 != 6
```
なぜ全体のデータ数が6になるのか分かりません．
このサービスの正しい使い方やエラーの意味が分かる方はいらっしゃいませんか．

空のサービスの中身は以下です．
```
rosservice call /SetTemplate "type: ''
image:
  header:
    seq: 0
    stamp: {secs: 0, nsecs: 0}
    frame_id: ''
  height: 0
  width: 0
  encoding: ''
  is_bigendian: 0
  step: 0
  data: ''
dimx: 0.0
dimy: 0.0
relativepose:
  position: {x: 0.0, y: 0.0, z: 0.0}
  orientation: {x: 0.0, y: 0.0, z: 0.0, w: 0.0}
savefilename: ''" 
```
以下が私の入力です．
```
 rosservice call /SetTemplate "type: 'img0001'
image:
  header:
    seq: 0
    stamp: {secs: 0, nsecs: 0}
    frame_id: ''
  height: 290
  width: 205
  encoding: 'bgr8'
  is_bigendian: 0
  step: 615
  data: '178350'
dimx: 0.1
dimy: 0.05
relativepose:
  position: {x: 0.0, y: 0.0, z: 0.0}
  orientation: {x: 0.0, y: 0.0, z: 0.0, w: 0.0}
savefilename: 'img0001.png'" 
```

step: 0
data: ''　のままにすると，
ERROR: service [/SetTemplate] responded with an error: Image is wrongly formed: step < width * byte_depth * num_channels  or  0 != 205 * 1 * 3
というエラーが出ます．

実行launchファイルの内容は以下です．
```
<launch>
  <node name="nodelet_manager"
         pkg="nodelet" type="nodelet"
         args="manager" />
  <node name="imagesift"
        pkg="nodelet" type="nodelet"
        args="load imagesift/ImageSift nodelet_manager">
    <remap from="image" to="/pepper_robot/naoqi_driver_node/camera/front/image_raw"/>
    <remap from="camera_info" to="/pepper_robot/naoqi_driver_node/camera/front/camera_info" />
  </node>

  <node name="point_pose_extractor" pkg="jsk_perception" type="point_pose_extractor"
             respawn="true" output="screen">
    <param name="template_filename" value="$(find MyPackageName)/picture/img0000.png"/>
    <param name="window_name" value="img0000"/>
  </node>
</launch>
```

wkentaro, 2016-04-06:
6になるのは、`data: '178350'`が6文字だからではないでしょうか。

wkentaro, 2016-04-06:
どうなると正解なのかわかりませんが、(`~debug_image`はでない）
https://github.com/wkentaro/20160406_point_pose_extractor
でサービスを呼ぶことはできるようです。

kochigami, 2016-04-07:
dataについて勘違いしていました．
(☓　data数を書く
◯  178350個分画素値が入る)
テンプレートを増やすことができました．

ありがとうございます．

closed 2016-04-07 by kochigami

wkentaro, 2016-04-08:
@kochigami 
今後このノードを使う人が同じような問題に当たらないために，[サンプル](https://github.com/jsk-ros-pkg/jsk_recognition/tree/master/jsk_perception/sample), [テスト](https://github.com/jsk-ros-pkg/jsk_recognition/tree/master/jsk_perception/test)または[ドキュメント](http://jsk-docs.readthedocs.org/en/latest/jsk_recognition/doc/jsk_perception/index.html)があるといいと思うのですが，
簡単に作っていただくことはできないでしょうか．

kochigami, 2016-04-10:
お返事が遅くなり申し訳ないです．
了解しました．