jsk-ros-pkg/jsk_recognition #1594: [error] Cannot build jsk_recognition

xMutzelx, 2016-04-19:
Hello,
I want to try/use JSK-Recognition, so I tried to build jsk_recognition with these steps:
```
mkdir -p jsk_ws/src
cd jsk_ws/src/
catkin_init_workspace
sudo git clone https://github.com/jsk-ros-pkg/jsk_recognition.git
cd ..
catkin_make
```
After some errors, I succsesfully ran: `rosdep install --from-paths src -r -y -i`
and `catkin_make` again.

Now I have the following errors:
```
-- Could NOT find PCL_GPU_KINFU_LARGE_SCALE (missing:  PCL_GPU_KINFU_LARGE_SCALE_LIBRARY PCL_GPU_KINFU_LARGE_SCALE_INCLUDE_DIR) 
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   program_options
CMake Error at /opt/ros/indigo/share/jsk_topic_tools/cmake/nodelet.cmake:26 (add_executable):
  add_executable cannot create target "normal_flip_to_frame" because another
  target with the same name already exists.  The existing target is an
  executable created in source directory
  "/home/marcel/jsk_ws/src/jsk_recognition/jsk_pcl_ros_utils".  See
  documentation for policy CMP0002 for more details.
Call Stack (most recent call first):
  jsk_recognition/jsk_pcl_ros/CMakeLists.txt:126 (jsk_nodelet)
  jsk_recognition/jsk_pcl_ros/CMakeLists.txt:139 (jsk_pcl_util_nodelet_deprecated)
```

After this error there are like 100 times similar errors.
Here is my CMakeError.log:
Determining if the pthread_create exist failed with the following output:
Change Dir: /home/marcel/jsk_ws/build/CMakeFiles/CMakeTmp

Run Build Command:/usr/bin/make "cmTryCompileExec3709113496/fast"
/usr/bin/make -f CMakeFiles/cmTryCompileExec3709113496.dir/build.make CMakeFiles/cmTryCompileExec3709113496.dir/build
make[1]: Entering directory `/home/marcel/jsk_ws/build/CMakeFiles/CMakeTmp'
/usr/bin/cmake -E cmake_progress_report /home/marcel/jsk_ws/build/CMakeFiles/CMakeTmp/CMakeFiles 1
Building C object CMakeFiles/cmTryCompileExec3709113496.dir/CheckSymbolExists.c.o
/usr/bin/cc    -o CMakeFiles/cmTryCompileExec3709113496.dir/CheckSymbolExists.c.o-   -c /home/marcel/jsk_ws/build/CMakeFiles/CMakeTmp/CheckSymbolExists.c
Linking C executable cmTryCompileExec3709113496
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTryCompileExec3709113496.dir/link.txt --verbose=1
/usr/bin/cc       CMakeFiles/cmTryCompileExec3709113496.dir/CheckSymbolExists.c.o  -o cmTryCompileExec3709113496 -rdynamic 
CMakeFiles/cmTryCompileExec3709113496.dir/CheckSymbolExists.c.o: In function `main':
CheckSymbolExists.c:(.text+0x16): undefined reference to `pthread_create'
collect2: error: ld returned 1 exit status
make[1]: Leaving directory `/home/marcel/jsk_ws/build/CMakeFiles/CMakeTmp'
make[1]: *** [cmTryCompileExec3709113496] Error 1
make: *** [cmTryCompileExec3709113496/fast] Error 2

File /home/marcel/jsk_ws/build/CMakeFiles/CMakeTmp/CheckSymbolExists.c:
/* */
#include <pthread.h>

int main(int argc, char** argv)
{
  (void)argv;
#ifndef pthread_create
  return ((int*)(&pthread_create))[argc];
#else
  (void)argc;
  return 0;
#endif
}

Determining if the function pthread_create exists in the pthreads failed with the following output:
Change Dir: /home/marcel/jsk_ws/build/CMakeFiles/CMakeTmp

Run Build Command:/usr/bin/make "cmTryCompileExec788283983/fast"
/usr/bin/make -f CMakeFiles/cmTryCompileExec788283983.dir/build.make CMakeFiles/cmTryCompileExec788283983.dir/build
make[1]: Entering directory `/home/marcel/jsk_ws/build/CMakeFiles/CMakeTmp'
/usr/bin/cmake -E cmake_progress_report /home/marcel/jsk_ws/build/CMakeFiles/CMakeTmp/CMakeFiles 1
Building C object CMakeFiles/cmTryCompileExec788283983.dir/CheckFunctionExists.c.o
/usr/bin/cc   -DCHECK_FUNCTION_EXISTS=pthread_create   -o CMakeFiles/cmTryCompileExec788283983.dir/CheckFunctionExists.c.o   -c /usr/share/cmake-2.8/Modules/CheckFunctionExists.c
Linking C executable cmTryCompileExec788283983
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTryCompileExec788283983.dir/link.txt --verbose=1
/usr/bin/cc   -DCHECK_FUNCTION_EXISTS=pthread_create    CMakeFiles/cmTryCompileExec788283983.dir/CheckFunctionExists.c.o  -o cmTryCompileExec788283983 -rdynamic -lpthreads 
/usr/bin/ld: cannot find -lpthreads
collect2: error: ld returned 1 exit status
make[1]: Leaving directory `/home/marcel/jsk_ws/build/CMakeFiles/CMakeTmp'
make[1]: *** [cmTryCompileExec788283983] Error 1
make: *** [cmTryCompileExec788283983/fast] Error 2

I checked libpthread and it is installed:
```
locate libpthread.so
/lib/x86_64-linux-gnu/libpthread.so.0
/lib32/libpthread.so.0
/libx32/libpthread.so.0
/usr/lib/x86_64-linux-gnu/libpthread.so
/usr/lib32/libpthread.so
/usr/libx32/libpthread.so

```
I am using Ubuntu 14.04 and Ros Indigo. 
Thank you very much for your help, I really appreciate it.

wkentaro, 2016-04-19:
The test passes with the version ec21810 .
https://travis-ci.org/jsk-ros-pkg/jsk_recognition/builds/124022356
In the travis test, it uses catkin_tools not catkin_make.

k-okada, 2016-04-19:
please try apt-get install ros-indigo-jsk-recognition

xMutzelx, 2016-04-19:
Thank you very much for your help. 
@k-okada When I use `apt-get install ros-indigo-jsk-recognition` some folders are missing compared to the git source (example: jsk_recognition/jsk_perception/scripts). That is the reason why I want to install it from source.
@wkentaro I tried to build it again with this version and now I have some different errors:
```
CMakeFiles/checkerboard_calibration.dir/src/checkerboard_calibration.cpp.o: In function `CheckerboardCalibration::image_cb(boost::shared_ptr<sensor_msgs::Image_<std::allocator<void> > const> const&)':
checkerboard_calibration.cpp:(.text._ZN23CheckerboardCalibration8image_cbERKN5boost10shared_ptrIKN11sensor_msgs6Image_ISaIvEEEEE[_ZN23CheckerboardCalibration8image_cbERKN5boost10shared_ptrIKN11sensor_msgs6Image_ISaIvEEEEE]+0xa5): undefined reference to `_IplImage::_IplImage(cv::Mat const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/marcel/jsk_ws/devel/.private/checkerboard_detector/lib/checkerboard_detector/checkerboard_calibration] Error 1
make[1]: *** [CMakeFiles/checkerboard_calibration.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
CMakeFiles/checkerboard_detector.dir/src/checkerboard_detector.cpp.o: In function `CheckerboardDetector::Detect(posedetection_msgs::ObjectDetection_<std::allocator<void> >&, sensor_msgs::Image_<std::allocator<void> > const&, sensor_msgs::CameraInfo_<std::allocator<void> > const&) [clone ._omp_fn.0]':
checkerboard_detector.cpp:(.text+0x1731): undefined reference to `cv::rectangle(cv::_InputOutputArray const&, cv::Point_<int>, cv::Point_<int>, cv::Scalar_<double> const&, int, int, int)'
checkerboard_detector.cpp:(.text+0x196b): undefined reference to `cv::cornerSubPix(cv::_InputArray const&, cv::_InputOutputArray const&, cv::Size_<int>, cv::Size_<int>, cv::TermCriteria)'
checkerboard_detector.cpp:(.text+0x1ce0): undefined reference to `cv::SimpleBlobDetector::create(cv::SimpleBlobDetector::Params const&)'
checkerboard_detector.cpp:(.text+0x1d78): undefined reference to `cv::findCirclesGrid(cv::_InputArray const&, cv::Size_<int>, cv::_OutputArray const&, int, cv::Ptr<cv::Feature2D> const&)'
checkerboard_detector.cpp:(.text+0x1e11): undefined reference to `cv::SimpleBlobDetector::create(cv::SimpleBlobDetector::Params const&)'
checkerboard_detector.cpp:(.text+0x1ea9): undefined reference to `cv::findCirclesGrid(cv::_InputArray const&, cv::Size_<int>, cv::_OutputArray const&, int, cv::Ptr<cv::Feature2D> const&)'
CMakeFiles/checkerboard_detector.dir/src/checkerboard_detector.cpp.o: In function `cv::String::String(char const*)':
checkerboard_detector.cpp:(.text._ZN2cv6StringC2EPKc[_ZN2cv6StringC5EPKc]+0x4f): undefined reference to `cv::String::allocate(unsigned long)'
CMakeFiles/checkerboard_detector.dir/src/checkerboard_detector.cpp.o: In function `cv::String::~String()':
checkerboard_detector.cpp:(.text._ZN2cv6StringD2Ev[_ZN2cv6StringD5Ev]+0x14): undefined reference to `cv::String::deallocate()'
CMakeFiles/checkerboard_detector.dir/src/checkerboard_detector.cpp.o: In function `CheckerboardDetector::CheckerboardDetector()':
checkerboard_detector.cpp:(.text._ZN20CheckerboardDetectorC2Ev[_ZN20CheckerboardDetectorC5Ev]+0x15af): undefined reference to `cv::namedWindow(cv::String const&, int)'
CMakeFiles/checkerboard_detector.dir/src/checkerboard_detector.cpp.o: In function `CheckerboardDetector::Detect(posedetection_msgs::ObjectDetection_<std::allocator<void> >&, sensor_msgs::Image_<std::allocator<void> > const&, sensor_msgs::CameraInfo_<std::allocator<void> > const&)':
checkerboard_detector.cpp:(.text._ZN20CheckerboardDetector6DetectERN18posedetection_msgs16ObjectDetection_ISaIvEEERKN11sensor_msgs6Image_IS2_EERKNS5_11CameraInfo_IS2_EE[_ZN20CheckerboardDetector6DetectERN18posedetection_msgs16ObjectDetection_ISaIvEEERKN11sensor_msgs6Image_IS2_EERKNS5_11CameraInfo_IS2_EE]+0x1556): undefined reference to `cv::line(cv::_InputOutputArray const&, cv::Point_<int>, cv::Point_<int>, cv::Scalar_<double> const&, int, int, int)'
checkerboard_detector.cpp:(.text._ZN20CheckerboardDetector6DetectERN18posedetection_msgs16ObjectDetection_ISaIvEEERKN11sensor_msgs6Image_IS2_EERKNS5_11CameraInfo_IS2_EE[_ZN20CheckerboardDetector6DetectERN18posedetection_msgs16ObjectDetection_ISaIvEEERKN11sensor_msgs6Image_IS2_EERKNS5_11CameraInfo_IS2_EE]+0x15eb): undefined reference to `cv::line(cv::_InputOutputArray const&, cv::Point_<int>, cv::Point_<int>, cv::Scalar_<double> const&, int, int, int)'
checkerboard_detector.cpp:(.text._ZN20CheckerboardDetector6DetectERN18posedetection_msgs16ObjectDetection_ISaIvEEERKN11sensor_msgs6Image_IS2_EERKNS5_11CameraInfo_IS2_EE[_ZN20CheckerboardDetector6DetectERN18posedetection_msgs16ObjectDetection_ISaIvEEERKN11sensor_msgs6Image_IS2_EERKNS5_11CameraInfo_IS2_EE]+0x1680): undefined reference to `cv::line(cv::_InputOutputArray const&, cv::Point_<int>, cv::Point_<int>, cv::Scalar_<double> const&, int, int, int)'
checkerboard_detector.cpp:(.text._ZN20CheckerboardDetector6DetectERN18posedetection_msgs16ObjectDetection_ISaIvEEERKN11sensor_msgs6Image_IS2_EERKNS5_11CameraInfo_IS2_EE[_ZN20CheckerboardDetector6DetectERN18posedetection_msgs16ObjectDetection_ISaIvEEERKN11sensor_msgs6Image_IS2_EERKNS5_11CameraInfo_IS2_EE]+0x1b75): undefined reference to `cv::circle(cv::_InputOutputArray const&, cv::Point_<int>, int, cv::Scalar_<double> const&, int, int, int)'
[... cut ...] `cv::imshow(cv::String const&, cv::_InputArray const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/marcel/jsk_ws/devel/.private/checkerboard_detector/lib/checkerboard_detector/checkerboard_detector] Error 1
make[1]: *** [CMakeFiles/checkerboard_detector.dir/all] Error 2
make: *** [all] Error 2
```
As it seems you are using OpenCV 2.4.8 and I am using OpenCV 3.1. Can this version difference cause my problems? Do you have any ideas?

Thank you very much, I really appreciate it.

wkentaro, 2016-04-19:
it should be fixed in the next release.

now the repo supports opencv2 only, so some patches are necessary.

xMutzelx, 2016-04-20:
Okay, I switched my OpenCV version to 2.4.11 and all the OpenCV related errors are solved. Now I have one error left:
```
Errors     << resized_image_transport:make /home/marcel/jsk_ws/logs/resized_image_transport/build.make.005.log       
/home/marcel/jsk_ws/src/jsk_recognition/resized_image_transport/src/image_processing_nodelet.cpp: In member function ‘void resized_image_transport::ImageProcessing::initPublishersAndSubscribers()’:
/home/marcel/jsk_ws/src/jsk_recognition/resized_image_transport/src/image_processing_nodelet.cpp:116:67: error: no matching function for call to ‘resized_image_transport::ImageProcessing::advertiseCamera(ros::NodeHandle&, const char [13], int&)’
       cp_ = advertiseCamera(*pnh_, "output/image", max_queue_size_);
                                                                   ^
/home/marcel/jsk_ws/src/jsk_recognition/resized_image_transport/src/image_processing_nodelet.cpp:116:67: note: candidate is:
In file included from /opt/ros/indigo/include/jsk_topic_tools/diagnostic_nodelet.h:44:0,
                 from /home/marcel/jsk_ws/src/jsk_recognition/resized_image_transport/include/resized_image_transport/image_processing_nodelet.h:26,
                 from /home/marcel/jsk_ws/src/jsk_recognition/resized_image_transport/src/image_processing_nodelet.cpp:1:
/opt/ros/indigo/include/jsk_topic_tools/connection_based_nodelet.h:214:5: note: image_transport::CameraPublisher jsk_topic_tools::ConnectionBasedNodelet::advertiseCamera(ros::NodeHandle&, image_transport::ImageTransport&, const string&, int)
     advertiseCamera(ros::NodeHandle& nh,
     ^
/opt/ros/indigo/include/jsk_topic_tools/connection_based_nodelet.h:214:5: note:   candidate expects 4 arguments, 3 provided
make[2]: *** [CMakeFiles/resized_image_transport.dir/src/image_processing_nodelet.cpp.o] Error 1
make[1]: *** [CMakeFiles/resized_image_transport.dir/all] Error 2
make: *** [all] Error 2
cd /home/marcel/jsk_ws/build/resized_image_transport; catkin build --get-env resized_image_transport | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
.....................................................................................................................
Failed     << resized_image_transport:make           [ Exited with code 2 ]                                          
Failed    <<< resized_image_transport                [ 8.4 seconds ]                                                 
Finished  <<< jsk_pcl_ros                            [ 1 minute and 58.1 seconds ]                                   
[build] Summary: 7 of 8 packages succeeded.                                                                          
[build]   Ignored:   1 packages were skipped or are blacklisted.                                                     
[build]   Warnings:  None.                                                                                           
[build]   Abandoned: None.                                                                                           
[build]   Failed:    1 packages failed.                                                                              
[build] Runtime: 2 minutes and 0.9 seconds total.                                 
```
Thank you so much for your help. Your support is great!

k-okada, 2016-04-20:
On Tue, Apr 19, 2016 at 8:16 PM, xMutzelx <notifications@github.com> wrote:

> @k-okada <https://github.com/k-okada> When I use apt-get install
> ros-indigo-jsk-recognition some folders are missing compared to the git
> source (example: jsk_recognition/jsk_perception/scripts). That is the
> reason why I want to install it from source.

That should be fixed in 0.3.20 (
https://github.com/jsk-ros-pkg/jsk_recognition/releases/tag/0.3.20) and
alreday released into shadow-fixed (
http://repositories.ros.org/status_page/ros_indigo_default.html?q=jsk_perception)
, and will be releasead release repository soon.

If you can not wait, please try using shadow-fixed repo
http://wiki.ros.org/ShadowRepository
--
◉ Kei Okada

xMutzelx, 2016-04-20:
@k-okada I tried your suggestion, but there are still missing folders (example: jsk_recognition/jsk_perception/scripts).
```
Setting up ros-indigo-jsk-recognition-msgs (0.3.21-0trusty-20160415-101236-0700) ...
Setting up ros-indigo-checkerboard-detector (0.3.21-0trusty-20160415-102438-0700) ...
Setting up ros-indigo-jsk-topic-tools (2.0.11-0trusty-20160405-210803-0700) ...
Setting up ros-indigo-jsk-recognition-utils (0.3.21-0trusty-20160415-102751-0700) ...
Setting up ros-indigo-jskeus (1.0.12-0trusty-20160404-030237-0700) ...
Setting up ros-indigo-roseus (1.5.0-0trusty-20160404-031122-0700) ...
Setting up ros-indigo-jsk-perception (0.3.21-0trusty-20160415-105052-0700) ...
Setting up ros-indigo-imagesift (0.3.21-0trusty-20160415-113610-0700) ...
Setting up ros-indigo-jsk-footstep-msgs (2.0.1-0trusty-20160321-184104-0700) ...
Setting up ros-indigo-jsk-pcl-ros-utils (0.3.21-0trusty-20160415-210810-0700) ...
Setting up ros-indigo-jsk-pcl-ros (0.3.21-0trusty-20160415-215100-0700) ...
Setting up ros-indigo-resized-image-transport (0.3.21-0trusty-20160415-101348-0700) ...
Setting up ros-indigo-jsk-recognition (0.3.21-0trusty-20160416-025111-0700) ...
marcel@marcel-H67A-UD3H-B3:~$ roscd jsk_perception/
marcel@marcel-H67A-UD3H-B3:/opt/ros/indigo/share/jsk_perception$ ls
cmake  euslisp  jsk_perception_nodelets.xml  launch  package.xml  sample  srv
```

wkentaro, 2016-04-20:
@xMutzelx the scripts in `jsk_recognition/jsk_perception/scripts/` directory is located in `/opt/ros/indigo/lib/jsk_perception` when you install it via apt.

k-okada, 2016-04-20:
btw, we're testing on opencv3 environment and it seems working -> https://travis-ci.org/jsk-ros-pkg/jsk_recognition/jobs/124450393

xMutzelx, 2016-04-21:
@wkentaro I fixed my problem with the installation from source. My pip version was 1.5.4, after I upgraded it to 8.1.1 and ran: `rosdep install --from-paths src -r -y -i`. After this is tried `catkin build` again and it worked perfectly. Thank you very much for all of your help.
@k-okada I found all of the "missing" folders after ```apt-get install ros-indigo-jsk-recognition```. Thank you very much for all of your help.

I can approve that both installations are now working fine for me.

closed 2016-04-21 by xMutzelx