jsk-ros-pkg/jsk_recognition #1616: [Problem] Bag of Features for Object Recognition

xMutzelx, 2016-04-25:
Hello,
I am following this example: http://jsk-recognition.readthedocs.org/en/latest/jsk_perception/bof_object_recognition.html

When I type:` rosrun jsk_perception create_sift_dataset.py 20150428_collected_images` I get the following error:
```
Traceback (most recent call last):
  File "/home/marcel/jsk_ws/src/jsk_recognition/jsk_perception/scripts/create_sift_dataset.py", line 31, in <module>
    import imagesift
ImportError: No module named imagesift
```
I succssfully installed jsk-recognition and jsk-3rdparty. I am using Ubuntu 14.04 and Ros Indigo. I installed OpenCV 2.4.11. Do you have any ideas how I can fix this?
Thank you for your help.

wkentaro, 2016-04-25:
This seems problem in `libsiftfastpy`. https://github.com/jsk-ros-pkg/jsk_3rdparty/tree/master/3rdparty/libsiftfast
because the `siftfastpy.so` should be installed and properly loaded after installed via apt.

xMutzelx, 2016-04-25:
I installed it from source and not via apt.
I can find siftfastpy.o by hand:
```
marcel@marcel-H67A-UD3H-B3:~/jsk_ws/build/libsiftfast/libsiftfast_svn/build/CMakeFiles/siftfastpy.dir$ ls
build.make         CXX.includecache  depend.internal  flags.make     link.txt       siftfastpy.o
cmake_clean.cmake  DependInfo.cmake  depend.make      libsiftfast.o  progress.make

```

wkentaro, 2016-04-25:
Ok, if so, it should be installed at:
```
~/jsk_ws/devel/lib/python2.7/dist-packages/siftfastpy.so
```

xMutzelx, 2016-04-25:
Here is no siftfastpy.so and I also cant find it in subfolders. 
```
marcel@marcel-H67A-UD3H-B3:~/jsk_ws/devel/lib/python2.7/dist-packages$ ls
bayesian                  jsk_pcl_ros        jsk_recognition_msgs   opt_camera
bayesian_belief_networks  jsk_pcl_ros_utils  jsk_recognition_utils  pgm_learner
imagesift                 jsk_perception     mini_maxwell           resized_image_transport
```
I also found a link at: `/home/marcel/jsk_ws/devel`

wkentaro, 2016-04-25:
There is a condition in `build/libsiftfast/libsiftfast_svn/CMakeLists.txt` as below:
```
if( HAVE_ALL_PYTHON_HEADERS )
      message(STATUS "python and boost-python found")
      include_directories(${PYTHON_INCLUDE_PATH} ${Boost_INCLUDE_DIRS})
      link_directories(${Boost_LIBRARY_DIRS})
      add_library(siftfastpy SHARED siftfastpy.cpp libsiftfast.cpp)
      # stdc++ has to be included before opengl libraries due to some ATI bug (http://wiki.fifengine.de/Segfault_in_cxa_allocate_exception#Workaround)
      target_link_libraries(siftfastpy ${STDC_LIBRARY} ${PYTHON_LIBRARIES} ${Boost_PYTHON_LIBRARY})
```
there is some missing dependencies in your environment.

wkentaro, 2016-04-25:
Please check the message on built time.

wkentaro, 2016-04-25:
I found catkin_tools 0.4.0 does not run execute_process cmake function, and it may cause the bug.

wkentaro, 2016-04-25:
Ah, it's my misunderstanding, https://github.com/jsk-ros-pkg/jsk_3rdparty/blob/master/3rdparty/libsiftfast/CMakeLists.txt#L33-L35 line fails because of changed catkin_tools api.

wkentaro, 2016-04-25:
https://github.com/jsk-ros-pkg/jsk_3rdparty/pull/62 should fix the problem of wrong location of siftfastpy.so.

xMutzelx, 2016-04-26:
Your fix works fine. Thank you for your help, I really appreciate it.

closed 2016-04-26 by xMutzelx