jsk-ros-pkg/jsk_recognition #1620: [question] Bag of Features for Object Recognition

xMutzelx, 2016-04-27:
I have a question about the "Bag of Features for Object Recognition" example (http://jsk-docs.readthedocs.org/en/latest/jsk_recognition/doc/jsk_perception/bof_object_recognition.html). I followed the example and everything worked fine (no errors or critical warnings) but I do not get such output windows, like shown in the example. I tried to edit the launch file and add one "image_view" without success. Can you please give me a hint or a short explanation?

wkentaro, 2016-04-27:
see #1622

xMutzelx, 2016-04-28:
Thank you very much it looks awesome! I have a problem with the histogram: 
```
/home/marcel/jsk_ws/src/jsk_visualization/jsk_rqt_plugins/src/jsk_rqt_plugins/hist.py:122: SyntaxWarning: The publisher should be created with an explicit keyword argument 'queue_size'. Please see http://wiki.ros.org/rospy/Overview/Publishers%20and%20Subscribers for more information.
  self.pub_image = rospy.Publisher(topic_name + "/histogram_image", Image)
Traceback (most recent call last):
  File "/home/marcel/jsk_ws/src/jsk_visualization/jsk_rqt_plugins/src/jsk_rqt_plugins/hist.py", line 162, in update_plot
    pos = np.arange(len(xs))
TypeError: object of type 'VectorArray' has no len()
```
This happens when I subscribe to the topic: `/bof_histogram_extractor/output` 
Is this the wrong topic? I do not have another topic from `bof_histogram_extractor`

wkentaro, 2016-04-28:
See the picture crefully.

xMutzelx, 2016-04-28:
Okay, I found my error ... `/bof_histogram_extractor/output/data`
I did not look at the picture, I just checked `rostopic list`. Thank you for your advice.

closed 2016-04-29 by xMutzelx