jsk-ros-pkg/jsk_recognition #1631: [doc] No Documentation Nodes

knorth55, 2016-04-29:
There is no document about nodes below.
Author or people familiar with node should write doc, or no one reuse it and others duplicate similar node.

```
jsk_pcl/InteractiveCuboidLikelihood
jsk_pcl/FindObjectOnPlane
jsk_pcl/DepthCalibration
jsk_pcl/LineSegmentCollector
jsk_pcl/LineSegmentDetector
jsk_pcl/BilateralFilter
jsk_pcl/JointStateStaticFilter
jsk_pcl/OrganizedPassThrough
jsk_pcl/BoundingBoxFilter
jsk_pcl/MultiPlaneSACSegmentation
jsk_pcl/EdgebasedCubeFinder
jsk_pcl/ParallelEdgeFinder
jsk_pcl/EdgeDepthRefinement
jsk_pcl/OrganizedEdgeDetector
jsk_pcl/HandleEstimator
jsk_pcl/GridSampler
jsk_pcl/EnvironmentPlaneModeling
jsk_pcl/ColorHistogramMatcher
jsk_pcl/NormalEstimationIntegralImage
jsk_pcl/SelectedClusterPublisher
jsk_pcl/ImageRotateNodelet
jsk_pcl/VoxelGridDownsampleManager
jsk_pcl/VoxelGridDownsampleDecoder
jsk_pcl/ColorizeRandomForest
jsk_pcl/ColorizeMapRandomForest
jsk_pcl/PointCloudMoveitFilter

jsk_pcl_utils/TransformPointcloudInBoundingBox
jsk_pcl_utils/DelayPointCloud
jsk_pcl_utils/NormalConcatenater
jsk_pcl_utils/TfTransformBoundingBoxArray
jsk_pcl_utils/ColorizeDistanceFromPlane
jsk_pcl_utils/PolygonArrayWrapper
jsk_pcl_utils/PolygonArrayUnwrapper
jsk_pcl_utils/PolygonAppender
jsk_pcl_utils/StaticPolygonArrayPublisher
jsk_pcl_utils/PlaneRejector
jsk_pcl_utils/PlaneReasoner
```

k-okada, 2016-05-06:
@chiwunau @YuOhara @iory @furushchev @wkentaro @otsubo @kochigami @mmurooka @iKrishneel とりあえず関係ありそう，勉強したら良さそう，という人にアサインしたので，中身をみてドキュメントを書いてみて下さい．ドキュメントの書き方は @knorth55 か @wkentaro に聞いて下さい．

wkentaro, 2016-05-06:
Alright.

wkentaro, 2016-05-09:
担当分送りました: https://github.com/jsk-ros-pkg/jsk_recognition/pull/1653 https://github.com/jsk-ros-pkg/jsk_recognition/pull/1652 https://github.com/jsk-ros-pkg/jsk_recognition/pull/1650 https://github.com/jsk-ros-pkg/jsk_recognition/pull/1649  https://github.com/jsk-ros-pkg/jsk_recognition/pull/1647

iory, 2016-05-09:
こちら見落としていました．
とりかかります．

k-okada, 2016-05-09:
上のチェックボックスonにできないかな？ > @ekentaro

wkentaro, 2016-05-09:
@k-okada マージ願います。

mmurooka, 2016-05-13:
担当分マージされましたので，チェックボックス付けました．https://github.com/jsk-ros-pkg/jsk_recognition/pull/1654

wkentaro, 2016-05-15:
> @mizomizo, @himlen1990 @ishiguroJSK @Horisu @mukuact とりあえず関係ありそう，勉強したら良さそう，という人にアサインしたので，中身をみてドキュメントを書いてみて下さい．ドキュメントの書き方は @knorth55 か @wkentaro に聞いて下さい．

ちなみにですが、ローカルでのドキュメントの生成方法はここにあります。

PR #2393 (2019-03-19): [doc] [jsk_pcl_ros_utils] [jsk_pcl_ros] Add documentation
Closes #1631 

This PR fixes code minimally and adds sample, test and documentation for below nodes.

__jsk_pcl_ros__
- [x] jsk_pcl/InteractiveCuboidLikelihood
- [x] jsk_pcl/FindObjectOnPlane
- [x] jsk_pcl/BilateralFilter
- [x] jsk_pcl/JointStateStaticFilter
- [x] jsk_pcl/MultiPlaneSACSegmentation
- [x] jsk_pcl/EdgebasedCubeFinder
- [x] jsk_pcl/HandleEstimator
- [x] jsk_pcl/EnvironmentPlaneModeling
- [x] jsk_pcl/SelectedClusterPublisher 
- [x] jsk_pcl/VoxelGridDownsampleManager
- [x] jsk_pcl/VoxelGridDownsampleDecoder
- [x] jsk_pcl/Colo [... cut]