jsk-ros-pkg/jsk_recognition #1643: [question] How to learn and recognize Objects

xMutzelx, 2016-05-02:
I would like to learn and recognize 3D-objects with the Kinect camera. I read the documentation from jsk_pcl_ros and I couldn't find some informations.
Here are my questions: 
- Which DB are you using to store the models from "IncrementalModelRegistration" or "LINEMODTrainer" ?  
- Do you have any tutorials which guide me through the first steps (Training and Detection)
- Is "LINEMODTrainer" the only detection nodelet, or do you use/offer some others?
Thank you very much for your help. I really appreciate it.

YuOhara, 2016-05-06:
For now, it may be better to use http://wg-perception.github.io/object_recognition_core/ which has linemod functions. It has good tutorials.
The performance of linemod in this repo is poor and the reason may because
* the performance of incremental registration(which uses icp) is not good, it does not have loop closing function.

also, I think it is better to use mesh file instead of point data for getting linemod-training data.
http://wg-perception.github.io/object_recognition_core/  use mesh data but jsk_package uses point data.

we want to improve the linemod code, however it may take time.
So please also check http://wg-perception.github.io/object_recognition_core.

wkentaro, 2016-10-30:
@xMutzelx Please comment or reopen issue when you have additional questions.

closed 2016-10-30 by wkentaro