jsk-ros-pkg/jsk_recognition #1644: [Documentation] Rename "PolygonToMaskImage"

xMutzelx, 2016-05-03:
Can you please rename: [link](http://jsk-docs.readthedocs.io/en/latest/jsk_recognition/doc/jsk_perception/nodes/robot_to_mask_image.html)?
This name is already used by another nodelet: [link](http://jsk-docs.readthedocs.io/en/latest/jsk_recognition/doc/jsk_perception/nodes/polygon_to_mask_image.html).

mmurooka, 2016-05-03:
Thank you for letting us know that.
[1] is the instruction of  [2].
[1] http://jsk-docs.readthedocs.io/en/latest/jsk_recognition/doc/jsk_perception/nodes/robot_to_mask_image.html
[2] https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/jsk_perception/src/robot_to_mask_image.cpp
Only the nodelet name in the document is mistaken and I fixed that.

wkentaro, 2016-05-25:
Fixed via #1645

closed 2016-05-25 by wkentaro