jsk-ros-pkg/jsk_recognition #1648: [doc/jsk_perception] add documentation

k-okada, 2016-05-06:
- [ ] jsk_perception/EdgeDetector @mizo-hana (if http://wiki.ros.org/opencv_apps#edge_detection	is ok, please remove) @mizmizo 
- [ ] jsk_perception/SparseImageEncoder  https://github.com/jsk-ros-pkg/jsk_recognition/pull/62 @yuyu2172 
- [ ] jsk_perception/SparseImageDecoder  https://github.com/jsk-ros-pkg/jsk_recognition/pull/62 @yuyu2172 
- [ ] jsk_perception/BackgroundSubstraction https://github.com/jsk-ros-pkg/jsk_recognition/pull/473 @mizmizo 
- [ ] jsk_perception/HoughCircleDetector (if http://wiki.ros.org/opencv_apps#hough_circles	is ok, please remove) @mizmizo 
- [ ] jsk_perception/GrabCut https://github.com/jsk-ros-pkg/jsk_recognition/pull/534 @himlen1990 
- [ ] jsk_perception/RGBDecomposer (https://github.com/jsk-ros-pkg/jsk_recognition/pull/578) need labeling? histogram? sample code? @ishiguroJSK 
- [ ] jsk_perception/HSVDecomposer (https://github.com/jsk-ros-pkg/jsk_recognition/pull/579) @ishiguroJSK 
- [ ] jsk_perception/ContourFinder (if http://wiki.ros.org/opencv_apps#find_contours is ok, please remove) @mizmizo 
- [ ] jsk_perception/SnakeSegmentation https://github.com/jsk-ros-pkg/jsk_recognition/pull/582 @Horisu 
- [ ] jsk_perception/ColorizeLabels https://github.com/jsk-ros-pkg/jsk_recognition/pull/589 @ishiguroJSK 
- [ ] jsk_perception/SingleChannelHistogram https://github.com/jsk-ros-pkg/jsk_recognition/pull/613 @ishiguroJSK 
- [ ] bof_histogram_extractor.py https://github.com/jsk-ros-pkg/jsk_recognition/pull/1254 @mukuact 
- [ ] create_feature0d_dataset.py https://github.com/jsk-ros-pkg/jsk_recognition/pull/1001 @mukuact 
- [ ] extract_images_sync  https://github.com/jsk-ros-pkg/jsk_recognition/pull/1420 @wesleypchan 
- [ ] fisheye_ray.py https://github.com/jsk-ros-pkg/jsk_recognition/pull/841 @iKrishneel 
- [ ] image_publisher.py https://github.com/jsk-ros-pkg/jsk_recognition/pull/696 @Horisu 
- [ ] image_saver_sync https://github.com/jsk-ros-pkg/jsk_recognition/pull/1420 @wkentaro 
- [ ] image_time_diff.py https://github.com/jsk-ros-pkg/jsk_recognition/pull/913 @Horisu 
- [ ] matchtemplate.py https://github.com/jsk-ros-pkg/jsk_recognition/commit/96869e7d22d40d901d5617cab84cd54e2be21e34 @mukuact 
- [ ] non_maximum_suppression.py https://github.com/jsk-ros-pkg/jsk_recognition/pull/866 @himlen1990 
- [ ] publish_header https://github.com/jsk-ros-pkg/jsk_recognition/pull/1420 @wkentaro 
- [ ] random_forest_server.py https://github.com/jsk-ros-pkg/jsk_recognition/pull/1426 @himlen1990 
- [ ] unwrap_histogram_with_range_array.py https://github.com/jsk-ros-pkg/jsk_recognition/pull/1392 @chiwunau

k-okada, 2016-05-06:
@mizomizo, @himlen1990 @ishiguroJSK @Horisu @mukuact  とりあえず関係ありそう，勉強したら良さそう，という人にアサインしたので，中身をみてドキュメントを書いてみて下さい．ドキュメントの書き方は @knorth55 か @wkentaro に聞いて下さい．

wkentaro, 2016-05-06:
@mukuact create_feature0d_dataset.py is deprecated. : https://github.com/jsk-ros-pkg/jsk_recognition/pull/1655

furushchev, 2016-05-06:
@k-okada maybe @yuyu2172 -> @furushchev ?

k-okada, 2016-05-09:
@horisu ping

wkentaro, 2016-05-09:
担当分送りました: https://github.com/jsk-ros-pkg/jsk_recognition/pull/1651 https://github.com/jsk-ros-pkg/jsk_recognition/pull/1655

mizmizo, 2016-05-13:
担当4つ中、一旦2つまで書きました。#1664

himlen1990, 2016-05-15:
I have pull requests in #1666, I am not sure whether #1426 is use for
training a model or use a exist model to classify data. I think this code
is wrote by @wkentaro <https://github.com/wkentaro>  and he knows more.

wkentaro, 2016-05-15:
@himlen1990 
Actually, I just migrated `sklearn` ros package  to `jsk_perception` with #1426 and https://github.com/jsk-ros-pkg/jsk_3rdparty/pull/53.
So I'm not the author and does not have answer for your question, but maybe you can try the [sample launch](https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/jsk_perception/sample/random_forest_sample.launch).

wkentaro, 2016-05-15:
> @mizomizo, @himlen1990 @ishiguroJSK @Horisu @mukuact とりあえず関係ありそう，勉強したら良さそう，という人にアサインしたので，中身をみてドキュメントを書いてみて下さい．ドキュメントの書き方は @knorth55 か @wkentaro に聞いて下さい．

ちなみにですが、ローカルでのドキュメントの生成方法は[ここ](https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/doc%2FREADME.md)にあります。

kochigami, 2016-05-20:
Hi, I'm preparing sample, test and doc for point_pose_extractor.cpp (#1582), 
but I haven't sent pull-request yet. 
(It is not included here though.)

wkentaro, 2016-08-01:
@yuyu2172 @mizmizo @himlen1990 @ishiguroJSK @Horisu @mukuact @iKrishneel Have you finished writing the documents?

yuyu2172, 2016-08-01:
@wkentaro 
not yet, sorry.

Ridhwanluthra, 2018-02-27:
I would like to do the documentation for non-maximum suppression nodelet.
Is anyone working on it?

knorth55, 2018-02-28:
@Ridhwanluthra 
`NonMaximumSuppression` is the service server used in `SlidingWindowObjectDetection` nodelet.
There is already a doc, but if you want to improve the doc, you can do it. http://jsk-recognition.readthedocs.io/en/latest/jsk_perception/nodes/non_maximum_suppression.html

Ridhwanluthra, 2018-03-04:
@knorth55 is explaining the process of nonMaximumSuppression there of any value?

knorth55, 2018-03-05:
@Ridhwanluthra I think the doc is not perfectly written.
If you want to tell more detailed information, you can write the doc.
But i recommend you to try `SlidingWindowDetection` first.

knorth55, 2019-01-30:
@YutoUchimi  :100: