jsk-ros-pkg/jsk_recognition #1669: [jsk_perception] bof_feature_extractor.py with old pickle does not work

wkentaro, 2016-05-16:
If you install sklearn with pip, it raise error with old pickled data. (because of changed api)
```
[ERROR] [WallTime: 1463393760.278256] bad callback: <bound method Subscriber.callback of <message_filters.Subscriber object at 0x7fe54077a1d0>>
Traceback (most recent call last):
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/topics.py", line 720, in _invoke_callback
    cb(msg)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/message_filters/__init__.py", line 74, in callback
    self.signalMessage(msg)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/message_filters/__init__.py", line 56, in signalMessage
    cb(*(msg + args))
  File "/opt/ros/indigo/lib/python2.7/dist-packages/message_filters/__init__.py", line 199, in add
    self.signalMessage(*msgs)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/message_filters/__init__.py", line 56, in signalMessage
    cb(*(msg + args))
  File "/home/baxter/ros/indigo/src/jsk-ros-pkg/jsk_recognition/jsk_perception/node_scripts/bof_histogram_extractor.py", line 75, in _apply
    X = self.bof.transform(X)
  File "/home/baxter/ros/indigo/src/jsk-ros-pkg/jsk_recognition/jsk_recognition_utils/python/jsk_recognition_utils/feature.py", line 24, in transform
    return np.vstack([self.make_hist(xi.reshape((-1, 128))) for xi in X])
  File "/home/baxter/ros/indigo/src/jsk-ros-pkg/jsk_recognition/jsk_recognition_utils/python/jsk_recognition_utils/feature.py", line 31, in make_hist
    indices = nn.kneighbors(descriptors, return_distance=False)
  File "/home/baxter/.local/lib/python2.7/site-packages/sklearn/neighbors/base.py", line 366, in kneighbors
    n_jobs = _get_n_jobs(self.n_jobs)
AttributeError: 'NearestNeighbors' object has no attribute 'n_jobs'
```

k-okada, 2016-05-17:
on ubuntu, we're expected to use sklearn from apt -> https://github.com/ros/rosdistro/blob/master/rosdep/python.yaml#L2007

wkentaro, 2016-05-17:
Sometimes I and others need latest scikit-learn.

closed 2016-05-19 by k-okada

rawalkhirodkar, 2016-08-22:
Sorry I am new to this. I am still facing this problem.
I woould appreciate some help, thank you

wkentaro, 2016-08-23:
@rawalkhirodkar Please open a new issue with error logs and the outputs of `pip show scikit-image` and `rosversion jsk_perception`.

PR #1670 (2016-05-19): [jsk_perception] Support latest sklearn in BoF feature extraction
Closes #1669